The-OpenROAD-Project/OpenLane #1002: Question about "XXX is not routable by TritonRoute. Move to special nets Issue Number"

0616ygh, 2022-03-14:
### Description
I used Cadence Genus to synthesize "BM64" (an example provided by OpenLane), which was mapped with sky130A. 
I imported the synthesized gate-level netlist into OpenLane for Interactive Mode.
The following problem occurred during "run_routing":
[image]

### Environment
```
Kernel: Linux v4.18.0-240.1.1.el8_3.x86_64
Distribution: centos 8
Python: v3.10.2 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: 5f20beb7928c7329ea1a199b9f494f48f2e6c080
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

5f20beb 2022-03-11T01:56:51+02:00 Fix issue with threads option in `run_designs.py` (#994) - Donn -  (HEAD -> master, tag: 2022.03.11_02.27.13, origin/master, origin/HEAD)
a82bf22 2022-03-11T01:47:51+02:00 scripts/checkers: fix report_file_relative scope (#989) - Johan Euphrosine -  ()
ec35dd7 2022-03-11T01:47:12+02:00 scripts/tcl_commands/scrot_klayout: add missing -log arguments (#990) - Johan Euphrosine -  (). 
```

### Reproduction Material
* Upload a tarball containing the relevant design.
[openlane_reproducible_9.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8245525/openlane_reproducible_9.zip)
//I think this zip is missing some files because in error.log there is something like this: [ERROR]: Failed to package reproducible.

* List the commands used to run the design.
`./flow.tcl -interactive`
`prep -design BM64`
`run_floorplan`
`run_placement`
`run_cts`
`run_routing`

### Expected behavior
The flow finishes without errors.

### Logs
```
OpenROAD 944855835623e651e7b9c7c50efcce1fb04b4fee 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /openlane/designs/BM64/runs/RUN_2022.03.13_08.38.40/tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/designs/BM64/runs/RUN_2022.03.13_08.38.40/tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: /openlane/designs/BM64/runs/RUN_2022.03.13_08.38.40/tmp/routing/28-global.def
[INFO ODB-0128] Design: BM64
[INFO ODB-0094] 		Created 100000 Insts
[INFO ODB-0130]     Created 1030 pins.
[INFO ODB-0131]     Created 124603 components and 509986 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 470976 connections.
[INFO ODB-0133]     Created 8544 nets and 39010 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/BM64/runs/RUN_2022.03.13_08.38.40/tmp/routing/28-global.def
[INFO ORD-0030] Using 2 thread(s).
[INFO DRT-0149] Reading tech and libs.

Units:                1000
Number of layers:     13
Number of macros:     441
Number of vias:       25
Number of viarulegen: 25

[INFO DRT-0150] Reading design.
[ERROR DRT-0305] Net zero_ of signal type GROUND is not routable by TritonRoute. Move to special nets.
Error: droute.tcl, 46 DRT-0305

---------------------------------------------------------------------------------------------------------------------
[ERROR]: Failed to package reproducible.
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[ERROR]: during executing: "openroad -python /openlane/scripts/tr_drc_to_klayout_drc.py -i /openlane/designs/BM64/runs/RUN_2022.03.13_08.38.40/reports/routing/detailed.drc -o /openlane/designs/BM64/runs/RUN_2022.03.13_08.38.40/reports/routing/detailed.klayout.xml --design-name BM64"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
OpenROAD 944855835623e651e7b9c7c50efcce1fb04b4fee 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Traceback (most recent call last):
  File "/openlane/scripts/tr_drc_to_klayout_drc.py", line 58, in <module>
    with open(input_file_name, "r") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/openlane/designs/BM64/runs/RUN_2022.03.13_08.38.40/reports/routing/detailed.drc'
```

arlpetergadfort, 2022-03-27:
@maliberty this is the same issue as I mentioned to you before. If you don't use genus to tie off the tie hi and tie lo nets, opensta creates zero_ and one_ for those nets and because openroad has no native way to insert the tieoffs they are left floating and incorrectly defined.

maliberty, 2022-03-27:
@arlpetergadfort what is different about the incoming verilog from versus what we get from yosys where we don't see this issue?  Does it not use 1'b1 / 1'b0 for constants?

maliberty, 2022-03-27:
I need to see the upstream .v not the the router input .def to understand the issue.

0616ygh, 2022-03-27:
I'm sorry I'm late.
I checked the relevant files and found a "28-global.v" file in "run/tmp/routing". I think this is upstream. v.
I also packaged genus-synthesized netlist and the Verilog files resized during placement and routing into a zip file.
Thank you for your help.
[verilogcode.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8357113/verilogcode.zip)

arlpetergadfort, 2022-03-27:
> @arlpetergadfort what is different about the incoming verilog from versus what we get from yosys where we don't see this issue?  Does it not use 1'b1 / 1'b0 for constants?

The netlist was generated in genus and not yosysy according to the OP.

maliberty, 2022-03-28:
I believe the Genus command is add_tieoffs

arlpetergadfort, 2022-03-28:
> Does Genus not have a command equivalent to yosys' hilomap? That would be surprising to me.
> 
> We could add an insert_tiecells command if needed.

What tool would `insert_tiecells` belong to? I already have some code that performs this for us and would be relatively easy to put a PR together for this. Since we didnt use genus to insert the tiecells either, it would be helpful for us as well.

maliberty, 2022-03-28:
It could go in ifp but is there a reason you don't insert ties with Genus?

arlpetergadfort, 2022-03-28:
> It could go in ifp but is there a reason you don't insert ties with Genus?

Because I could do it in Innovus. No real reason except that we didn't.

maliberty, 2022-03-28:
In https://github.com/The-OpenROAD-Project/OpenROAD/pull/1724 @arlpetergadfort added an insert_tiecells command.  So you can solve this either in Genus or in OR.

vijayank88, 2022-04-27:
@donn 
For this issue, is any configuration variable there to define tiecell list, so OpenROAD can insert tie cells during floorplan?

vijayank88, 2022-05-31:
@donn 
How do I enable insert_tiecells command with OpenLane flow?

arlpetergadfort, 2022-05-31:
> @donn How do I enable `insert_tiecells` command with OpenLane flow?

I added the following to the ORFS floorplanning script for us:
```
insert_tiecells [join $env(TIELO_CELL_AND_PORT) /]
insert_tiecells [join $env(TIEHI_CELL_AND_PORT) /]
```

vijayank88, 2022-07-05:
@donn 
Any update for openlane flow scripts, how to set `insert_tiecells`?

donn, 2022-07-05:
You may start a new enhancement issue for that.

closed 2022-07-05 by donn