The-OpenROAD-Project/OpenLane #1004: Design reports after routing show technically-true, but practically-useless, utilization numbers

tdene, 2022-03-14:
### Description
During the implementation of any design, OpenLane will create a `reports` folder.
This `reports` folder has, among others, an area report.
The area report displays both raw area, as well as % utilization.

All 

### Environment
```
Please run the following command in the OpenLane folder:

python3 ./env.py issue-survey

And copy and paste the ENTIRE output between the triple-backticks. Please do not gzip and upload the output.

If there's no env.py, you are using an out of date version of OpenLane and should probably update. 
```

### Reproduction Material
* Upload a tarball containing the relevant design.
* List the commands used to run the design.

If you see a message like `Reproducible packaged: Please tarball and upload <PATH> if you're going to submit an issue` in your logs, please also tarball and include that path. This will greatly speed up the fixing process.

### Expected behavior
A clear and concise description of what you expected to happen.

### Logs
```
Add any relevant logs here. Please do ensure they're enclosed by the triple-backticks.
```

maliberty, 2022-03-18:
This is because OL chooses to fill with decap cells not filler cells.  Filler cells are excluded but decaps are not fillers.  It is an OL decision and not a tool issue.  @donn

kareefardi, 2023-02-02:
I don't think that is applicable anymore. I just checked the reports and utilization number seems to ignore both decap and fill cells. Perhaps that is happening since both have `CLASS CORE SPACER`. @maliberty If you think my interpretation is wrong and believe this might still be an issue please re-open it.

closed 2023-02-02 by kareefardi