The-OpenROAD-Project/OpenLane #997: GND and VCC short circuit error

Sukru-Uzun, 2022-03-14:
```
Kernel: Linux v5.11.0-49-generic
Distribution: ubuntu 21.04
Python: v3.9.5 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: -dev
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

e4bfdd7 2022-02-22T16:59:14-03:00 Remove pip install, no longer needed (#953) - Vitor Bandeira -  (grafted, HEAD, tag: 2022.02.23_02.50.41)
```

I have tried to run openlane flow for my design but it has below strange error. It gave me lvs error due to short circuit of vcc and gnd. I could not find what causes that situation. GL and RTL verilog files seems connected correctly. Besides, I run `make user_proj_example` command to synthesize inherent modules. However, the error was from `user_project_wrapper`.

```
 Subcircuit pins:
 Circuit 1: user_project_wrapper            |Circuit 2: user_project_wrapper
 -------------------------------------------|-------------------------------------------
 vssd2                                                   |vccd1 **Mismatch**
 vssd1                                                   |vssd1
 wb_clk_i                                              |wb_clk_i
 wb_rst_i                                               |wb_rst_i
...
```

vijayank88, 2022-03-14:
@Sukru-Uzun 
Can you attach issue_reproducible based on the issue: https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

Sukru-Uzun, 2022-03-14:
There is no issue_reproducible file for this error. Here is my lvs log file which mentions error.
[30-user_project_wrapper.lef.log](https://github.com/The-OpenROAD-Project/OpenLane/files/8244031/30-user_project_wrapper.lef.log)

d-m-bailey, 2022-03-14:
@Sukru-Uzun Can you check your extract output? It might show a short there. Also if you have all the data in your repo, you could share that and I could take a look at it.

Sukru-Uzun, 2022-03-14:
@d-m-bailey, what do you mean by `extract output`? Is it a log file or somt like .gds?
[31-ext2gds.spice.feedback.txt](https://github.com/The-OpenROAD-Project/OpenLane/files/8245319/31-ext2gds.spice.feedback.txt)

This is my [repo](https://github.com/Procenne-Digital-Design/secure-memory/tree/sram) but it does not have post-processing file in it.

Sukru-Uzun, 2022-03-14:
> 

I uploaded post-processing files

d-m-bailey, 2022-03-14:
Something like `runs/<tag>/logs/magic/15-magic_gds.spice.log`

Sukru-Uzun, 2022-03-14:
I could find this one. I uploaded all my log files if you need other files.
[31-gds.spice.log](https://github.com/The-OpenROAD-Project/OpenLane/files/8246920/31-gds.spice.log)

d-m-bailey, 2022-03-14:
@Sukru-Uzun I cloned your repo. I assume you're on branch `sram`, you didn't say.
`31-gds.spice.log` is for `user_proj_example` not `user_project_wrapper`.
I could not find a `user_project_wrapper.gds` file in your repo.
Maybe you don't have these files because LVS fails at the LEF stage.
In your openlane directory, does `find user_project_wrapper -name '*gds*'` find anything?

closed 2022-03-17 by Sukru-Uzun

Prit1097, 2022-03-23:
>Can you also upload the DRC Router Report? I also may reflect errors