The-OpenROAD-Project/OpenLane #998: Pin vdd of instance xxx is not connected to any net

0616ygh, 2022-03-14:
### Description
I use 4 SRAM macros generated by OpenRAM in my CPU design. 
For reasons of SystemVerilog syntax support，I synthesized the RTL files with Candence Genus (using the sky130A process). 
I imported the synthesized netlist into OpenLane and used interactive mode for Floorplan and PnR.
 In the process, the problem as shown in the title appeared：
Pin vdd of instance DATA_RAM is not connected to any net
[image]
When I put 1 macro in a very simple example (like _manual_macro_placement_test_) and use OpenLane for Synthesis&PnR, there is no such problem. Hence I am confused.  Besides, this error appears in the latest version of OpenLane, earlier versions will pass the PDN with warning and proceed to PnR smoothly.
### Environment
```
Kernel: Linux v4.18.0-240.1.1.el8_3.x86_64
Distribution: centos 8
Python: v3.10.2 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: 5f20beb7928c7329ea1a199b9f494f48f2e6c080
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

5f20beb 2022-03-11T01:56:51+02:00 Fix issue with threads option in `run_designs.py` (#994) - Donn -  (HEAD -> master, tag: 2022.03.11_02.27.13, origin/master, origin/HEAD)
a82bf22 2022-03-11T01:47:51+02:00 scripts/checkers: fix report_file_relative scope (#989) - Johan Euphrosine -  ()
ec35dd7 2022-03-11T01:47:12+02:00 scripts/tcl_commands/scrot_klayout: add missing -log arguments (#990) - Johan Euphrosine -  ()
```

### Reproduction Material
* 
[openlane_reproducible_3.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8243299/openlane_reproducible_3.zip)

* Run command:
`./flow.tcl -interactive`
`prep -design orv64`
`*run_floorplan`

### Expected behavior
I would like to know the cause of the error and then fix it.
The flow finishes without errors.

### Logs
```
[INFO TAP-0001] Found 4 macro blocks.
[INFO TAP-0002] Original rows: 1650
[INFO TAP-0003] Created 0 rows for a total of 1650 rows.
[INFO TAP-0004] Inserted 3300 endcaps.
[INFO TAP-0005] Inserted 190614 tapcells.
[INFO ODB-0222] Reading LEF file: /openlane/designs/orv64/runs/RUN_2022.03.11_10.44.43/tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 443 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/designs/orv64/runs/RUN_2022.03.11_10.44.43/tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: /openlane/designs/orv64/runs/RUN_2022.03.11_10.44.43/results/floorplan/orv64.def
[INFO ODB-0128] Design: orv64
[INFO ODB-0094] 		Created 100000 Insts
[INFO ODB-0094] 		Created 200000 Insts
[INFO ODB-0094] 		Created 300000 Insts
[INFO ODB-0097] 		Created 100000 Nets
[INFO ODB-0097] 		Created 200000 Nets
[INFO ODB-0130]     Created 2055 pins.
[INFO ODB-0131]     Created 399285 components and 1879388 component-terminals.
[INFO ODB-0133]     Created 211916 nets and 663484 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/orv64/runs/RUN_2022.03.11_10.44.43/results/floorplan/orv64.def
[INFO PDN-0016] Power Delivery Network Generator: Generating PDN
  config: /openlane/scripts/openroad/pdn_cfg.tcl
[INFO PDN-0008] Design name is orv64.
[INFO PDN-0009] Reading technology data.
[ERROR PDN-0037] Pin vdd of instance ic_top_u_ic_l1_u_PER_WAY\[0\].DATA_RAM is not connected to any net.
PDN-0037
```

maliberty, 2022-03-16:
You have no add_global_connection for pin vdd in pdn_cfg.tcl.  This is identical to https://github.com/The-OpenROAD-Project/OpenROAD/issues/1692

@donn this seems to be a hole in the OL methodology for incorporating macros easily.

0616ygh, 2022-03-17:
> 

Thank you. I will make adjustments according to your suggestions and give feedback as soon as possible

donn, 2022-06-14:
Staled out.

closed 2022-06-14 by donn