The-OpenROAD-Project/OpenLane #999: The parameters suggested in PnR don't seem to work:[ERROR GPL-0302] Use a higher –density or re-floorplan with a larger core area

0616ygh, 2022-03-14:
### Description
OpenLane gives the target-density value it suggests. I followed the suggested parameter to reconfigure the config file and still encountered the same error. I would like to ask if the advice given by the program is not necessarily valid.
[image]

### Environment
```
Kernel: Linux v4.18.0-240.1.1.el8_3.x86_64
Distribution: centos 8
Python: v3.10.2 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: 5f20beb7928c7329ea1a199b9f494f48f2e6c080
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

5f20beb 2022-03-11T01:56:51+02:00 Fix issue with threads option in `run_designs.py` (#994) - Donn -  (HEAD -> master, tag: 2022.03.11_02.27.13, origin/master, origin/HEAD)
a82bf22 2022-03-11T01:47:51+02:00 scripts/checkers: fix report_file_relative scope (#989) - Johan Euphrosine -  ()
ec35dd7 2022-03-11T01:47:12+02:00 scripts/tcl_commands/scrot_klayout: add missing -log arguments (#990) - Johan Euphrosine -  (). 
```

### Reproduction Material
* Upload a tarball containing the relevant design.
[openlane_reproducible_4.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8244849/openlane_reproducible_4.zip)

* List the commands used to run the design.
`./flow.tcl -interactive`
`prep -design orv64`
`run_floorplan`
`run_placement`

### Expected behavior
Know why the advice doesn't work. Solve such problems so that flow can run normally.

### Logs
```
OpenROAD 944855835623e651e7b9c7c50efcce1fb04b4fee 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /openlane/designs/orv64/runs/RUN_2022.03.11_14.02.35/tmp/merged.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 443 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/designs/orv64/runs/RUN_2022.03.11_14.02.35/tmp/merged.lef
[INFO ODB-0127] Reading DEF file: /openlane/designs/orv64/runs/RUN_2022.03.11_14.02.35/tmp/floorplan/1-initial_fp.def
[INFO ODB-0128] Design: orv64
[INFO ODB-0094] 		Created 100000 Insts
[INFO ODB-0094] 		Created 200000 Insts
[INFO ODB-0097] 		Created 100000 Nets
[INFO ODB-0097] 		Created 200000 Nets
[INFO ODB-0130]     Created 2055 pins.
[INFO ODB-0131]     Created 229068 components and 1763016 component-terminals.
[INFO ODB-0133]     Created 241353 nets and 846744 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/orv64/runs/RUN_2022.03.11_14.02.35/tmp/floorplan/1-initial_fp.def
Found 0 macro blocks.
Using 1u default distance from corners.
[INFO PPL-0007] Random pin placement.
[INFO ODB-0222] Reading LEF file: /openlane/designs/orv64/runs/RUN_2022.03.11_14.02.35/tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 443 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/designs/orv64/runs/RUN_2022.03.11_14.02.35/tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: /openlane/designs/orv64/runs/RUN_2022.03.11_14.02.35/tmp/floorplan/2-io.def
[INFO ODB-0128] Design: orv64
[INFO ODB-0094] 		Created 100000 Insts
[INFO ODB-0094] 		Created 200000 Insts
[INFO ODB-0097] 		Created 100000 Nets
[INFO ODB-0097] 		Created 200000 Nets
[INFO ODB-0130]     Created 2055 pins.
[INFO ODB-0131]     Created 229068 components and 1763016 component-terminals.
[INFO ODB-0133]     Created 241353 nets and 846744 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/orv64/runs/RUN_2022.03.11_14.02.35/tmp/floorplan/2-io.def
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: 460 2720
[INFO GPL-0004] CoreAreaLxLy: 5520 10880
[INFO GPL-0005] CoreAreaUxUy: 2656500 2660160
[INFO GPL-0006] NumInstances: 229068
[INFO GPL-0007] NumPlaceInstances: 229068
[INFO GPL-0008] NumFixedInstances: 0
[INFO GPL-0009] NumDummyInstances: 0
[INFO GPL-0010] NumNets: 241332
[INFO GPL-0011] NumPins: 847940
[INFO GPL-0012] DieAreaLxLy: 0 0
[INFO GPL-0013] DieAreaUxUy: 2662330 2673050
[INFO GPL-0014] CoreAreaLxLy: 5520 10880
[INFO GPL-0015] CoreAreaUxUy: 2656500 2660160
[INFO GPL-0016] CoreArea: 7023188294400
[INFO GPL-0017] NonPlaceInstsArea: 0
[INFO GPL-0018] PlaceInstsArea: 2811732924800
[INFO GPL-0019] Util(%): 40.03
[INFO GPL-0020] StdInstsArea: 2811732924800
[INFO GPL-0021] MacroInstsArea: 0
[InitialPlace]  Iter: 1 CG Error: 0.00022569 HPWL: 6756712652
[InitialPlace]  Iter: 2 CG Error: 0.00007202 HPWL: 5830882928
[InitialPlace]  Iter: 3 CG Error: 0.00003200 HPWL: 5842351930
[InitialPlace]  Iter: 4 CG Error: 0.00010106 HPWL: 5808646974
[InitialPlace]  Iter: 5 CG Error: 0.00007618 HPWL: 5684175691
[InitialPlace]  Iter: 6 CG Error: 0.00009629 HPWL: 5533127588
[InitialPlace]  Iter: 7 CG Error: 0.00002845 HPWL: 5377541256
[InitialPlace]  Iter: 8 CG Error: 0.00001709 HPWL: 5286123883
[InitialPlace]  Iter: 9 CG Error: 0.00001476 HPWL: 5230608024
[InitialPlace]  Iter: 10 CG Error: 0.00000734 HPWL: 5214300892
[ERROR GPL-0302] Use a higher -density or re-floorplan with a larger core area.
Given target density: 0.40
Suggested target density: 0.40
Error: replace.tcl, 90 GPL-0302
```

vijayank88, 2022-03-15:
@0616ygh 
Always use `suggested target density+ 0.01` will works.(greater than suggested value)
Here 0.41 will resolve your issue.

maliberty, 2022-03-15:
@vijayank88 if that is the case we should just change to the tool reported value.

0616ygh, 2022-03-16:
> @vijayank88 resolved with [The-OpenROAD-Project/OpenROAD#1697](https://github.com/The-OpenROAD-Project/OpenROAD/pull/1697)
Ok, thank you very much!

0616ygh, 2022-03-16:
> 

Your suggestion is effective. Thank you for your answer.