The-OpenROAD-Project/OpenLane #1005: Routing congestion fault

Sukru-Uzun, 2022-03-15:
I have a congestion fault. I have tried different size of area but it didn't change the result. My density is around 0.1.
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8251830/openroad_issue_reproducible.tar.gz)

vijayank88, 2022-03-15:
@Sukru-Uzun 
Please follow issue template its must.
OS/OpenLane version..etc.,

Sukru-Uzun, 2022-03-15:
```
Kernel: Linux v5.11.0-49-generic
Distribution: ubuntu 21.04
Python: v3.9.5 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: -dev
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

e4bfdd7 2022-02-22T16:59:14-03:00 Remove pip install, no longer needed (#953) - Vitor Bandeira -  (grafted, HEAD, tag: 2022.02.23_02.50.41)
```

vijayank88, 2022-03-15:
@Sukru-Uzun 
share user_proj_example/runs/<tag>/config.tcl

vijayank88, 2022-03-15:
@Sukru-Uzun 
Is rtl and openlane directory DB is up-to-date right?

Sukru-Uzun, 2022-03-15:
@vijayank88, yes

vijayank88, 2022-03-15:
@Sukru-Uzun keep the repo up-to-date. So its easy for the experts to debug the issue and provide the solution ASAP.

vijayank88, 2022-03-16:
@Sukru-Uzun Close with comment how you resolved the issue. It will useful for other community users. thanks