The-OpenROAD-Project/OpenLane #1010: [ERROR DPL-0036] Detailed placement failed.

Hagiwara-shc, 2022-03-18:
### Description
I got the following error and process exited abnormally during implementing my design.

[ERROR DPL-0036] Detailed placement failed.
Error: resizer.tcl, 78 DPL-0036
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

### Environment
OpenLane environment has been installed as part of
[Caravel User Project - mpw-5c release](https://github.com/efabless/caravel_user_project/tree/mpw-5c)

"python3 ./env.py issue-survey" failed as follows...
Traceback (most recent call last):
  File "./env.py", line 255, in <module>
    main()
  File "./env.py", line 251, in main
    commands[args[0]]()
  File "./env.py", line 154, in issue_survey
    % get_tag()
  File "/home/shc/Research/efabless/openlane/dependencies/get_tag.py", line 66, in get_tag
    raise e
  File "/home/shc/Research/efabless/openlane/dependencies/get_tag.py", line 48, in get_tag
    f"Cannot determine git branch. Please specify OPENLANE_IMAGE_NAME manually.\nFull output: {branch_name_data.stderr.decode('utf8').strip()}"
dependencies.get_tag.NoGitException: Cannot determine git branch. Please specify OPENLANE_IMAGE_NAME manually.
Full output: error: unknown option `show-current'
usage: git branch [<options>] [-r | -a] [--merged | --no-merged]
   or: git branch [<options>] [-l] [-f] <branch-name> [<start-point>]
   or: git branch [<options>] [-r] (-d | -D) <branch-name>...
   or: git branch [<options>] (-m | -M) [<old-branch>] <new-branch>
   or: git branch [<options>] (-c | -C) [<old-branch>] <new-branch>
   or: git branch [<options>] [-r | -a] [--points-at]
   or: git branch [<options>] [-r | -a] [--format]

### Reproduction Material
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8297599/openroad_issue_reproducible.tar.gz)

* git clone https://github.com/Hagiwara-shc/marmot_asic.git
* Install OpenLane and PDK as described at https://github.com/efabless/caravel_user_project/blob/mpw-5c/docs/source/quickstart.rst
* make marmot

### Expected behavior
Flow finish without error.

### Logs
...
set_max_fanout 5.0000 [current_design]
[INFO]: Setting RC values...
[INFO RSZ-0027] Inserted 637 input buffers.
[INFO RSZ-0028] Inserted 296 output buffers.
[INFO RSZ-0058] Using max wire length 2319um.
[INFO RSZ-0034] Found 4 slew violations.
[INFO RSZ-0036] Found 2 capacitance violations.
[INFO RSZ-0038] Inserted 14 buffers in 4 nets.
[INFO RSZ-0039] Resized 92648 instances.
[INFO RSZ-0042] Inserted 196 tie sky130_fd_sc_hd__conb_1 instances.
[INFO RSZ-0042] Inserted 14 tie sky130_fd_sc_hd__conb_1 instances.
[INFO DPL-0034] Detailed placement failed on:
[INFO DPL-0035]  _138361_
[INFO DPL-0035]  _138866_
  ... (snip, about 1000 lines)
[INFO DPL-0035]  _115609_
[INFO DPL-0035]  _115632_
[INFO DPL-0035] message limit reached, this message will no longer print
[ERROR DPL-0036] Detailed placement failed.
Error: resizer.tcl, 78 DPL-0036
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

vijayank88, 2022-03-18:
@Hagiwara-shc 
Regarding `"python3 ./env.py issue-survey" failed as follows...` this issue,

Try to use python3.6 and above version

vijayank88, 2022-03-18:
@Hagiwara-shc 
`set ::env(FP_CORE_UTIL) 50` enable this in config.tcl and try lower value to utilize more core for placement, cell placement congestion high. Try `set ::env(FP_CORE_UTIL) 30`
[image]

vijayank88, 2022-03-18:
@Hagiwara-shc 
With following updates in `config.tcl` am able to complete the flow successfully.
```
set ::env(FP_CORE_UTIL) 30
set ::env(CELL_PAD) 2
set ::env(PL_BASIC_PLACEMENT) 0
```

Hagiwara-shc, 2022-03-18:
@vijayank88
Thank you for your comment.

After I updated config.tcl as you suggested, [ERROR DPL-0036]  disappeared.
However, the flow stopped due to another problem.
Reproducible packaged and the log is as follows:

[openroad_issue_reproducible_2.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8308035/openroad_issue_reproducible_2.tar.gz)

[INFO DRT-0181] Start track assignment.
[INFO DRT-0184] Done with 484755 vertical wires in 9 frboxes and 283467 horizontal wires in 10 frboxes.
[INFO DRT-0186] Done with 60026 vertical wires in 9 frboxes and 75809 horizontal wires in 10 frboxes.
[INFO DRT-0182] Complete track assignment.
[INFO DRT-0267] cpu time = 00:05:10, elapsed time = 00:11:47, memory = 3979.47 (MB), peak = 4457.57 (MB)
[INFO DRT-0187] Start routing data preparation.
[INFO DRT-0267] cpu time = 00:00:01, elapsed time = 00:00:03, memory = 4002.71 (MB), peak = 4457.57 (MB)
[INFO DRT-0194] Start detail routing.
[INFO DRT-0195] Start 0th optimization iteration.
    Completing 10% with 0 violations.
    elapsed time = 00:04:12, memory = 4361.95 (MB).
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: kill signal

Do you have any clue to fix the problem?
Should I file as another issue?

vijayank88, 2022-03-19:
How much RAM does your machine have?

Hagiwara-shc, 2022-03-19:
RAM is 4.8GB, and I added 19GB swap file.

$ free -h
              total        used        free      shared  buff/cache   available
Mem:           4.8G        735M        1.1G         61M        3.0G        3.8G
Swap:           19G        953M         19G

vijayank88, 2022-03-19:
Increase your RAM size it will work. I will confirm later how much RAM
exactly you need to complete the routing. Also as per current configuration
flow failed with hold violations

Hagiwara-shc, 2022-03-19:
Ok, I will increase RAM size and try again.
Thank you.

vijayank88, 2022-03-19:
@Hagiwara-shc 
Peak memory took to complete routing
```
[INFO DRT-0267] cpu time = 02:14:29, elapsed time = 01:10:16, memory = 32422.46 (MB), peak = 35505.38 (MB)
```
Closed to 35GB need

Hagiwara-shc, 2022-03-19:
@vijayank88 
My trial with larger RAM (48GB) has just completed! (with hold violations...). The log indicates 35GB peak too.

Thank you for your support.

closed 2022-03-19 by Hagiwara-shc

tanvi-shewale, 2023-04-12:
Where can I find the config.tcl? I am facing a similar issue.