The-OpenROAD-Project/OpenLane #1012: LVS error net mismatch

gncemre23, 2022-03-18:
### Description
While hardening the core, which has SRAM macro and another digital which is hardened separately without any error, I get the LVS error given below.
[image]

### Environment
```
Kernel: Linux v5.11.0-49-generic
Distribution: ubuntu 21.04
Python: v3.9.5 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: -dev
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
```

### Reproduction Material
in the **[repository](https://github.com/Procenne-Digital-Design/secure-memory/tree/uart)**, I run the following command 
`flow.tcl -design user_project_wrapper -tag with_user_proj_0.5 -overwrite -disable_output`

### Expected behavior
No lvs error
### Logs
```
Please find the attached log file.
[29-user_project_wrapper.lef.log](https://github.com/The-OpenROAD-Project/OpenLane/files/8302756/29-user_project_wrapper.lef.log)
```

d-m-bailey, 2022-03-18:
@gncemre23 Looks like you have a power short. Can you post your gds?

gncemre23, 2022-03-18:
Sure! Please find the attached
[user_project_wrapper.gds.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8306732/user_project_wrapper.gds.zip)
d

d-m-bailey, 2022-03-18:
@gncemre23 Looks like your `user_proj_example` has met5 routing layers that are shorting with the `user_project_wrapper` power grid. Should be easy to see if you just display met5. Maybe set your max routing layer to met4.
[image]

I don't believe this is an openroad issue and suggest the issue be closed.

armleo, 2022-03-19:
Also you have typo here:
set ::env(VDD_NET) {vcdd1}
set ::env(GND_NET) {vssd1}

Sukru-Uzun, 2022-03-19:
> @gncemre23 Looks like your `user_proj_example` has met5 routing layers that are shorting with the `user_project_wrapper` power grid. Should be easy to see if you just display met5. Maybe set your [...]

Is this klayout?

d-m-bailey, 2022-03-19:
Yes, klayout.

Sukru-Uzun, 2022-03-19:
Can we observe that on the Openroad -gui ? I cannot import gds or lef files in openroad.

gncemre23, 2022-03-19:
I realized that I did not set DESIG_IS_CORE parameter as 0 while hardening the user_proj_example macro. I guess that causes the tool is using met5 even if I set rt_max_layer as met4. Now, with this adjustment, I don't have any LVS errors.
Thank you @d-m-bailey

closed 2022-03-19 by gncemre23