The-OpenROAD-Project/OpenLane #1013: Tool hangs during report_power calculation

vijayank88, 2022-03-18:
### Description
This is from slack channel #openlane https://skywater-pdk.slack.com/archives/C016H8WJMBR/p1647550327496089 conversation.
Raising the issue behalf of user.

Hello, has anyone tried synthesizing an asynchronous design using openlane? In my case, openlane hangs at report_power. If I disable the power report( by commenting it out in sta.tcl ) it then hangs when generating the PDN.

### Environment
```
mpw-5c
```

### Reproduction Material
* Upload a tarball containing the relevant design. 
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8303290/openroad_issue_reproducible.zip)

* List the commands used to run the design.
* unzip the attached file and run #openroad > source run.tcl

### Expected behavior
Should not hang at PDN stage

### Logs
```
[INFO ODB-0222] Reading LEF file: ./tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  ./tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: ./in.def
[INFO ODB-0128] Design: user_proj_example
[INFO ODB-0130]     Created 607 pins.
[INFO ODB-0131]     Created 13326 components and 56322 component-terminals.
[INFO ODB-0133]     Created 5993 nets and 17459 connections.
[INFO ODB-0134] Finished DEF file: ./in.def
[INFO PDN-0016] Power Delivery Network Generator: Generating PDN
  config: openlane/scripts/openroad/pdn_cfg.tcl
[INFO PDN-0008] Design name is user_proj_example.
[INFO PDN-0009] Reading technology data.
[INFO PDN-0011] ****** INFO ******
Type: stdcell, stdcell_grid
    Stdcell Rails
      Layer: met1  -  width: 0.480  pitch: 5.440 
    Straps
      Layer: met4  -  width: 1.600  pitch: 153.600  offset: 16.320 
    Connect: {met1 met4}
Type: macro, CORE_macro_grid_1
    Macro orientation: R0 R180 MX MY R90 R270 MXR90 MYR90
    Straps
    Connect: {met4_PIN_ver met5}
[INFO PDN-0012] **** END INFO ****
[INFO PDN-0013] Inserting stdcell grid - stdcell_grid.
[INFO PDN-0015] Writing to database.
[INFO]: Setting RC values...
[INFO PSM-0002] Output voltage file is specified as: ./tmp/floorplan/6-pdn.pga.rpt.
[WARNING PSM-0016] Voltage pad location (VSRC) file not specified, defaulting pad location to checkerboard pattern on core area.
[WARNING PSM-0017] X direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0018] Y direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0019] Voltage on net vccd1 is not explicitly set.
[WARNING PSM-0022] Using voltage 1.800V for VDD network.
[WARNING PSM-0065] VSRC location not specified, using default checkerboard pattern with one VDD every size bumps in x-direction and one in two bumps in the y-direction
[WARNING PSM-0030] VSRC location at (5.520um, 10.880um) and size 10.000um, is not located on a power stripe. Moving to closest stripe at (21.840um, 10.800um).
[WARNING PSM-0030] VSRC location at (285.520um, 150.880um) and size 10.000um, is not located on a power stripe. Moving to closest stripe at (329.040um, 151.200um).
[WARNING PSM-0030] VSRC location at (565.520um, 290.880um) and size 10.000um, is not located on a power stripe. Moving to closest stripe at (636.240um, 291.600um).
[WARNING PSM-0030] VSRC location at (5.520um, 430.880um) and size 10.000um, is not located on a power stripe. Moving to closest stripe at (21.840um, 432.000um).
[INFO PSM-0031] Number of PDN nodes on net vccd1 = 21954.

```

vijayank88, 2022-03-18:
By disable IR drop calculation able to complete PDN stage.
```
set ::env(FP_PDN_IRDROP) 0
```

6Kotnk, 2022-03-18:
Thank you, this appears to have fixed my problem. However, I've had to comment out the code in pdn.tcl. Setting the environment variable in config.tcl does not seem to do anything.

vijayank88, 2022-03-19:
@donn 
This is the solution to go forward or will be fixed in tool?

maliberty, 2022-03-19:
@vijayank88 this is a workaround and the tool should still be fixed.  In such cases be sure to let the OR team know of the problem.

VidyaChhabria, 2022-03-20:
@maliberty  IR drop analysis inherently calls the subroutines of report_power to get the current values. So if report_power hangs, IR drop is hung too.

maliberty, 2022-03-22:
@donn @shalan although the tool shouldn't hang it makes no sense to run analyze_power_grid before placement.  You have no idea where the power draw will be on the grid.  In ORFS we run it as part of the final report.

jjcherry56, 2022-03-23:
OR commit ba46a7a7c fixes this

dralabeing, 2022-03-30:
@mousa do you want to validate this fix?

mousaq92, 2022-03-30:
How can I get OR commit ba46a7a7c?

Thanks,

Mousa

mousaq92, 2022-03-31:
Hello,

I re-installed OR. I can confirm that pdn passes. I see a lot of warnings. Here's a screenshot for your reference.

[image]

Thanks,

Mousa

maliberty, 2022-03-31:
running ir drop without detailed placement is going to be virtually meaningless.

closed 2022-06-14 by donn