The-OpenROAD-Project/OpenLane #1016: Error during PDN

ratul619, 2022-03-21:
### Description

Error in PDN step : 

Floorplan seems OK with spacing in between macros 

### Environment
```
Please run the following command in the OpenLane folder:
Kernel: Linux v5.13.0-35-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: UNKNOWN vUNKNOWN (UNSUPPORTED)
OpenLane Git Version: 9602463a6dae6cfe4ad025251d91a8310b417bfd
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

9602463 2022-03-03T00:07:19+02:00 Add Commit-Based Docker Tags (#967) - Donn -  (HEAD -> master, tag: 2022.03.03_02.36.44, origin/master, origin/HEAD)
b2c49ef 2022-03-02T21:39:51+02:00 [BOT] Update openroad_app (#961) - Openlane Bot -  ()
4076f3e 2022-03-02T01:24:37+02:00 Synthesis Exploration Tweaks (#973) - Donn -  (tag: 2022.03.02_02.59.05)

```

### Reproduction Material
* Upload a tarball containing the relevant design.
* List the commands used to run the design.

If you see a message like `Reproducible packaged: Please tarball and upload <PATH> if you're going to submit an issue` in your logs, please also tarball and include that path. This will greatly speed up the fixing process.

### Expected behavior
A clear and concise description of what you expected to happen.

### Logs
```

[WARNING PSM-0038] Unconnected PDN node on net gnd at location (4012.200um, 4889.725um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net gnd at location (3979.800um, 4889.725um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net gnd at location (3952.800um, 4889.725um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net gnd at location (3920.400um, 4889.725um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net gnd at location (3888.000um, 4889.725um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net gnd at location (3893.400um, 4889.725um), layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net gnd at location (3861.000um, 4889.725um), layer: 5.
[ERROR PSM-0069] Check connectivity failed.
Error: pdn.tcl, 44 PSM-0069
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[INFO]: Saving runtime environment...
[INFO]: Creating reproducible...

or_issue.py OpenROAD Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /openlane/designs/vector_processor_v1/runs/RUN_2022.03.20_15.22.15/openroad_issue_reproducible…
[FIN] Done.
[INFO]: Reproducible packaged: Please tarball and upload designs/vector_processor_v1/runs/RUN_2022.03.20_15.22.15/openroad_issue_reproducible if you're going to submit an issue.
[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: top
Run Directory: /openlane/designs/vector_processor_v1/runs/RUN_2022.03.20_15.22.15
Source not found.
----------------------------------------

LVS Summary:
Source: /openlane/designs/vector_processor_v1/runs/RUN_2022.03.20_15.22.15/logs/finishing/top.lvs.lef.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openlane/designs/vector_processor_v1/runs/RUN_2022.03.20_15.22.15/reports/final_summary_report.csv
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
[INFO]: The failure may have been because of the following warnings:
[WARNING]: All internal macros will not be connected to power.

    while executing
"flow_fail"
    (procedure "run_openroad_script" line 65)
    invoked from within
"run_openroad_script $::env(SCRIPTS_DIR)/openroad/pdn.tcl  |& -indexed_log [index_file $::env(floorplan_logs)/pdn.log]"
    (procedure "gen_pdn" line 9)
    invoked from within
"gen_pdn"
    (procedure "run_power_grid_generation" line 94)
    invoked from within
"run_power_grid_generation"
    (procedure "run_floorplan" line 55)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 55)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	puts_info "Note, that post_run_hook..."
    (file "./flow.tcl" line 412)

```

ratul619, 2022-03-21:
Attaching collaterals

ratul619, 2022-03-21:
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8313207/openroad_issue_reproducible.tar.gz)
[sram_64_128_sky130A_SS_1p8V_25C.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8313211/sram_64_128_sky130A_SS_1p8V_25C.zip)
[sram_64_128_sky130A.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8313213/sram_64_128_sky130A.zip)

vijayank88, 2022-03-21:
@ratul619 
Set following variable in `config.tcl` and re-try
`set ::env(FP_PDN_CHECK_NODES) 0`

ratul619, 2022-03-21:
Thanks. Went ahead with flow , but hit below issue : 

Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
[INFO GRT-0012] Antenna violations: 127
[WARNING GRT-0054] Placement of diode ANTENNA_158 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_214 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_215 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_216 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_217 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_236 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_237 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_238 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_239 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_240 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_241 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_242 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_338 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_339 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_340 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_341 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_342 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_343 will be legalized by detailed placement.
[INFO GRT-0015] 399 diodes inserted.
[WARNING DPL-0004] Placed in rows check failed (9).
[ERROR DPL-0033] detailed placement checks failed.
Error: groute.tcl, 75 DPL-0033
[ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[INFO]: Saving runtime environment...
[INFO]: Creating reproducible...

or_issue.py OpenROAD Issue Packager

ratul619, 2022-03-21:
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8313680/openroad_issue_reproducible.tar.gz)

vijayank88, 2022-03-21:
@ratul619  can you share your `config.tcl`?

Check your `config.tcl` for variable `CELL_PAD` and share its value.

ratul619, 2022-03-21:
set ::env(CELL_PAD) {4}
set ::env(CELL_PAD_EXCLUDE) {sky130_fd_sc_hd__tap* sky130_fd_sc_hd__decap* sky130_fd_sc_hd__fill*}
set ::env(CELL_PAD) {4}
set ::env(CELL_PAD_EXCLUDE) {sky130_fd_sc_hd__tap* sky130_fd_sc_hd__decap* sky130_fd_sc_hd__fill*}
set ::env(CELL_PAD) {4}
set ::env(CELL_PAD_EXCLUDE) {sky130_fd_sc_hd__tap* sky130_fd_sc_hd__decap* sky130_fd_sc_hd__fill*}

ratul619, 2022-03-21:
[config.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8313734/config.zip)

vijayank88, 2022-03-21:
@ratul619 
Can you re-try the flow with value of `CELL_PAD` with `2`?

ratul619, 2022-03-21:
Still fails

Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
[INFO GRT-0012] Antenna violations: 116
[WARNING GRT-0054] Placement of diode ANTENNA_188 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_189 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_190 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_191 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_210 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_211 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_212 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_213 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_214 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_215 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_216 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_312 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_313 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_314 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_315 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_316 will be legalized by detailed placement.
[WARNING GRT-0054] Placement of diode ANTENNA_317 will be legalized by detailed placement.
[INFO GRT-0015] 379 diodes inserted.
[WARNING DPL-0004] Placed in rows check failed (10).
[ERROR DPL-0033] detailed placement checks failed.
Error: groute.tcl, 75 DPL-0033
[ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

ratul619, 2022-03-21:
Any other suggestions

vijayank88, 2022-03-21:
@ratul619 
There are two options here to fix it.
1. set ::env(DIODE_PADDING) 0 
2. Change the diode insertion strategy other than 3 will resolve this issue. `set ::env(DIODE_INSERTION_STRATEGY) 4`
Any one of above will woks and passing global route

ratul619, 2022-03-21:
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8314491/openroad_issue_reproducible.tar.gz)

vijayank88, 2022-03-21:
@ratul619 
child killed signal probably a RAM issue. Check your  detailed router log for how much PEAK memory its took? 
What is your RAM size available?

ratul619, 2022-03-21:
#macroNoAp             = 0
[INFO DRT-0166] Complete pin access.
[INFO DRT-0267] cpu time = 00:00:25, elapsed time = 00:00:15, memory = 5851.34 (MB), peak = 6812.30 (MB)
[INFO DRT-0151] Reading guide.

The peak doesnt seem much. Running on the 24GB RAM machine

ratul619, 2022-03-21:
Checking on : logs/routing/24-detailed.log

vijayank88, 2022-03-21:
@ratul619 
To start detail routing itself is need close to 10GB. see my logs here
```
[INFO DRT-0185] Post process initialize RPin region query.
[INFO DRT-0181] Start track assignment.
[INFO DRT-0184] Done with 8569 vertical wires in 15 frboxes and 5386 horizontal wires in 15 frboxes.
[INFO DRT-0186] Done with 937 vertical wires in 15 frboxes and 688 horizontal wires in 15 frboxes.
[INFO DRT-0182] Complete track assignment.
[INFO DRT-0267] cpu time = 00:02:48, elapsed time = 00:01:31, memory = 8139.52 (MB), peak = 9680.57 (MB)
[INFO DRT-0187] Start routing data preparation.
[INFO DRT-0267] cpu time = 00:00:00, elapsed time = 00:00:00, memory = 8139.52 (MB), peak = 9680.57 (MB)
[INFO DRT-0194] Start detail routing.
```
It may need more memory to run further. I'll let you know once my routing completes how much memory its took for routing.

vijayank88, 2022-03-21:
I'm facing some new issues.
```
Error: net1109 1 pin not visited #guides = 7
Error: net1110 1 pin not visited #guides = 8
Error: net1111 1 pin not visited #guides = 7
Error: net1112 1 pin not visited #guides = 7
Error: net1113 1 pin not visited #guides = 7
Error: net1114 1 pin not visited #guides = 7
Error: net1115 1 pin not visited #guides = 7
Error: net1116 1 pin not visited #guides = 7
Error: checkConnectivity break, net net1053
Objs not visited:
frPathSeg: begin (1326640 4411500 ) end ( 1326640 4414900 ) layerNum 10
	beginStyle: 1
	endStyle: 0
frPathSeg: begin (1326640 4411500 ) end ( 1327100 4411500 ) layerNum 10
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (1327100 4408780 ) end ( 1327100 4411500 ) layerNum 10
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (1326870 4408780 ) end ( 1327100 4408780 ) layerNum 8
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (1326870 4402490 ) end ( 1326870 4408780 ) layerNum 6
	beginStyle: 1
	endStyle: 1
frVia: at ( 1327100 4408780 )
VIA DEF:
VIA M3M4_PR DEFAULT
  RECT -190 -160 190 160
  RECT -100 -100 100 100
  RECT -165 -165 165 165
frVia: at ( 1326870 4408780 )
VIA DEF:
VIA M2M3_PR DEFAULT
  RECT -140 -185 140 185
  RECT -100 -100 100 100
  RECT -165 -165 165 165
frVia: at ( 1326870 4402490 )
VIA DEF:
VIA L1M1_PR_MR DEFAULT
  RECT -85 -85 85 85
  RECT -85 -85 85 85
  RECT -145 -115 145 115
frVia: at ( 1326870 4402490 )
VIA DEF:
VIA M1M2_PR DEFAULT
  RECT -160 -130 160 130
  RECT -75 -75 75 75
  RECT -130 -160 130 160
INSTTERM: (INST/CELL/TERM/NET) _2790__1053 sky130_fd_sc_hd__conb_1 LO net1053
Segmentation fault
```
Have you got any other issues during global routing?

vijayank88, 2022-03-21:
@maliberty 
Please have a look into this issue.

Is that related to macro halo spacing, has router not able to access the pins?

vijayank88, 2022-03-21:
@ratul619  
inside `runs` directory there is another `config.tcl` can you share that again?

mousaq92, 2022-03-21:
Hello,

I just saw this issue, and I want to join in on this if that's ok.

I downloaded the first reproducible that you posted.

In run.tcl, I changed set :: env(FP_PDN_ENABLE_MACROS_GRID) to 1

PDN now passes. Please let me know if this works.

Thanks,

Mousa

ratul619, 2022-03-21:
I increased the RAM , still fails here. 

[INFO DRT-0036] met1 guide region query size = 4151.
[INFO DRT-0036] via guide region query size = 0.
[INFO DRT-0036] met2 guide region query size = 2740.
[INFO DRT-0036] via2 guide region query size = 0.
[INFO DRT-0036] met3 guide region query size = 1229.
[INFO DRT-0036] via3 guide region query size = 0.
[INFO DRT-0036] met4 guide region query size = 1077.
[INFO DRT-0036] via4 guide region query size = 0.
[INFO DRT-0036] met5 guide region query size = 6.
[INFO DRT-0179] Init gr pin query.
[INFO DRT-0185] Post process initialize RPin region query.
[INFO DRT-0181] Start track assignment.
[INFO DRT-0184] Done with 8569 vertical wires in 15 frboxes and 5386 horizontal wires in 15 frboxes.
[INFO DRT-0186] Done with 937 vertical wires in 15 frboxes and 688 horizontal wires in 15 frboxes.
[INFO DRT-0182] Complete track assignment.
[INFO DRT-0267] cpu time = 00:02:49, elapsed time = 00:01:31, memory = 8152.80 (MB), peak = 9693.86 (MB)
[INFO DRT-0187] Start routing data preparation.
[INFO DRT-0267] cpu time = 00:00:00, elapsed time = 00:00:00, memory = 8152.80 (MB), peak = 9693.86 (MB)
[INFO DRT-0194] Start detail routing.
[INFO DRT-0195] Start 0th optimization iteration.
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: kill signal

[INFO]: Saving runtime environment...
[INFO]: Creating reproducible...

ratul619, 2022-03-21:
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8316689/openroad_issue_reproducible.tar.gz)

ratul619, 2022-03-21:
@vijayank88  -- Did not face any issues in groute step

vijayank88, 2022-03-21:
@ratul619
It's took close to 22GB in 1st optimization itself

vijayank88, 2022-03-22:
@ratul619 
Can you share your DB/ repo?

vijayank88, 2022-03-22:
@ratul619 
As your design have 16 Macros count, its need better floorplaning with routing obstruction.
I see there are  gnd and pwr rails goes over the macro. It should not happen.

Need to find where issue coming from and your design configuration too confusing.

ratul619, 2022-03-24:
https://github.com/ratul619/vector_processor_v1 

Above is the github link for full design folder...

maliberty, 2022-03-24:
@Stephanommg since you are looking at the connectivity checker please look into this one.

ratul619, 2022-03-28:
HI , Any update on this ? 

Thanks

mousaq92, 2022-03-28:
Hello @ratul619 ,

The design that you provided uses the following Verilog file for the sram blocks sram_1rw1r_32_256_8_sky130.v.

The lef and lib files that you provided are for a completely different sram block

sram_64_128_sky130A.lef
sram_64_128_sky130A_SS_1p8V_25C.lib

Can you please upload the right verilog, lef, lib, and gds files for the sram blocks?

Thanks,

Mousa

Stephanommg, 2022-03-28:
I am using this link to reproduce the error: https://github.com/The-OpenROAD-Project/OpenLane/files/8316689/openroad_issue_reproducible.tar.gz

Is it still up to date?

ratul619, 2022-03-29:
Yes . its upto date. No changes have been made

antonblanchard, 2022-03-30:
Possibly a reproduce of https://github.com/The-OpenROAD-Project/OpenROAD/issues/1677

Stephanommg, 2022-03-30:
The problem is solved. The solution is currently in this branch: https://github.com/The-OpenROAD-Project-private/OpenROAD/tree/secure-TR_APsWithShapes

However it is still under more testing.

ratul619, 2022-03-31:
Anyway i can take the solution at this point ?

maliberty, 2022-03-31:
You can always build OR from source however that branch does have some issues found in testing so caveat emptor

Stephanommg, 2022-04-12:
Please test again with the master branch. It should be solved.

closed 2022-04-27 by Stephanommg