The-OpenROAD-Project/OpenLane #1028: [mpw-5] groute.tcl exits abnormally on analog GDS integration, possibly congestion error?

3x10e8, 2022-03-31:
### Description
Global router errors out when attempting to make user_project_wrapper for a custom analog GDS layout of ReRAM cells along with a stacked MiM cap array using M3-M5 layers in sky130B. We have tried changing parameter settings on [config.tcl](https://github.com/jainsoumil2/ReRAM-16x16/blob/design/openlane/user_project_wrapper/config.tcl) but we are still unable to get the flow to complete, and the error output is somewhat cryptic. Any pointers to debug analog GDS integration with the wrapper would be appreciated!

### Environment
```
Kernel: Linux v5.13.0-37-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.13 (OK)
OpenLane Git Version: -dev
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

e4bfdd7 2022-02-22T16:59:14-03:00 Remove pip install, no longer needed (#953) - Vitor Bandeira -  (grafted, HEAD, tag: 2022.02.23_02.50.41)
```

### Reproduction Material
* Project repo: https://github.com/jainsoumil2/ReRAM-16x16
* Verilog files: [user_project_wrapper.v](https://github.com/jainsoumil2/ReRAM-16x16/blob/design/verilog/rtl/user_project_wrapper.v), for wrapping blackboxed [top.v](https://github.com/jainsoumil2/ReRAM-16x16/blob/design/verilog/rtl/top.v) 
* Commands used: 
```
sudo OPENLANE_ROOT=~/OpenLane PDK_ROOT=~/pdks make user_project_wrapper
```
* Reproducible packaged from [openroad_issue_reproducible](https://github.com/jainsoumil2/ReRAM-16x16/tree/design/openlane/user_project_wrapper/runs/user_project_wrapper/openroad_issue_reproducible): [openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8385271/openroad_issue_reproducible.zip)

### Expected behavior
We were expecting the GDS flow to complete -- but we do not understand what is causing the global router to error out. 

### Logs
```
[INFO]: Obstructions added over met5 1200 1200 1800 1800,
		met4 1200 1200 1800 1800
[INFO]: Changing layout from /home/soumil/Desktop/ReRAM-16x16/openlane/user_project_wrapper/runs/user_project_wrapper/results/placement/user_project_wrapper.def to /home/soumil/Desktop/ReRAM-16x16/openlane/user_project_wrapper/runs/user_project_wrapper/results/placement/user_project_wrapper.obs.def
[INFO]: Incremented step index to 13.
[INFO]: Running Detailed Placement...
[INFO]: Changing layout from /home/soumil/Desktop/ReRAM-16x16/openlane/user_project_wrapper/runs/user_project_wrapper/results/placement/user_project_wrapper.obs.def to /home/soumil/Desktop/ReRAM-16x16/openlane/user_project_wrapper/runs/user_project_wrapper/results/placement/user_project_wrapper.obs.def
[INFO]: Changing layout from /home/soumil/Desktop/ReRAM-16x16/openlane/user_project_wrapper/runs/user_project_wrapper/results/placement/user_project_wrapper.obs.def to /home/soumil/Desktop/ReRAM-16x16/openlane/user_project_wrapper/runs/user_project_wrapper/results/placement/user_project_wrapper.obs.def
[INFO]: Incremented step index to 14.
[INFO]: Running Fill Insertion...
[INFO]: Changing layout from /home/soumil/Desktop/ReRAM-16x16/openlane/user_project_wrapper/runs/user_project_wrapper/results/placement/user_project_wrapper.obs.def to /home/soumil/Desktop/ReRAM-16x16/openlane/user_project_wrapper/runs/user_project_wrapper/tmp/routing/15-fill.def
[INFO]: Incremented step index to 15.
[INFO]: Running Global Routing...
[ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally
```

maliberty, 2022-03-31:
I think you'll have better success if you bring the pins out to the edge of your macro rather trying to get the router to see all the gds level details and find a connection.

vijayank88, 2022-03-31:
@3x10e8 
Your design looks highly congested
[image]

Check zoomed view
[image]

maliberty, 2022-03-31:
@vijayank88 those are just fill cells.  Looking at the routing congestion is more useful for global routing issues.  The macro is the area of high congestion

vijayank88, 2022-04-01:
@3x10e8 
Can you explain how you defined this
```
set ::env(GLB_RT_OBS) "met5 1200 1200 1800 1800,
		met4 1200 1200 1800 1800"
```
https://github.com/jainsoumil2/ReRAM-16x16/blob/bb4e2123fda891e71311465667a7af3908684db8/openlane/user_project_wrapper/config.tcl#L76

3x10e8, 2022-04-01:
@vijayank88 thank you for catching that! The design top_module has an array of MiM caps using M3, M4, and M5 layers, and our goal here was to obstruct out M4/M5 over the MiM caps, but those values are all wrong, sorry!

I have gone back to the macro placement coordinates used by the mprj counter example included with caravel_user_project: [macro.cfg](https://github.com/jainsoumil2/ReRAM-16x16/blob/droute_error/openlane/user_project_wrapper/macro.cfg)
```
top_module 1175 1690 N
```
And I have corrected the M4/M5 obstructions to approximately avoid the MiM cap array ([config.tcl](https://github.com/jainsoumil2/ReRAM-16x16/blob/droute_error/openlane/user_project_wrapper/config.tcl)):
```
set ::env(GLB_RT_OBS)  "met5 1175 1813 1418 2053,
			met4 1175 1813 1418 2053"
```
This should now look as below, where grey is the user_project_wrapper, blue is the top_module, and the yellow ring encloses the M4/M5 obstruction area:
[image]

This by itself doesn't overcome routing errors ([openroad_issue_reproducible](https://github.com/jainsoumil2/ReRAM-16x16/tree/droute_error/openlane/user_project_wrapper/runs/user_project_wrapper/openroad_issue_reproducible) committed to repo):
```
[INFO]: Incremented step index to 17.
[INFO]: Running Detailed Routing...
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: kill signal
```
And using OpenROAD GUI with: 
read_lef [merged_unpadded.lef](https://github.com/jainsoumil2/ReRAM-16x16/blob/droute_error/openlane/user_project_wrapper/runs/user_project_wrapper/openroad_issue_reproducible/tmp/merged_unpadded.lef)
read_def [in.def](https://github.com/jainsoumil2/ReRAM-16x16/blob/droute_error/openlane/user_project_wrapper/runs/user_project_wrapper/openroad_issue_reproducible/in.def)
global_route

This still shows high congestion:
[image]

So we'll go ahead and bring all pins out to the edges of the macro as @maliberty suggests, and try again! Thanks!!

vijayank88, 2022-04-04:
@3x10e8 
`child killed signal` related to system resource issue.
 Starting at 0th routing optimization its tooks more than 25GB RAM.
I'll update once flow completes how RAM required to complete routing stage.

Increase your system resource and try it.

vijayank88, 2022-04-04:
Routing completed successfully with 28GB RAM. Here's routing view
[image]

3x10e8, 2022-04-04:
Oh I guess we ran out of RAM. Could you share the final output GDS?

vijayank88, 2022-04-05:
@3x10e8 
attached out.def for your reference.
[routing.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8416827/routing.zip)

vijayank88, 2022-04-05:
@3x10e8  attached file here
[routing.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8416852/routing.zip)

vijayank88, 2022-04-05:
@3x10e8 
Considering its related to system resource, closing this issue. Open new issue if you had any other design issues.

closed 2022-04-05 by vijayank88

jainsoumil2, 2022-04-19:
@vijayank88 apologies for replying this late to the thread, but is it still possible that you can share the final output in GDS format? I was having trouble reading the .def file you shared in magic. It's unable to read decap standard cells., amongst other standard cells. It'll be very appreciated!

vijayank88, 2022-04-19:
@jainsoumil2 
Please open a new issue and where you get struck with Magic while loading .def file?
share your command tried to load .def into magic.