The-OpenROAD-Project/OpenLane #1029: LVS mismatch in user_project_wrapper

hdemir93, 2022-03-31:
### Description
Hello everyone, I tried to run user_project_wrapper with my design inside. There is a mismatch in LVS stage, which outputs the log message below.
```
LVS reports:
    net count difference = 0
    device count difference = 0
    unmatched nets = 0
    unmatched devices = 0
    unmatched pins = 2
    property failures = 0

Total errors = 2
``` 
When I search the mismatches in the logs, I have seen these nets caused the error. (This part is not the entire log, the log is inside reproducibles_and_logs.tar.gz)
```
vccd1_uq0                                  |vccd1 **Mismatch**                         
vssd1_uq0                                  |vssd1 **Mismatch** 
vccd1                                      |(no matching pin)                          
vssd1                                      |(no matching pin)                          
vdda2_uq0                                  |(no matching pin)                          
vdda1_uq0                                  |(no matching pin)                          
vccd2_uq0                                  |(no matching pin)                          
vssd2_uq0                                  |(no matching pin)                          
vssa2_uq0                                  |(no matching pin)                          
vssa1_uq0                                  |(no matching pin)  
```
There were no generated tarballs, so I tried my best to include relevant files in it, I can upload another files if needed.
### Environment
```
Kernel: Linux v5.13.0-35-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: -dev
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

e4bfdd7 2022-02-22T16:59:14-03:00 Remove pip install, no longer needed (#953) - Vitor Bandeira -  (grafted, HEAD, tag: 2022.02.23_02.50.41)
```

### Expected behavior
No LVS mismatches expected.

### Reproducibles and Logs
```
[reproducibles_and_logs.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8386738/reproducibles_and_logs.tar.gz)
```

d-m-bailey, 2022-03-31:
@hdemir93 Looks like you have multiple unconnected `vccd1` and `vssd1` texts/labels in your layout. You could try removing the unused ones.

hdemir93, 2022-03-31:
I only used vccd1 and vssd1 in my RTL, it seems the tool automatically synthesizes them. So I can not simply delete them from my RTL, I need to know how I can prevent the tool to synthesize it.

vijayank88, 2022-05-12:
@hdemir93 
Please try with latest openlane and it may resolve.
If you're using more than one macro, follow this comment https://github.com/The-OpenROAD-Project/OpenLane/issues/1076#issuecomment-1124407624 , it may help to resolve your issue.

donn, 2023-04-13:
Staled out.

closed 2023-04-13 by donn