The-OpenROAD-Project/OpenLane #1030: ECO flow doesn't insert fill cells

antonblanchard, 2022-04-01:
The recent ECO update inserts fill cells in a different place. We used to insert it after ECO was complete, now we insert it inside the 
` while {$::env(ECO_FINISH) != 1} {` loop. Amongst other things, this means we never insert fill cells if there were no hold violations in the first place to fix up.

I'm also not sure where we would remove the fill cells on a second iteration.

antonblanchard, 2022-04-12:
@donn your commit f2f5d8358795a14f3eb1c7c9b0a992b2ab2c49f0 seems to have changed some of the logic:

- Previously we'd call `reorg_reports.sh` in each iteration of the loop, now we don't.
- Previously we'd call `ins_fill_cells` once, after the loop is complete. Now we call it once every loop.

donghl17, 2022-04-13:
I'm taking a look at it, and I may probably agree with @antonblanchard because I remember we fixed this problem before.

msaligane, 2022-06-30:
I am happy to help maintain it, but this first version was only supposed to help fix hold in a simple way. Flow could be improved, for instance, to:
 - replace fast cells on fast paths with slower ones (for power reduction/leakage)
 - upsize / downsize buffers
 
Also, I would recommend adding CI checks to ensure updates to the flow do not break the ECO loop.
 We will get to these at some point. Anton is doing a fantastic job stressing and maintaining it.

donn, 2022-07-01:
Ah, alright. Sounds good then. I'll look into CI if you can share a design with me that you use for testing the ECO flow.

closed 2022-07-01 by donn