The-OpenROAD-Project/OpenLane #1032: OpenSTA reports impossibly low numbers

kop316, 2022-04-04:
### Description
In running Openlance, I added `report_units` to `sta.tcl` and `sta_multi_corner.tcl` as shown here:

https://github.com/The-OpenROAD-Project/OpenLane/pull/1031

and I get:

```
 time 1ns
 capacitance 1pF
 resistance 1kohm
 voltage 1v
 current 1mA
 power 1nW
 distance 1um
power_report

===========================================================================
 report_power
============================================================================
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power
----------------------------------------------------------------
Sequential             2.83e-04   5.37e-05   5.49e-10   3.36e-04  64.1%
Combinational          6.50e-05   1.23e-04   1.73e-09   1.88e-04  35.9%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  3.48e-04   1.77e-04   2.28e-09   5.25e-04 100.0%
                          66.3%      33.7%       0.0%
power_report_end
```
Power is in nW, but the Power is in thr 10^-4 range, which is impossible.

Parasitics STA has the same issue:
```
 time 1ns
 capacitance 1pF
 resistance 1kohm
 voltage 1v
 current 1mA
 power 1nW
 distance 1um
power_report

===========================================================================
 report_power
============================================================================
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power
----------------------------------------------------------------
Sequential             2.78e-04   1.28e-04   5.62e-10   4.06e-04  38.7%
Combinational          1.55e-04   4.89e-04   1.23e-07   6.44e-04  61.3%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  4.33e-04   6.17e-04   1.24e-07   1.05e-03 100.0%
                          41.3%      58.7%       0.0%
power_report_end
```

### Environment
```
OpenLane Container (e4bfdd7):/openlane$ python3 ./env.py issue-survey
Kernel: Linux v5.16.0-5-amd64
Distribution: centos 7
Python: v3.6.8 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found: {e}.
OpenLane Git Version: e4bfdd7
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

e4bfdd7 2022-02-22T16:59:14-03:00 Remove pip install, no longer needed (#953) - Vitor Bandeira -  (grafted, HEAD, tag: 2022.02.23_02.50.41)
```

### Reproduction Material
https://github.com/The-OpenROAD-Project/OpenLane/pull/1031

and run `make user_proj_example`

Look in 2-sta.log:

```
 time 1ns
 capacitance 1pF
 resistance 1kohm
 voltage 1v
 current 1mA
 power 1nW
 distance 1um
power_report

===========================================================================
 report_power
============================================================================
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power
----------------------------------------------------------------
Sequential             2.83e-04   5.37e-05   5.49e-10   3.36e-04  64.1%
Combinational          6.50e-05   1.23e-04   1.73e-09   1.88e-04  35.9%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  3.48e-04   1.77e-04   2.28e-09   5.25e-04 100.0%
                          66.3%      33.7%       0.0%
power_report_end
```
```
 time 1ns
 capacitance 1pF
 resistance 1kohm
 voltage 1v
 current 1mA
 power 1nW
 distance 1um
power_report

===========================================================================
 report_power
============================================================================
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power
----------------------------------------------------------------
Sequential             2.78e-04   1.28e-04   5.62e-10   4.06e-04  38.7%
Combinational          1.55e-04   4.89e-04   1.23e-07   6.44e-04  61.3%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  4.33e-04   6.17e-04   1.24e-07   1.05e-03 100.0%
                          41.3%      58.7%       0.0%
power_report_end
```

### Expected behavior
I think the units should be in Watts??

### Logs
```
[26-parasitics_sta.log](https://github.com/The-OpenROAD-Project/OpenLane/files/8406129/26-parasitics_sta.log)
[2-sta.log](https://github.com/The-OpenROAD-Project/OpenLane/files/8406130/2-sta.log)
```

maliberty, 2022-04-04:
The sta docs say "Power values are reported in watts." for report_power.

maliberty, 2022-04-04:
https://github.com/The-OpenROAD-Project/OpenSTA/blob/master/doc/OpenSTA.pdf

kop316, 2022-04-04:
> The sta docs say "Power values are reported in watts." for report_power.

From a UI perspective, that is confusing then. Why does `report_units` report `nW` if `report_power` is in watts? I would think they should be consistent.

Additonally, If I replace https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/openroad/sta.tcl

with only the line:
```
report_units
```

I get:

```
 time 1s
 capacitance 1F
 resistance 1ohm
 voltage 1v
 current 1A
 power 1W
 distance 1m
```

So somehow sta.tcl is modifying `report_units` to be `1nw`.

maliberty, 2022-04-04:
The units are taken from the 1st .lib read

jjcherry56, 2022-04-15:
Units come from the first liberty file, but in the case of report_power it makes no sense to use nW to report chip power so watts are used, documented, and now included in the report for those that don't read documentation.

closed 2022-04-15 by jjcherry56