The-OpenROAD-Project/OpenLane #1037: OpenLANE now fails on PDN with latest OpenROAD changes.

cbalint13, 2022-04-09:
OpenLANE fails sin PDN step due to OpenROAD [pdn: convert to c++ #1749](https://github.com/The-OpenROAD-Project/OpenROAD/pull/1749) .

Error steps [reproductibe 
[pdn-issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8456577/pdn-issue.tar.gz)
attached].
```
$ openlane-flow.tcl -design spm
OpenLane 2022.03.31.02.35.19-20220401.0.git2f276c62.fc37
{...}
[STEP 6]
[INFO]: Generating PDN...
[ERROR]: during executing openroad script /usr/share/openlane/scripts/openroad/pdn.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
[INFO ODB-0130]     Created 36 pins.
[INFO ODB-0131]     Created 478 components and 2698 component-terminals.
[INFO ODB-0133]     Created 352 nets and 984 connections.
[INFO ODB-0134] Finished DEF file: /home/cbalint/rpmbuild/SPECS/MAIN/RUN_2022.04.09_11.42.16/results/floorplan/spm.def
[WARNING PDN-1024] -verbose has been deprecated.
[WARNING PDN-1000] Using legacy PDNGEN.
Consider using "convert_pdn_config /usr/share/openlane/scripts/openroad/pdn_cfg.tcl" to convert the legacy configuration.
[ERROR PDN-0175] Pitch 28.8270 is too small for, must be atleast 28.8300
Error: pdn_cfg.tcl, 39 PDN-0175
{...}
```
OpenROAD version:
```
$ openroad -version
2.0-20220407.git34e0ea38.fc37 
```

---

All [openlane builds fails](https://copr.fedorainfracloud.org/coprs/rezso/VLSI/build/4123317) at post-build regression tests ```%check``` stage. 
Unsure what possible solution would be beyond just having ```convert_pdn_config```.

Cc @arlpetergadfort , @maliberty 
[pdn-issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8456578/pdn-issue.tar.gz)

vijayank88, 2022-04-09:
@cbalint13 
Currently checked with OR version: OpenROAD v2.0-3078-g4e7b85576 
Not facing any issue with test case attached. Successfully completed PDN stage.

```
[INFO PSM-0064] Number of voltage sources = 1.
[INFO PSM-0040] All PDN stripes on net VPWR are connected.
########## IR report #################
Worstcase voltage: 1.80e+00 V
Average IR drop  : 6.48e-10 V
Worstcase IR drop: 1.08e-08 V
######################################
```

cbalint13, 2022-04-09:
> 7b85576

> @cbalint13 Currently checked with OR version: OpenROAD v2.0-3078-g4e7b85576 Not facing any issue with test case attached. Successfully completed PDN stage.

@vijayank88 ,

* The v2.0-3078-g4e7b85576 is one month **before** mentioned PDN changes in openroad.
```
commit 4e7b85576cca102f0ac3cb5ed66c13ada8b3bb75
Merge: 9ef24c565 698596e64
Author: Matt Liberty <mliberty@eng.ucsd.edu>
Date:   Wed Mar 2 16:03:09 2022 -0800
```

> Also try with latest OpenLane tag : [2022.04.08_02.45.18](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.04.08_02.45.18)

To summ up:

* openlane (all versions, including latest) fails with the **newer** openroad.
* openlane suffer this recent openroad change: [pdn: convert to c++ #1749 ](https://github.com/The-OpenROAD-Project/OpenROAD/pull/1749) / 07-Apr-2022.

arlpetergadfort, 2022-04-09:
@cbalint13 I'll take a look, but there are a couple of things that changed.
1. To use the "tcl based" version of  TCL you must use a CFG file (not a hybrid set of commands). This method will be deprecated eventually, To switch you will need to update the pdn.tcl script to something like:
```
# load the grid definitions 
if {[catch {source $::env(PDN_CFG)} errmsg]} {
    puts stderr $errmsg
    exit 1
}
# run PDNGEN
if {[catch {pdngen} errmsg]} {
    puts stderr $errmsg
    exit 1
}
```
2. 28.8270um doesn't appear to match the manufacturing grid of 0.005, so you would probably need to correct that as well. Since the spacing was not specified PDN is computing it and ensuring that the spacing will match the grid (https://github.com/The-OpenROAD-Project/OpenROAD/blob/a2a04f41555c3d7b00108d45d76bf21834ce39e6/src/pdn/src/straps.cpp#L71). PDN probably need to check the inputs will fit on grid too, but since the spacing computing could have been valid (ignoring the issue with the pitch, it should compute the spacing to fit the requested spacing.)

tdene, 2022-04-11:
See [this notebook](https://colab.research.google.com/drive/1bqAWs2To8suxx5acmCYp10iWlKI-Qsn4#scrollTo=3wjq8EkRhv1i) for a publicly reproducible instance of this issue.

The link above should scroll one directly to the `###Create OpenLane config file for our design` cell under `## Run Sklansky adder through OpenLane` under `# Implement adders through OpenLane`.

Note this section:
```
# The following switches are related to OpenLane issue #1037
!echo "set ::env(FP_PDN_AUTO_ADJUST) 0" >> sklansky_config.tcl
!echo "set ::env(FP_PDN_VPITCH) 26.07" >> sklansky_config.tcl
!echo "set ::env(FP_PDN_VOFFSET) 52.14" >> sklansky_config.tcl
```

The issue can be replicated by copying the notebook, removing that specific section, and running through the flow.

The error is of the same origin as that reported by @cbalint13 , and seems to directly stem from a rounding difference. What its root cause may be, I do not have the expertise to venture, but regardless there exists a fatal error related to `FP_PDN_AUTO_ADJUST`.

tdene, 2022-04-12:
To clarify, I believe this issue lies within OpenLane, and is related to recent changes to OpenROAD's PDN code.

The issue, as I see it, is with OpenLane's `FP_PDN_AUTO_ADJUST` setting PDN pitch / offset values that are no longer seen as valid by OpenROAD.

I think there was some confusion in the PR thread. The entity that is dividing by 3 is not @cbalint13, but rather OpenLane itself.

That said, this is not a one-off error, but rather a pervasive one. Any design that uses `FP_PDN_AUTO_ADJUST`, has a small core area, and does not have well-divisible-by-3 parameters, will fatally error out.

I am encountering this issue repeatedly for all of my designs, as is @proppy.

I do not have the expertise to conduct further investigation, and see why exactly [this block of code](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/tcl_commands/floorplan.tcl#L67) no longer works. Perhaps @cbalint13's suggested solution of rounding on lines 72-76 is sufficient, perhaps a more fundamental change is required. But this does need to be fixed.

@maliberty @arlpetergadfort

proppy, 2022-04-12:
> I am encountering this issue repeatedly for all of my designs, as is @proppy.

yep, I can confirm I'm getting the same error with the built-in inverter example: https://colab.research.google.com/gist/proppy/8c8ee486c297dc948f33a14b6efe36f6/openlane-with-conda-eda.ipynb#scrollTo=VP60fdObiP15
```
INFO]: Generating PDN...
[ERROR]: during executing openroad script /content/OpenLane/scripts/openroad/pdn.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
[INFO ODB-0226] Finished LEF file:  /content/runs/RUN_2022.04.12_04.23.43/tmp/merged.unpadded.nom.lef
[INFO ODB-0127] Reading DEF file: /content/runs/RUN_2022.04.12_04.23.43/results/floorplan/inverter.def
[INFO ODB-0128] Design: inverter
[INFO ODB-0130]     Created 2 pins.
[INFO ODB-0131]     Created 32 components and 116 component-terminals.
[INFO ODB-0133]     Created 2 nets and 2 connections.
[INFO ODB-0134] Finished DEF file: /content/runs/RUN_2022.04.12_04.23.43/results/floorplan/inverter.def
[ERROR PDN-0191] Pitch of 7.9870 does not fit the manufacturing grid of 0.0050.
Error: pdn_cfg.tcl, 39 PDN-0191
child process exited abnormally
```

tdene, 2022-04-13:
I recommend that this be closed with #1044 now merged in.

Though it should be noted that this was merely patched over, not properly fixed.

proppy, 2022-04-14:
> Though it should be noted that this was merely patched over, not properly fixed.

is there already another issue documentating what the proper fix should be?

maliberty, 2022-04-14:
@proppy in #1044 there is a discussion of the proper fix.

tdene, 2022-04-14:
@proppy you're absolutely right. See #1050!

cbalint13, 2022-05-11:
#1059  solves the issue.

closed 2022-05-11 by cbalint13