The-OpenROAD-Project/OpenLane #1063: Issue with detail placement

jaquerinte, 2022-04-18:
### Description
User project example with the MPW-6b tag does not work. Ending without providing  an error of what happen in the detail placement. 

### Environment
```
python3 ./env.py issue-survey
Kernel: Linux v5.4.0-107-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.14 (OK)
OpenLane Git Version: 85b6ceca9e38a2187226457ad398f8da3353e8e5
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

85b6cec 2022-04-16T18:04:28+02:00 Fix Jenkins to work with Volare (#1058) - Vitor Bandeira -  (HEAD -> master, tag: 2022.04.17_01.59.58, origin/master, origin/HEAD)
251acb3 2022-04-15T14:19:06+02:00 Quick CI Fix - Mohamed Gaber -  (tag: 2022.04.16_01.55.31)
bb89ac9 2022-04-15T13:15:46+02:00 Integrate Volare for PDK Building and Mangement, Part 2 (#1052) - Mohamed Gaber -  ()
```

### Reproduction Material
make user_proj_example
[issue_openlane.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8506762/issue_openlane.tar.gz)
### Expected behavior
openlane finish correcly or outputs an error to know what happen

### Logs
```
Add any relevant logs here. Please do ensure they're enclosed by the triple-backticks.
```

vijayank88, 2022-04-19:
@jaquerinte 
Its fail during detailed placement. Log shown below:
```
[INFO DPL-0035]  ANTENNA__077570__C1
[INFO DPL-0035]  ANTENNA__096455__A0
[INFO DPL-0035]  ANTENNA__116471__B2
[INFO DPL-0035]  ANTENNA__116987__C1
[INFO DPL-0035]  ANTENNA__116972__A1
[INFO DPL-0035]  ANTENNA__076786__B1
[INFO DPL-0035]  ANTENNA__113405__A
[INFO DPL-0035]  ANTENNA__113405__A
[INFO DPL-0035]  ANTENNA__116989__A2
[INFO DPL-0035]  ANTENNA__092017__B1
[INFO DPL-0035] message limit reached, this message will no longer print
[ERROR DPL-0036] Detailed placement failed.
Error: opendp.tcl, 32 DPL-0036
```
Your placement density looks high.
[image]

If you're using default `CORE_UTIL` mean reduce the value further and try to utilize more core area, also try to relax `PL_TARGET_DENSITY` will resolve this issue.

jaquerinte, 2022-04-19:
Ok, I just re-try increasing the size of the area and reducing density seems to pass that point, but now I have an congestion error. But seems weird, is the same desing that I tape out for MPW-2 and is taking a lot more area and less density. I mantain this issue or I closed beause the issue has change?

maliberty, 2022-04-19:
There are a multitude of possible changes that could be the cause of the difference from MPW-2 (tool, flow, pdk, etc).  If you want to dig into that and narrow it down to an actionable issue it would be interesting.

I think I would make a new issue for the congestion.

jaquerinte, 2022-04-19:
perfect thanks, I will try to investigate that, I will try so solve the congestion first if not I will post a new issue

closed 2022-04-19 by jaquerinte