The-OpenROAD-Project/OpenLane #1064: MPW CI fails with mpw-6b tag and above

vijayank88, 2022-04-20:
### Description
We're maintaining all MPW shuttle designs here: https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI

Earlier stable OpenLane tag we tested is  [2022.03.03_02.36.44](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.03.03_02.36.44).
CI status available here: https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public/detail/PR-20/2/pipeline

Recently I test with OpenLane tag [mpw-6b](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/mpw-6b) and above tags. 
CI status available here: https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public/detail/PR-20/10/pipeline/850

All latest OpenLane tags fails for CI for counter designs with hold and setup violations.
@tspyrou told that openrcx data not correlate with latest commits, so timing violations occurs.

@dralabeing  FYI

antonblanchard, 2022-04-20:
Some discussion of the RC parasitics changing over time in https://github.com/RTimothyEdwards/open_pdks/issues/251

dralabeing, 2022-04-20:
@tspyrou explained that the setrc  tcl file , see example here (https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/sky130hd/setRC.tcl) must be updated so that the global route based timing estimates will correlate with the signoff values after openrcx. Otherwise optimization will not see slack change and hence the timing violations. This was pending. 
@donn what is the status of this?

antonblanchard, 2022-04-20:
I think the issue is not the pre detailed routing parasitic capacitance estimates but bad openrcx data that was fixed recently by @donn and @rtimothyedwards. I'm working with Openlane trunk and the parasitic capacitance estimates are the closest they have ever been.

antonblanchard, 2022-04-20:
FYI Those fixes are post `mpw-6b`

dralabeing, 2022-04-20:
@antonblanchard Thanks for the update and it is good to know that openrcx data updates potentially fixes this problem and that there is better co-relation. (hopefully) from gr timing and openrcx signoff.
@donn will these fixes be committed  into mpw-6b soon ?

vijayank88, 2022-04-21:
With OpenLane tag [2022.04.17_01.59.58](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.04.17_01.59.58) timing violations issue got fixed. Closing this issue.

closed 2022-04-21 by vijayank88