The-OpenROAD-Project/OpenLane #1067: LVS Mismatch Error

vijayank88, 2022-04-22:
### Description
Using OpenLane tag [2022.04.17_01.59.58](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.04.17_01.59.58) fails with LVS.
Its MPW5  precheck passed designs. But its failing with LVS stage with above OpenLane tag.

### Environment
```
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.13 (OK)
OpenLane Git Version: c29cdb4379acb11b67506b8e43f4d85249297498
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

c29cdb4 2022-04-18T17:20:57+02:00 Don't call read_verilog if we are reading a DEF (#1051) - Anton Blanchard -  (HEAD -> master, tag: 2022.04.19_01.43.34, origin/master, origin/HEAD)
754b537 2022-04-18T16:46:39+02:00 Fix `get_tag.py` (#1062) - Mohamed Gaber -  ()
85b6cec 2022-04-16T18:04:28+02:00 Fix Jenkins to work with Volare (#1058) - Vitor Bandeira -  (tag: 2022.04.17_01.59.58)

```

### Reproduction Material
-Unzip the file 
[FMAC_wrapper.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8539952/FMAC_wrapper.zip)

-Copy directory `FMAC_wrapper` to OpenLane/designs/
-cd OpenLane
-make mount
-cd designs/FMAC_wrapper
-source run.tcl

### Expected behavior
LVS clean pass

### Logs
```
vccd2                                      |vccd2                                      
vdda1                                      |vdda1                                      
vdda2                                      |vdda2                                      
vssa1                                      |vssa1                                      
vssa2                                      |vssa2                                      
vssd2                                      |vssd2                                      
(no matching pin)                          |vccd1                                      
(no matching pin)                          |vssd1                                      
vccd1                                      |(no matching pin)                          
vssd1                                      |(no matching pin)                          
---------------------------------------------------------------------------------------
Cell pin lists for user_project_wrapper and user_project_wrapper altered to match.
Cells failed matching, or top level cell failed pin matching.

[ERROR]: There are LVS errors in the design..
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'runs/RUN_2022.04.22_09.22.08/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'runs/RUN_2022.04.22_09.22.08/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

```

antonblanchard, 2022-04-23:
It looks a bit like the macro is unpowered:

```
+ la_oenb[98] la_oenb[99] la_oenb[9] mac_core/vccd1 mac_core/vssd1 wb_clk_i wb_rst_i
```

Perhaps something went wrong with the PDN? Can you check that there are vias connecting met5 to met4 power?

vijayank88, 2022-04-23:
@antonblanchard 
User repo available here: https://github.com/The-OpenROAD-Project/caravel_user_project/tree/KashifInayat/F-MAC

While we trying to push MPW5 design to MPW CI facing such LVS error for many designs with latest openlane tag.

CI logs https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public/detail/PR-26/1/pipeline/1102

antonblanchard, 2022-04-24:
I'm not seeing any vias from met5 to met4 on the macro:

[image]

The PDN code has changed in OpenROAD recently and I'm not sure if Openlane is up to date. @donn or @maliberty might have an idea.

maliberty, 2022-04-25:
I believe OL is still on the older pdngen.   I have a pending PR to update it to the new version.

maliberty, 2022-04-25:
Please package the pdngen step as that's where the connection should be made.

vijayank88, 2022-04-25:
@maliberty
Attached test case here: 
[RUN_2022.04.23_07.19.58_pdn_packaged.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8552074/RUN_2022.04.23_07.19.58_pdn_packaged.zip)

maliberty, 2022-05-10:
@vijayank88 please try with the latest OL which is using the new pdngen.  If the problem persists please package a test case from that version.  The prior test cases are no longer useful as they use the old pdngen.

vijayank88, 2022-05-11:
@maliberty 
Now its fail at LEF LVS
```
[STEP 30]
[INFO]: Running LEF LVS...
[ERROR]: There are LVS errors in the design: See 'runs/RUN_2022.05.11_07.40.10/logs/signoff/30-user_project_wrapper.lvs.lef.log' for details.
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'runs/RUN_2022.05.11_07.40.10/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'runs/RUN_2022.05.11_07.40.10/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

vijayank88, 2022-05-11:
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8668323/openroad_issue_reproducible.zip)

Attached PDN stage DEF file

antonblanchard, 2022-05-11:
```
[WARNING PDN-0189] Supply pin vccd1 of instance mac_core is not connected to any net.
[WARNING PDN-0189] Supply pin vssd1 of instance mac_core is not connected to any net.
```

Do you have the right `FP_PDN_MACRO_HOOKS`? It looks like it might be set to:

```
set ::env(FP_PDN_MACRO_HOOKS) "\
       mprj vccd1 vssd1"
```

Where it should be set to:
```
set ::env(FP_PDN_MACRO_HOOKS) "\
       mac_core vccd1 vssd1"
```

vijayank88, 2022-05-11:
@antonblanchard 
Yes user defined here: https://github.com/KashifInayat/F-MAC/blob/b1e72225d0e1ce3fc3ac89341ce06afc9c5a5db4/openlane/user_project_wrapper/config.tcl#L48

I will modify and test it

vijayank88, 2022-05-11:
Closing this via #1059

closed 2022-05-11 by vijayank88

LvmhZhD, 2022-09-27:
> 我在宏上没有看到从met5到met4的任何通孔：
> 
> [image]
> 
> 最近 OpenRO 中的 PDN 发生了变化，我打开 AD 通道是否是最新的。@donn或者@maliberty可能有一个联想。

hi, i encounter a problem with that `no via insertion between met4 and met5`, and about 150 relative messages totally ,how can i fix it? 

my defination:

set ::env(VDD_NETS) [list {vccd1} {vccd2} {vdda1} {vdda2}]
set ::env(GND_NETS) [list {vssd1} {vssd2} {vssa1} {vssa2}]
set ::env(SYNTH_USE_PG_PINS_DEFINES) "USE_POWER_PINS"

set ::env(VDD_PIN) "vccd1"
set ::env(GND_PIN) "vssd1"

set ::env(FP_PDN_MACRO_HOOKS) project_name vccd1 vssd1 vccd1 vssd1
...

thx

maliberty, 2022-09-27:
@LvmhZhD this seem unrelated - you should open a new issue with a test case

LvmhZhD, 2022-09-27:
> @LvmhZhD这似乎无关 - 您应该使用测试用例打开一个新问题

ok,thx,i will open a new issue