The-OpenROAD-Project/OpenLane #1070: unmapped primitive in synthesis for module $_ALDFF_PP_

vijayank88, 2022-04-25:
### Description
I am facing issue during initial floorplan with module `$_ALDFF_PP_` not found. Its MPW passed design.

### Environment
```
Please run the following set of commands in the OpenLane folder:

make survey || python3 ./env.py issue-survey
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.13 (OK)
OpenLane Git Version: 2022.04.17_01.59.58
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---

```

### Reproduction Material
- Unzip file: 
[GPS_Baseband.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8553049/GPS_Baseband.zip)
- copy `GPS_Baseband` directory inside OpenLane/designs
- cd OpenLane
- make mount
- ./flow.tcl -design GPS_Baseband
This will stop during floorplan stage with primitive module not found error

### Expected behavior
Pass floorplan stage.

### Logs
```
[INFO]: Current run directory is /openlane/designs/GPS_Baseband/runs/RUN_2022.04.25_07.22.53
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[INFO]: Running Synthesis...
[INFO]: Running Single-Corner Static Timing Analysis...
[INFO]: Running Initial Floorplanning...
[ERROR]: Floorplanning failed
[ERROR]: module $_ALDFF_PP_ not found in /openlane/designs/GPS_Baseband/runs/RUN_2022.04.25_07.22.53/tmp/merged.nom.lef
[ERROR]: module $_ALDFF_PN_ not found in /openlane/designs/GPS_Baseband/runs/RUN_2022.04.25_07.22.53/tmp/merged.nom.lef
[ERROR]: Check whether EXTRA_LEFS is set appropriately
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'runs/RUN_2022.04.25_07.22.53/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'runs/RUN_2022.04.25_07.22.53/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

vijayank88, 2022-07-18:
@donn 
Is any configuration there to fix this error?

donn, 2022-07-18:
Try a different synth strategy

vijayank88, 2022-07-18:
Tried all options. Still same error:
```
set ::env(SYNTH_STRATEGY) "DELAY 1"
set ::env(SYNTH_STRATEGY) "DELAY 0"
set ::env(SYNTH_STRATEGY) "AREA 0"
set ::env(SYNTH_STRATEGY) "AREA 1"
set ::env(SYNTH_STRATEGY) "AREA 2"
set ::env(SYNTH_STRATEGY) "DELAY 4"
set ::env(SYNTH_STRATEGY) "DELAY 2"
set ::env(SYNTH_STRATEGY) "DELAY 3"
```

The synthesis netlist snippet as follows:
```
\$_ALDFF_PP_  \gps_engine_i.ch6.ccl.codgen.gold.g2codereg_reg[1]  /* _54382_ */ (
    .AD(\gps_engine_i.ch6.ccl.codgen.gold.code_wire[1] ),
    .C(\gps_engine_i.ch6.ccl.clktempt ),
    .D(_00652_),
    .L(_00653_),
    .Q(\gps_engine_i.ch6.ccl.codgen.gold.g2codereg[1] )
  );

\$_ALDFF_PN_  \gps_engine_i.ch6.clkgen.newword_cod2_reg[0]  /* _54450_ */ (
    .AD(\gps_engine_i.ch6.clkgen.freqword_cod2[0] ),
    .C(\gps_engine_i.ch6.clkgen.accclr01_clk1ms ),
    .D(_00666_),
    .L(io_in[36]),
    .Q(\gps_engine_i.ch6.clkgen.newword_cod2[0] )
  );

```

donn, 2022-07-18:
I'll look into it. There's been a couple regressions in Yosys, but unfortunately upgrading Yosys *further* breaks even more things.

kareefardi, 2022-08-30:
There is no technology mapping for `$_ALDFF_PP_`. You should be able to provide the techmap using `SYNTH_EXTRA_MAPPING_FILE` similar to https://github.com/efabless/caravel/blob/main/openlane/gpio_control_block/yosys_mapping.v

vijayank88, 2022-08-30:
Then its should fall back to design owner only.

Is possible to get link for `ALDFF_PP` mapping file if you're aware of it? @kareefardi

vijayank88, 2022-08-30:
got it here: https://github.com/RTimothyEdwards/chaos_automaton/blob/26c763e32bd86d9d870354874988d2bd03bf1009/openlane/user_project_wrapper/yosys_mapping.v#L22

RTimothyEdwards, 2022-08-30:
I was just going to post that one for you, but you already found it!

vijayank88, 2022-09-06:
@donn 
With `yosys_mapping.v` I'm able resolve this issue.
Is that possible to add this file part of OpenLane repo or some fixed location to use along with OpenLane flow?

kareefardi, 2024-07-17:
I believe it is best for users to add mapping that are not by default provided by the std cell library.

closed 2024-07-17 by kareefardi