The-OpenROAD-Project/OpenLane #1077: Unable to run ./flow.tcl

saji87, 2022-04-30:
While running ./flow.tcl, the following error is displayed.

can't find package json
    while executing
"package require json"
    (file "/home/saji/projects/OpenLane/scripts/tcl_commands/all.tcl" line 15)
    invoked from within
"source /home/saji/projects/OpenLane/scripts/tcl_commands/all.tcl"
    ("package ifneeded openlane 0.9" script)
    invoked from within
"package require openlane"
    (file "./flow.tcl" line 26)

make test ran succesfully. However the docker repository needed to be hard coded in Makefile to pull it as it was empy (efabless/openlane:latest)

vijayank88, 2022-04-30:
@saji87 
Can you please list the commands used before running `./flow.tcl`?
Seems you're not given `-design <Design_Name>`

Make sure followed below steps or follow it, it will resolve your issue.
```
cd OpenLane
make mount
./flow.tcl -design spm
```

saji87, 2022-04-30:
My issue seems to be error in `OPENLANE_IMAGE_NAME` in Makefile. Somehow its empty. 

`ifeq ($(OPENLANE_IMAGE_NAME),)`
`OPENLANE_DOCKER_TAG ?= $(shell $(PYTHON_BIN) ./dependencies/get_tag.py)`
`ifneq ($(OPENLANE_DOCKER_TAG),)`
`export OPENLANE_IMAGE_NAME ?= efabless/openlane:$(OPENLANE_DOCKER_TAG)`
`endif`
`endif`
`

After hardcoding it to efabless/openlane the run is succesful

`[SUCCESS]: Flow complete.
[INFO]: Note that the following warnings have been generated:
..............................
`

May I ask what `THREADS ?= 1`  stands for in Makefile

vijayank88, 2022-04-30:
Can you display?
`which python3`

saji87, 2022-04-30:
`/usr/bin/python3`

vijayank88, 2022-04-30:
You need to do following command before make mount:
```
OPENLANE_IMAGE_NAME=efabless/openlane:current make mount
```

vijayank88, 2022-04-30:
Make sure `pyhton3` default points to `python3.8`

saji87, 2022-04-30:
`>>make survey || python3 ./env.py issue-survey
python3 ./env.py issue-survey
Kernel: Linux v5.13.0-40-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)`

vijayank88, 2022-05-02:
@saji87 
If your issue got fixed, you may close this issue.

closed 2022-05-02 by saji87