The-OpenROAD-Project/OpenLane #1079: ECO Flow failed

vijayank88, 2022-05-02:
### Description
Running mpw-5 design repo: https://github.com/antonblanchard/microwatt-mpw5 with openlane tag: https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.04.28_01.44.59. Its failed during`eco_gen_buffer` by not able to find 
```
no files matched glob pattern "*_unpadded.lef"
    while executing
"glob -directory $::env(RUN_DIR)/tmp  *_unpadded.lef"
```

@antonblanchard FYI

I think recent update in OpenLane `*_unpadded.lef` changed to `*.unpadded.nom.lef`. For other STA corners `min` and `max`. So the `eco_gen_buffer` proc should updated accordingly.

### Environment
```
Please run the following set of commands in the OpenLane folder:

make survey || python3 ./env.py issue-survey
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.13 (OK)
OpenLane Git Version: 2022.04.28_01.44.59
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---

```

@dralabeing FYI

### Reproduction Material
MPW repo: https://github.com/antonblanchard/microwatt-mpw5/blob/main/openlane/Microwatt_FP_DFFRFile/config.tcl

### Expected behavior
ECO flow should pass.

### Logs
```
[INFO]: Starting ECO flow...
[INFO]: [ECO: 0] Checking output...
no files matched glob pattern "*_unpadded.lef"
    while executing
"glob -directory $::env(RUN_DIR)/tmp  *_unpadded.lef"
    (procedure "eco_gen_buffer" line 5)
    invoked from within
"eco_gen_buffer"
    (procedure "eco_output_check" line 4)
    invoked from within
"eco_output_check"
    (procedure "run_eco_flow" line 7)
    invoked from within
"run_eco_flow"
    (procedure "run_eco_step" line 4)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 55)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	if { [info exists arg_values(-file)] } {
		run_file [file normalize $a..."
    (file "./flow.tcl" line 427)
make: *** [quick_run] Error 1
make: Leaving directory `/home/vijayan/FORK_OPENROAD/OL_MPW_CI/OpenLane-MPW-CI/OpenLane'

```

dralabeing, 2022-05-02:
@vijayank88 if the suggested fix works in your repo, please submit a pr.

antonblanchard, 2022-05-02:
Thanks for looking into this @vijayank88. You might also trip over https://github.com/The-OpenROAD-Project/OpenLane/issues/1030

vijayank88, 2022-05-02:
> @vijayank88 if the suggested fix works in your repo, please submit a pr.

@dralabeing 
I am not found any fix for it.

dralabeing, 2022-05-02:
@donn fyi

ramsraj, 2022-06-10:
I have been attempting ECO hold fixes as well, I'm seeing that the re-router iteration post buffer insertion falls over with a cryptic openroad error - any assistance in debug would be much appreciated. Can upload the testcase if anyone would like to reproduce it, please let me know.

Need to mention that I'm using the HS Std-cell library - although don't see how that would make a difference.
Many Thanks, Ram

`[STEP 28]
[INFO]: Running Multi-Corner Static Timing Analysis at the nom process corner...
[INFO]: Starting ECO flow...
[INFO]: [ECO: 0] Checking output...
[INFO]: [ECO: 0] Generating buffer insertion script...
[INFO]: [ECO: 0] Timing violations found, performing another ECO iteration...
[INFO]: [ECO: 1] Starting iteration...
[INFO]: [ECO: 1] Applying fixes...
[INFO]: Routing...
[STEP 29]
[INFO]: Running Global Routing Resizer Timing Optimizations...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/dfdecode7x6/runs/RUN_2022.06.10_10.32.41/logs/routing/29-resizer.log
[ERROR]: Last 10 lines:
47# 0x00007F4A2E028F1E in /lib64/libtcl8.5.so
48# Tcl_EvalEx in /lib64/libtcl8.5.so
49# Tcl_Eval in /lib64/libtcl8.5.so
50# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
51# ord::tclAppInit(Tcl_Interp*) in openroad
52# Tcl_Main in /lib64/libtcl8.5.so
53# main in openroad
54# __libc_start_main in /lib64/libc.so.6
55# 0x0000000001028727 in openroad
child process exited abnormally
`

PR #1685 (2023-03-21): Exit Cleanup
~ Move `QUIT_ON` variables outside of checker functions
~ Move final timing checks to a flow step: `run_timing_check_step`
~ Replace a couple of loose `file exists` with `assert_files_exist` calls
~ Replace all `flow_fail` across the flow with `throw_error`, which behaves accordingly:
    * If running an interactive script, `flow_fail` is called
    * Else, the errors are propagated upwards, where `flow.tcl` is to catch it
~ Rename `try_catch` to `try_exec` as it now propagates the error, with translation behavior
- Remove UMich experimental ECO flow
---
Resolves #1079
Resolves #1 [... cut]
closed 2023-03-21 by donn