The-OpenROAD-Project/OpenLane #1080: Inexplicable DRCs from Magic

maliberty, 2022-05-05:
### Description
Magic produces DRCs that seem false.    It reports violations like:
```
----------------------------------------
Metal2 overlap of via2 < 0.045um in one direction (via2.4a - via2.4)
----------------------------------------
 9.735um 46.335um 9.750um 46.380um
```
where the via looks legal:
[image]

(the small black square in the upper left is a violation marker).  This is viewing the inverter.magic.gds in klayout.
### Environment

This is generated by my fork in https://github.com/The-OpenROAD-Project/OpenLane/pull/1059

python3 ./env.py issue-survey
Kernel: Linux v3.10.0-1160.45.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.14 (OK)
OpenLane Git Version: pdn-update-dev
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

840b5d9 2022-05-04T15:13:27-07:00 check FP_PDN_MACRO_HOOKS is set - Matt Liberty -  (HEAD -> pdn-update, origin/pdn-update)
df89226 2022-05-04T13:38:52-07:00 Update pdn scripts for new pdngen in OR - Matt Liberty -  ()
758fceb 2022-05-04T13:37:58-07:00 Modifed caravel_upw to have to instances connected to different supplies - Matt Liberty -  ()

### Reproduction Material
Attached
[bug.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8628506/bug.zip)

This is just a run of my run area as I don't see a packager for magic issues.  The gds is included.

### Expected behavior
An understandable DRC report.

d-m-bailey, 2022-05-05:
Looking at the magic data, the overlap appears to be only 0.03um. On cifout, magic converts the cut area to vias that are drc clean.
[image]

maliberty, 2022-05-05:
The overlap between what gds layer/datatype exactly?  It does not have that appears in klayout or openroad.

RTimothyEdwards, 2022-05-05:
@maliberty :  So the problem here is that magic gives you error messages relevant to magic's layers, not klayout's.  Magic is giving you an error message referring to a derived layer (magic's "via2") and a rule distance referring to two vendor rules: `(via2.4a - via2.4)`.  Magic's rule for its derived via2 layer is a minimum overlap in one direction of 0.045um, which is the vendor overlap rule (0.085um) minus the minimum overlap that is already included in the derived "via2" layer (0.04um).  The vendor rule states that a minimum overlap of metal2 over (mask layer) via2 in one direction is 0.085um.  Your overlap is only 0.7um, which is why the layout is flagged with a DRC error.

maliberty, 2022-05-06:
I found out what the issue was in OR though I still find magic's report to be hard to grok.

maliberty, 2022-05-06:
@RTimothyEdwards where is the definition of these derived layers?

closed 2022-05-06 by maliberty

RTimothyEdwards, 2022-05-06:
There is no specific definition of the derived layers because I haven't thought about a specific place and format for that information, but I guess it would be helpful to create a page for it in some of the online PDK documentation.

Magic's diagnostic output is written to be meaningful within the context of magic's layout view, and it will always be pretty inscrutable if you pull that diagnostic text out and then try to match it up to the view in another tool like klayout instead of magic.

maliberty, 2022-05-06:
@RTimothyEdwards The input I'm giving is my def file.  It is very confusing that you are modifying the data  I provide and then giving errors against the modified data.  I guess I'll have to learn to use magic as there is no other way to work but it is odd.  I don't have this issue with klayout or Calibre DRCs.

RTimothyEdwards, 2022-05-06:
That's the way magic works.  If magic is the only tool that will read DEF, then your other choice is a little roundabout, but you can just use magic to read DEF and write GDS, then run klayout's DRC checker on the GDS, which will give you results more directly meaningful to the GDS mask-level data.

maliberty, 2022-05-06:
@RTimothyEdwards I don't have a choice as the unit tests are setup to use magic.