The-OpenROAD-Project/OpenLane #1082: chip_integration.md

studyreserve, 2022-05-06:
Greetings.

OpenLane/docs/source/chip_integration.md documentation file in the master branch includes broken links in "Hardening The Full Chip" chapter:
1) "You could take this [full chip](https://github.com/efabless/caravel/blob/mpw-one-b/openlane/caravel/interactive.tcl) as an example."
2) "First you need to harden the padframe as a separate macro, check [this flow](https://github.com/efabless/caravel/blob/mpw-one-b/openlane/chip_io/interactive.tcl) as an example on how to do so."
3) "if it's presented in a chip_io module, otherwise you can harden it separately as indicated in [this flow](https://github.com/efabless/caravel/blob/mpw-one-b/openlane/chip_io/interactive.tcl)."
4,5) "[Here](https://github.com/efabless/caravel/blob/mpw-one-b/openlane/chip_io/padframe.cfg) is an example. Or a hardened chip_io.gds and chip_io.lef following [this](https://github.com/efabless/caravel/blob/mpw-one-b/openlane/chip_io/interactive.tcl)."
6) "[The interactive script for the IOs](https://github.com/efabless/caravel/blob/mpw-one-b/openlane/chip_io/interactive.tcl) does the following:"
7) "Given these inputs the following [interactive script](https://github.com/efabless/caravel/blob/mpw-one-b/openlane/caravel/interactive.tcl) script. Mainly, it does the following steps:"
8) "Run DRC and LVS checks [here](https://github.com/efabless/caravel/blob/mpw-one-b/openlane/caravel/interactive.lvs.tcl)."
9) "Otherwise, refer to [this](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/pdn/doc/PDN.md) for more details about the syntax."

It seems, that caravel project "mpw-one-b" no longer exists, there is "mpw-one" project only, should it leads there?

The last broken link leads to non existing PDN.md file. I could not figure out where it was supposed to lead.

donn, 2022-05-18:
Sadly, this is a more general situation than any one file- OpenLane's documentation is in general need of a concerted update effort.

Some of our team members are doing some work on that, but frankly we're currently swamped with Caravel work. So community input would be sincerely appreciated: including entirely different strategies to handle documentation, if applicable. It's the current biggest sore spot with the flow, unfortunately.

@maliberty What strategy do you use for documenting OpenROAD?

maliberty, 2022-05-18:
@donn are you asking technically (e.g. markdown, sphinx) or more process (probably no better than yours)?

donn, 2022-05-18:
Both, frankly. You have relatively good documentation for the Tcl APIs.

maliberty, 2022-05-18:
@donn We write our documentation with markdown.  @vvbandeira can comment on how we process that to produce the readthedocs.  

TCL commands are pretty straight forward in that when something changes the PR should include a doc update.  Code review is the main enforcement tool, though it still relies on a human check rather than anything automated.  The place where we have more trouble is with high level descriptive documentation as that doesn't belong to any single change/owner.

mithro, 2022-05-18:
BTW - You should use Sphinx

@umacor can also provide advice as he has been working on improving the F4PGA documentation (and hopefully soon the SkyWater PDK documentation too).

vvbandeira, 2022-05-19:
> @vvbandeira can comment on how we process that to produce the readthedocs.

There's nothing fancy here. Just a simple setup and webhook triggered on commits to the master branch to generate the final HTML. A quick look at OL `conf.py` and `.readthedocs.yml` suggests we are using a very similar approach.

donn, 2022-05-29:
@proppy's suggesting the following:

### Prompt

The following page often gets out of date w/ the change to the default TCL configs as shown in #1087:
https://github.com/The-OpenROAD-Project/OpenLane/blob/master/configuration/README.md
https://github.com/The-OpenROAD-Project/OpenLane/blob/master/regression_results/datapoint_definitions.md

### Proposal

It would be nice to find a way to autogenerate those, maybe some of the options listed in https://wiki.tcl-lang.org/page/Source+Documentation+Tools could be explored?

PR #1333 (2022-09-07): Documentation Fixes
\+ Local installer re-documented, volare integrated
\+ Create new "Macros/Chip Integration" section in `configuration.md`, document EXTRA_LIBS
\+ Document `CVC_SCRIPTS_DIR` in PDK variables ~ Replace broken links with permalinks `chip_integration.md`
~ `SYNTH_CLOCK_UNCERTAINITY` -> `SYNTH_CLOCK_UNCERTAINTY` (with translation behavior)
\- Remove unused `FP_CORE_MARGIN`

---
Resolves #1082
Resolves #1124 
Resolves #1144
Resolves #1332
closed 2022-09-07 by donn