The-OpenROAD-Project/OpenLane #1090: MPW-6 Openlane flow fails with [ERROR GRT-0118] Routing congestion too high - Same design routed sucessfully in MPW5 tool set

dineshannayya, 2022-05-12:
### Description
Latest version (MPW-6) of Openlane flow fails with [ERROR GRT-0118] Routing congestion too high  ERROR - 
Same design was previously successfully routed using MPW-5 tools

### Environment
This failure observed with latest version of Openlane : 
<pre><font color="#C4A000">commit 8120faaedf752714e65fb7ff91993a8e6630a664 (</font><font color="#06989A"><b>HEAD -&gt; </b></font><font color="#4E9A06"><b>master</b></font><font color="#C4A000">, </font><font color="#C4A000"><b>tag: 2022.05.12_01.39.47</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/HEAD</b></font><font color="#C4A000">)</font>
Author: Mohamed Gaber &lt;me@donn.website&gt;
Date:   Wed May 11 19:42:22 2022 +0200

    Make `RUN_SPEF_EXTRACTION` skip STA entirely, remove pandas (#1089)
    
    ~ `RUN_MAGIC` check moved to flow.tcl
    ~ `RUN_SPEF_EXTRACTION` now skips STA entirely
    - Remove pandas (it&apos;s overkill for what we need + greatly inflates install times)
</pre>

Latest Openroad version: 
<pre><font color="#C4A000">commit 6d0f5c0116ba235c93c181dfe80246ee5b142bd4 (</font><font color="#06989A"><b>HEAD -&gt; </b></font><font color="#4E9A06"><b>master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/HEAD</b></font><font color="#C4A000">)</font>
Merge: 2eebc2524 e439643eb
Author: arlpetergadfort &lt;46405338+arlpetergadfort@users.noreply.github.com&gt;
Date:   Wed May 11 13:25:56 2022 -0400

    Merge pull request #1844 from The-OpenROAD-Project-staging/gui-speedup
    
    gui: speedup when handling callbacks
</pre>

### Reproduction Material
Attach tarball containing the relevant design share by the openlane flow.
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8676897/openroad_issue_reproducible.tar.gz)

### Expected behavior
A clear and concise description of what you expected to happen.

### Logs
```
Add any relevant logs here. Please do ensure they're enclosed by the triple-backticks.
```

vijayank88, 2022-05-12:
@dineshannayya 
Please update following variable in `config.tcl` will resolve congestion issue:
```
set ::env(GLB_RT_ADJUSTMENT) "0.25"
```

dineshannayya, 2022-05-12:
@vijayank88 Thanks for quick response .. GLB_RT_ADJUSTMENT value changing from 0.3 to 0.25 helped in completion of routing .. Not Sure how this parameter helped in resoling the congestion issue .. I see below comment for this variable

`GLB_RT_ADJUSTMENT` | Reduction in the routing capacity of the edges between the cells in the global routing graph. Values range from 0 to 1. <br> 1 = most reduction, 0 = least reduction  <br> (Default: `0`)|

One more observation after routing phase there no hold violation and RCX extracted sta has timing hold violation.  
[21-grt_sta.min.rpt.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8679005/21-grt_sta.min.rpt.gz)
[30-rcx_sta.min.rpt.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8679006/30-rcx_sta.min.rpt.gz)

Let me know any additional data needed to analysis this issue.

vijayank88, 2022-05-12:
Please share your repo details and design you're trying. I will check at my end and let you know.

dineshannayya, 2022-05-12:
@vijayank88  My Repo is : https://github.com/dineshannayya/riscduino_dcore.git

maliberty, 2022-05-12:
What is the OR commit for mpw5 that was successful?  Do you mean global routing from MPW5 could route this design or that the total flow worked in mpw5.  It is possible that a totally different input to grt is generated in MPW5 and this isn't a grt issue at all.

maliberty, 2022-05-12:
Its also possible it is related to the PDK changes and has nothing to do with OR.

vijayank88, 2022-05-13:
@mithro Please check In MPW CI search for rdq_uart_i2cm_usb_spi_top: https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI/pull/38/checks?check_run_id=6406215768 latest CI run.

No issues found. Also @dineshannayya  using custom port configuration, but we are running auto pin placement, due to PIN_CFG not supported in OpenLane.

dineshannayya, 2022-05-13:
@vijayank88  I have attached custom change removed pin_order file for uart_i2cm_usb_spi_top 
[pin_order.cfg.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8684283/pin_order.cfg.gz)

antonblanchard, 2022-05-13:
As @maliberty mentioned, so much has changed. I know I'm always on the edge of not being able to get microwatt through global routing, but a tweak to here and there (eg `GLB_RT_ADJUSTMENT`) gets me through.

I wonder if our time would be better spent looking at the specific reasons some designs aren't able to pass global routing. In Openlane global placement isn't doing any routability optimisations, so turning it on seems like a good start. I created https://github.com/The-OpenROAD-Project/OpenLane/pull/1092

vijayank88, 2022-05-13:
> @vijayank88 I have attached custom change removed pin_order file for uart_i2cm_usb_spi_top [pin_order.cfg.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8684283/pin_order.cfg.gz)

@dineshannayya 
I just removed `PIN_ORDER_CFG` from your `config.tcl`. So its took auto pin placement and flow got completed without any violations, the repo what you shared.
```
[WARNING]: There are max slew violations in the design at the typical corner. Please refer to 'runs/RUN_2022.05.12_13.43.36/reports/signoff/30-rcx_sta.slew.rpt'.
[INFO]: There are no hold violations in the design at the typical corner.
[WARNING]: There are setup violations in the design at the typical corner. Please refer to 'runs/RUN_2022.05.12_13.43.36/reports/signoff/30-rcx_sta.max.rpt'.
[SUCCESS]: Flow complete.
[INFO]: Note that the following warnings have been generated:
[WARNING]: There are max slew violations in the design at the typical corner. Please refer to 'runs/RUN_2022.05.12_13.43.36/reports/signoff/30-rcx_sta.slew.rpt'.
[WARNING]: There are setup violations in the design at the typical corner. Please refer to 'runs/RUN_2022.05.12_13.43.36/reports/signoff/30-rcx_sta.max.rpt'.
```

vijayank88, 2022-05-13:
Here is max path report:
```
===========================================================================
 report_tns
============================================================================
tns -0.17
tns_report_end
wns_report

===========================================================================
 report_wns
============================================================================
wns -0.17
wns_report_end
worst_slack

===========================================================================
 report_worst_slack -max (Setup)
============================================================================
worst slack -0.17

===========================================================================
 report_worst_slack -min (Hold)
============================================================================
worst slack 0.08
worst_slack_end
clock_skew
```

dineshannayya, 2022-05-13:
@vijayank88 Please share the Git version of Openlane and Openroad tools & openlane docker used for your run.. 

I Don't think PIN Order file can impact the timing fix .. Tools should fix timing issue based on path delays ..

**Openroad app Git log used from my side**
<pre><font color="#C4A000">commit 6d0f5c0116ba235c93c181dfe80246ee5b142bd4 (</font><font color="#06989A"><b>HEAD -&gt; </b></font><font color="#4E9A06"><b>master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/HEAD</b></font><font color="#C4A000">)</font>
Merge: 2eebc2524 e439643eb
Author: arlpetergadfort &lt;46405338+arlpetergadfort@users.noreply.github.com&gt;
Date:   Wed May 11 13:25:56 2022 -0400

    Merge pull request #1844 from The-OpenROAD-Project-staging/gui-speedup
    
    gui: speedup when handling callbacks
</pre>

**Openlane Git log**
<pre><font color="#C4A000">commit 8120faaedf752714e65fb7ff91993a8e6630a664 (</font><font color="#06989A"><b>HEAD -&gt; </b></font><font color="#4E9A06"><b>master</b></font><font color="#C4A000">, </font><font color="#C4A000"><b>tag: 2022.05.12_01.39.47</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/HEAD</b></font><font color="#C4A000">)</font>
Author: Mohamed Gaber &lt;me@donn.website&gt;
Date:   Wed May 11 19:42:22 2022 +0200

    Make `RUN_SPEF_EXTRACTION` skip STA entirely, remove pandas (#1089)
    
    ~ `RUN_MAGIC` check moved to flow.tcl
    ~ `RUN_SPEF_EXTRACTION` now skips STA entirely
    - Remove pandas (it&apos;s overkill for what we need + greatly inflates install times)
</pre>

vijayank88, 2022-05-13:
As per above message you're running with OpenLane: tag: 2022.05.12_01.39.47.
I will use same tag and let you know.

vijayank88, 2022-05-13:
@dineshannayya 
I've updated OpenLane tag: 2022.05.12_01.39.47
OpenROAD commit:   commit: 7510fd4dbe19babb99cbf39525c1b4dac1f2be2d

Again my worst path slack is `-0.17` only getting

dineshannayya, 2022-05-13:
Attached .yml [tool_metadata.yml.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8685717/tool_metadata.yml.gz)

vijayank88, 2022-05-13:
[30-parasitics_sta.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8685784/30-parasitics_sta.zip)
Attached STA report post routing.

Seems you've latest OpenLane tag. But we are not able to reproduce you hold violation issue.

dineshannayya, 2022-05-13:
@vijayank88 I have removed the custom pin-order change + Latest Openlane  +  docker efabless/openlane: latest and still see same hold violation
[30-rcx_sta.min.rpt.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8686267/30-rcx_sta.min.rpt.gz)

Did you tried with efabless/openlane: latest docker ? 

Only other difference i can think of the is PDK, Not sure it can create difference.

vijayank88, 2022-05-13:
`efabless/openlane:2022.05.12_01.39.47` OPENLANE_IMAGE_NAME

vijayank88, 2022-05-13:
Just move your existing pdks directory to some other name and try to update OpenLane.

Because current PDK installation moved to volare. Its fast and keep up-to-date PDK version.

dineshannayya, 2022-05-13:
@vijayank88 Re-installing the latest PDK made the difference, Now there is no hold and slew violation.  

<pre>[INFO]: There are no max slew violations in the design at the typical corner.
[INFO]: There are no hold violations in the design at the typical corner.
[WARNING]: There are setup violations in the design at the typical corner. Please refer to &apos;uart_i2cm_usb_spi_top/runs/uart_i2cm_usb_spi_top/reports/signoff/30-rcx_sta.max.rpt&apos;.
[SUCCESS]: Flow complete.
</pre>

Not sure how PDK update helped here as PDK should have impact same way for both routing and rcx extraction.

Thanks for your effort, we can close this issue.

maliberty, 2022-05-13:
The PDK can affect timing and therefore resizing which affects congestion.

dineshannayya, 2022-05-13:
@maliberty 
PDK is common to both Post Routing Timing analysis and Rcx Extracted (spef) STA analysis. 
Conceptually both timing report should be nearly same.  
Previously Post Route Timing analysis was clean, but Rcx Extract(spef) STA had hold violation.
Look like Post Route and Rcx Extract scripts uses different set of PDK  files and there were some inconsistency in previous version of PDK ?

vijayank88, 2022-05-16:
@dineshannayya 
The recent update in OpenLane includes PDK and RCX extraction results. So your timing report vary from previous shuttle.
If your issue got resolved, please close this issue.

closed 2022-05-16 by dineshannayya