The-OpenROAD-Project/OpenLane #1093: [ERROR STA-0402] Error while pdn generation

ufukyill, 2022-05-14:
### Description
After running "make module_name" I am facing with this error. I have tried:
-Updating openlane then run
-Clonning mpw5 tag then run
-Adding  "set ::env(QUIT_ON_MISMATCHES) 0 "to the config.tcl 
-Changing openroad commit ID from tool_metadata.yml to 7510fd4dbe19babb99cbf39525c1b4dac1f2be2d then "make build-openroad_app" and then "make merge"
### Environment
```
python3 ./env.py issue-survey
Kernel: Linux v5.4.0-110-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.15 (OK)
OpenLane Git Version: 8120faaedf752714e65fb7ff91993a8e6630a664
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

8120faa 2022-05-11T19:42:22+02:00 Make `RUN_SPEF_EXTRACTION` skip STA entirely, remove pandas (#1089) - Mohamed Gaber -  (HEAD -> master, tag: 2022.05.12_01.39.47, origin/master, origin/HEAD)
0c611a3 2022-05-09T08:39:57-07:00 Update to use the new C++ pdngen in OR (#1059) - Matt Liberty -  (tag: 2022.05.11_01.53.47)
47c9f22 2022-05-06T14:32:57+02:00 Remove Hard-coded Paths in Sample Designs (#1083) - Balint Cristian -  (tag: 2022.05.07_01.50.30)

```

### Reproduction Material
"/home/caravel-user/Desktop/AES_accelerator/caravel_example/openlane/AES_top_wb/runs/AES_top_wb/openroad_issue_reproducible"
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8692817/openroad_issue_reproducible.zip)

### Expected behavior
make module_name successfully

### Logs
```
caravel-user@caravel-vm:~/Desktop/AES_accelerator/caravel_example$ make AES_top_wb
export CARAVEL_ROOT=/home/caravel-user/Desktop/AES_accelerator/caravel_example/caravel && cd openlane && make AES_top_wb
make[1]: Entering directory '/home/caravel-user/Desktop/AES_accelerator/caravel_example/openlane'
Makefile:43: warning: undefined variable 'PDK'
Makefile:43: warning: undefined variable 'PDK'
###############################################
OpenLane 8120faaedf752714e65fb7ff91993a8e6630a664
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using design configuration at /home/caravel-user/Desktop/AES_accelerator/caravel_example/openlane/AES_top_wb/config.tcl
[INFO]: Sourcing Configurations from /home/caravel-user/Desktop/AES_accelerator/caravel_example/openlane/AES_top_wb/config.tcl
[INFO]: PDKs root directory: /home/caravel-user/Desktop/OpenLane/pdks
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/caravel-user/Desktop/OpenLane/pdks/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /home/caravel-user/Desktop/AES_accelerator/caravel_example/openlane/AES_top_wb/config.tcl
[WARNING]: Removing existing run at /home/caravel-user/Desktop/AES_accelerator/caravel_example/openlane/AES_top_wb/runs/AES_top_wb...
[INFO]: Current run directory is /home/caravel-user/Desktop/AES_accelerator/caravel_example/openlane/AES_top_wb/runs/AES_top_wb
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[STEP 1]
[INFO]: Running Synthesis...
[STEP 2]
[INFO]: Running Single-Corner Static Timing Analysis...
[STEP 3]
[INFO]: Running Initial Floorplanning...
[INFO]: Setting Core Dimensions...
[STEP 4]
[INFO]: Running IO Placement...
[STEP 5]
[INFO]: Running Tap/Decap Insertion...
[INFO]: Power planning with power {vccd1} and ground {vssd1}...
[STEP 6]
[INFO]: Generating PDN...
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: AES_top_wb/runs/AES_top_wb/logs/floorplan/6-pdn.log
[ERROR]: Last 10 lines:
[INFO ODB-0097] 		Created 100000 Nets
[INFO ODB-0097] 		Created 200000 Nets
[INFO ODB-0097] 		Created 300000 Nets
[INFO ODB-0130]     Created 106 pins.
[INFO ODB-0131]     Created 385036 components and 2480077 component-terminals.
[INFO ODB-0133]     Created 308127 nets and 1090520 connections.
[INFO ODB-0134] Finished DEF file: /home/caravel-user/Desktop/AES_accelerator/caravel_example/openlane/AES_top_wb/runs/AES_top_wb/results/floorplan/AES_top_wb.def
[ERROR STA-0402] define_pdn_grid -default is not a known keyword or flag.
Error: pdn_cfg.tcl, 161 STA-0402
child process exited abnormally

[INFO]: Saving runtime environment...
[INFO]: Creating reproducible...

or_issue.py OpenROAD Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /home/caravel-user/Desktop/AES_accelerator/caravel_example/openlane/AES_top_wb/runs/AES_top_wb/openroad_issue_reproducible…
[FIN] Done.
[INFO]: Reproducible packaged: Please tarball and upload AES_top_wb/runs/AES_top_wb/openroad_issue_reproducible if you're going to submit an issue.
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'AES_top_wb/runs/AES_top_wb/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'AES_top_wb/runs/AES_top_wb/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    (procedure "run_openroad_script" line 66)
    invoked from within
"run_openroad_script $::env(SCRIPTS_DIR)/openroad/pdn.tcl  |& -indexed_log [index_file $::env(floorplan_logs)/pdn.log]"
    (procedure "gen_pdn" line 9)
    invoked from within
"gen_pdn"
    (procedure "run_power_grid_generation" line 65)
    invoked from within
"run_power_grid_generation"
    (procedure "run_floorplan" line 54)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 55)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	if { [info exists arg_values(-file)] } {
		run_file [file normalize $a..."
    (file "/openlane/flow.tcl" line 434)
make[1]: *** [Makefile:43: AES_top_wb] Error 1
make[1]: Leaving directory '/home/caravel-user/Desktop/AES_accelerator/caravel_example/openlane'
make: *** [Makefile:70: AES_top_wb] Error 2

```

vijayank88, 2022-05-16:
@ufukyill  
In `pdn_cfg.tcl` remove line `158` and continue, until get fix from upstream.

vijayank88, 2022-05-16:
@maliberty  @donn 
While using one of private PDK I also faced same issue during PDN stage.

maliberty, 2022-05-16:
I ran your test case with the latest OR (8da44250940f63186c203c95953c068e44fde2d1) and I get no error.   Perhaps try with the latest OL  (260dd7f80102e4b1726fdee8cd596b06822ca252) and see if it still occurs.

maliberty, 2022-05-16:
@vijayank88 are you using private PDKs with OL?

ufukyill, 2022-05-17:
@vijayank88
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8707045/openroad_issue_reproducible.zip)

vijayank88, 2022-05-17:
```
INFO DPL-0035]  rebuffer1940
[INFO DPL-0035]  rebuffer1940
[INFO DPL-0035]  rebuffer2435
[INFO DPL-0035]  rebuffer2435
[INFO DPL-0035]  rebuffer2449
[INFO DPL-0035]  rebuffer2532
[INFO DPL-0035]  rebuffer2532
[INFO DPL-0035]  rebuffer2532
[INFO DPL-0035]  rebuffer2532
[ERROR DPL-0036] Detailed placement failed.
Error: resizer_timing.tcl, 79 DPL-0036
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_timing.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[INFO]: Creating reproducible...
[INFO]: Saving runtime environment...

or_issue.py OpenROAD Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /home/caravel-user/Desktop/AES_accelerator/caravelwdesign/openlane/AES_top_wb/runs/AES_top_wb/openroad_issue_reproducible…
[WRN] /home/caravel-user/Desktop/AES_accelerator/caravelwdesign/openlane/AES_top_wb/runs/AES_top_wb/tmp/cts/13-resizer_timing.sdc was not found, might be a product. Skipping
[FIN] Done.
[INFO]: Reproducible packaged: Please tarball and upload AES_top_wb/runs/AES_top_wb/openroad_issue_reproducible if you're going to submit an issue.
[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: AES_top_wb
Run Directory: /home/caravel-user/Desktop/AES_accelerator/caravelwdesign/openlane/AES_top_wb/runs/AES_top_wb
Source not found.
----------------------------------------

LVS Summary:
Source: /home/caravel-user/Desktop/AES_accelerator/caravelwdesign/openlane/AES_top_wb/runs/AES_top_wb/logs/finishing/AES_top_wb.lvs.lef.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /home/caravel-user/Desktop/AES_accelerator/caravelwdesign/openlane/AES_top_wb/runs/AES_top_wb/reports/final_summary_report.csv
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    (procedure "run_openroad_script" line 62)
    invoked from within
"run_openroad_script $::env(SCRIPTS_DIR)/openroad/resizer_timing.tcl -indexed_log [index_file $::env(cts_logs)/resizer.log]"
    (procedure "run_resizer_timing" line 8)
    invoked from within
"run_resizer_timing"
    (procedure "run_cts_step" line 9)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 55)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	puts_info "Note, that post_run_hook..."
    (file "/openlane/flow.tcl" line 412)
make[1]: *** [Makefile:43: AES_top_wb] Error 1
make[1]: Leaving directory '/home/caravel-user/Desktop/AES_accelerator/caravelwdesign/openlane'
make: *** [Makefile:72: AES_top_wb] Error 2
```

vijayank88, 2022-05-17:
@ufukyill 
Update below variable in `config.tcl` and re-run the flow.
```
set ::env(CELL_PAD) 2
```

vijayank88, 2022-05-17:
@ufukyill 
Try again with `set ::env(GLB_RT_ADJUSTMENT) "0.15"`

maliberty, 2022-05-17:
This issue is no longer about pdngen and should not be taken over for a different problem.  Is there any remaining pdngen issue here?

donn, 2022-05-18:
That is indeed a different issue, and one that's probably fixed in #1092.

closed 2022-05-18 by donn

proppy, 2022-05-27:
> Is there any remaining pdngen issue here?
I can confirm that updating openroad does seem to fix this issue.