The-OpenROAD-Project/OpenLane #1095: Add OpenROAD remove_buffers to flow

antonblanchard, 2022-05-16:
@jjcherry56 points out that we should have the option of calling `remove_buffers` before we call `repair_design`, to undo any buffer addition that yosys/abc does.

To complicate matters, Openlane currently has an internal command called `remove_buffers`. It's controlled by an undocumented environment variable `DONT_BUFFER_PORTS`. What is it used for? @donn any ideas?

I have a distant memory that it was to remove buffers inserted around tristates. I did find it a mention of it in caravel, but it's commented out:

```
# mgmt_gpio_in is driven by a tristate cell
# set ::env(DONT_BUFFER_PORTS) "mgmt_gpio_in"
```

If we are inserting buffers around tristates, I would think we should fix OpenROAD, and just remove this.

maliberty, 2022-05-16:
There was a bug a while back where resizer but a buffer after a tristate but that was fixed.

antonblanchard, 2022-05-17:
> There was a bug a while back where resizer but a buffer after a tristate but that was fixed.

Thanks @maliberty, I created https://github.com/The-OpenROAD-Project/OpenLane/pull/1098