The-OpenROAD-Project/OpenLane #1096: PDN flow/script enhancement : There should be independent control on core ring and power Strip generation

dineshannayya, 2022-05-17:
### Prompt
Current MPW-6 Openlane has common config to decide on number of core ring and power strip.

Issue: In MPW shuttle user project need  have core-ring for all the 8 power (vccd1 vccd2 vdda1 vdda2 vssd1 vssd2 vssa1 vssa2), even though user project need one set of power domain (vccd1 vssd1). 
Default 8 power strips requirement make keep minimum 8 power-strip space between each hard macros. Highly congested SOC design will not have enough free space to support this.  

### Proposal
Given separate control 
> For Core Ring, use VDD_PINS/GND_PINS to decide on number of power in core ring
> For Power Strip, use VDD_NETS/GND_NETS to decide on number of power strip

maliberty, 2022-05-17:
@arlpetergadfort FYI

maliberty, 2022-05-19:
@donn OL should pick up the latest OR with Peter's fix and expose a set of flow controls.  I believe this is done from an OR perspective.

mattvenn, 2022-05-20:
this would be great for me too, as I'm trying to get 450 small designs into the area and so I need a much finer PDN. With all 4 psus I can't manage it. With 1, it works but then precheck job fails.

egorxe, 2022-05-26:
> this would be great for me too, as I'm trying to get 450 small designs into the area and so I need a much finer PDN. With all 4 psus I can't manage it. With 1, it works but then precheck job fails.

I could confirm that with the latest OR there is an easy way to increase PDN density by disabling secondary supply stripes. All that is needed is to add options like these to add_stripes in pdn_cfg.tcl:
```
-spacing [expr {($::env(FP_PDN_VPITCH) / 2) - $::env(FP_PDN_VWIDTH)}] \
-nets [list $::env(VDD_NET) $::env(GND_NET)]
```
It would be even more convenient if OR recalculated spacing automatically. Together with setting FP_PDN_CORE_RING_VOFFSET & FP_PDN_CORE_RING_HOFFSET to 12.45 it allows you to pass XOR precheck with high PDN density.

I could provide a pull request with per supply control of stripes generation by OL variables if required.

maliberty, 2022-05-26:
You should look at https://github.com/The-OpenROAD-Project/OpenLane/pull/1109

egorxe, 2022-05-26:
> You should look at #1109

#1109 is about power rings order and this issue is about separate control for rings and straps. They are related, but not the same.

dineshannayya, 2022-11-11:
This is old ticket, PDN has this support now

closed 2022-11-11 by dineshannayya