The-OpenROAD-Project/OpenLane #1100: allow separate padding for global & detailed placement

maliberty, 2022-05-17:
### Prompt
Using a single padding value limits density and makes placement more difficult

### Proposal
In ORFS we have separate values for detailed and global [padding](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/1182d2619e25a721b0ce065d2221d298939e5989/flow/platforms/sky130hd/config.mk#L111):
```
export CELL_PAD_IN_SITES_GLOBAL_PLACEMENT ?= 1
export CELL_PAD_IN_SITES_DETAIL_PLACEMENT ?= 0
```
Whereas in OL you use a single CELL_PAD (default 4).  This makes both gpl & dpl use the same padding and makes it more difficult to accommodate added/upsized cells during resizing.   I suggest OL adopt a similar approach to ORFS.

maliberty, 2022-05-17:
@antonblanchard FYI

antonblanchard, 2022-05-18:
@maliberty sounds good. I also note that we only add padding to the right side, except for clock tree buffers and diodes, where we only add padding to the left side:

```
scripts/openroad/replace.tcl:set_placement_padding -global -right $::env(CELL_PAD)
scripts/openroad/opendp.tcl:set_placement_padding -global -right $::env(CELL_PAD)
scripts/openroad/resizer.tcl:set_placement_padding -global -right $::env(CELL_PAD)
scripts/openroad/resizer_routing_timing.tcl:# set_placement_padding -global -right $::env(CELL_PAD)
scripts/openroad/resizer_timing.tcl:set_placement_padding -global -right $::env(CELL_PAD)
```

```
scripts/openroad/cts.tcl:set_placement_padding -masters $buffers -left $::env(CELL_PAD)
scripts/openroad/diodes.tcl:set_placement_padding -masters $::env(DIODE_CELL) -left $::env(DIODE_PADDING)
scripts/openroad/groute.tcl:    set_placement_padding -masters $::env(DIODE_CELL) -left $::env(DIODE_PADDING)
```

I know it is unlikely to make much difference, but it is odd enough that I trip over it every few months and wonder what the rationale was.

maliberty, 2022-05-18:
@antonblanchard I have no clue why - in ORFS we use balanced padding.

vijayank88, 2022-05-18:
I like to make this changes in OpenLane What @maliberty suggesting.

For small design `CELL_PAD=4` works ok. For little complex designs its failing during DPL/GRT stage.

Most of the new user facing this issue and am pointing to change `CELL_PAD=2 or 1` which ever resolves the issue.

maliberty, 2022-05-19:
To be clear I am hoping the OL team with implement this (@donn) not me

antonblanchard, 2022-06-29:
I started looking at this. We aren't applying any padding during global placement because we are using the wrong interface:

```
set_placement_padding -global -right $::env(CELL_PAD)

if { $::env(CELL_PAD_EXCLUDE) != "" } {
    set_placement_padding -masters $::env(CELL_PAD_EXCLUDE) -right 0 -left 0
}

global_placement {*}$arg_list
```

For global placement padding we need to use `global_placement -pad_right $::env(CELL_PAD) ...`

That explains some of the detailed placement issues we've seen. Fixing this is going to mean almost every design will need a tweak to `PL_TARGET_DENSITY` however.

maliberty, 2022-06-29:
@antonblanchard good catch!

antonblanchard, 2022-06-29:
As expected, a lot of CI breakage with this fix: https://github.com/The-OpenROAD-Project/OpenLane/pull/1172

vijayank88, 2022-06-29:
We need @donn  help here for metrics change?

antonblanchard, 2022-06-29:
@vijayank88 the problem is any design with a `PL_TARGET_DENSITY` set near the limit is going to break with any value of padding. Reducing the values will help, although it may break some designs that relied on larger spacing in detailed placement to help with routing.

donn, 2022-07-21:
> For global placement padding we need to use global_placement -pad_right $::env(CELL_PAD) ...

`-pad_right`? I thought we were balancing?

PR #1226 (2022-07-25): Tie Cells at FP Stage, Improved Cell Padding
\+ Added "logging to" information for long-running steps (CTS and Routing)
\+ Added variable `GPL_CELL_PADDING` for global placement, with a default value of 0 (value divided by 2 and applied to both the left and right of the cell)
\+ Add feature to also print mounted scripts' version if the container version does not match
\+ Add `insert_tiecells` after floorplanning
~ `replace.tcl`, `opendp.tcl` -> `gpl.tcl`, `dpl.tcl`
~ `CELL_PAD` renamed `DPL_CELL_PADDING` for detailed placement, default value still 4, value divided by 2 and applied to both the  [... cut]
closed 2022-07-25 by donn