The-OpenROAD-Project/OpenLane #1102: [ERROR PDN-0165] Conflict found, instance is part of two grid definitions (CORE_macro_grid_2, CORE_macro_grid_1)

jurevreca12, 2022-05-23:
### Description
I've been trying to harden my SoC using openlane. I was successful in hardening all my macros, but the flow fails when I try to integrate the macros in the final design. The concrete error output is:
```
###############################################
[INFO]: 
	___   ____   ___  ____   _       ____  ____     ___
	/   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
	|   | |  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
	| O | |   _/    _]|  |  || |___ |     ||  |  ||    _]
	|   | |  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
	\___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Version: 2022.02.23_02.50.41
[INFO]: Running non-interactively
[INFO]: Using design configuration at /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/config.tcl
[INFO]: Sourcing Configurations from /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/config.tcl
[INFO]: PDKs root directory: /home/jure/Projekti/PDK
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/jure/Projekti/PDK/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/config.tcl
[WARNING]: Removing exisiting run /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper
[INFO]: Current run directory is /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper
[INFO]: Storing configs into config.tcl ...
[INFO]: Preparing LEF Files
[INFO]: Extracting the number of available metal layers from /home/jure/Projekti/PDK/sky130A/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd.tlef
[INFO]: The available metal layers (6) are li1  met1  met2  met3  met4  met5
[INFO]: Merging LEF Files...
mergeLef.py : Merging LEFs
sky130_fd_sc_hd.lef: SITEs matched found: 0
sky130_fd_sc_hd.lef: MACROs matched found: 441
mergeLef.py : Merging LEFs complete
mergeLef.py : Merging LEFs
sky130_sram_2kbyte_1rw1r_32x512_8.lef: SITEs matched found: 0
sky130_sram_2kbyte_1rw1r_32x512_8.lef: MACROs matched found: 1
wbuart_wrap.lef: SITEs matched found: 0
wbuart_wrap.lef: MACROs matched found: 1
rvj1_caravel_soc.lef: SITEs matched found: 0
rvj1_caravel_soc.lef: MACROs matched found: 1
mergeLef.py : Merging LEFs complete
[INFO]: Merging the following extra LEFs:  /home/jure/Projekti/PDK/sky130A/libs.ref/sky130_sram_macros/lef/sky130_sram_2kbyte_1rw1r_32x512_8.lef  /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/../../lef/wbuart_wrap.lef  /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/../../lef/rvj1_caravel_soc.lef
[INFO]: Trimming Liberty...
[INFO]: Generating Exclude List...
[INFO]: Generating Exclude List...
[INFO]: Creating ::env(DONT_USE_CELLS)...
[INFO]: Looking for files defined in ::env(EXTRA_GDS_FILES)  /home/jure/Projekti/PDK/sky130A/libs.ref/sky130_sram_macros/gds/sky130_sram_2kbyte_1rw1r_32x512_8.gds  /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/../../gds/wbuart_wrap.gds  /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/../../gds/rvj1_caravel_soc.gds ...
[INFO]: /home/jure/Projekti/PDK/sky130A/libs.ref/sky130_sram_macros/gds/sky130_sram_2kbyte_1rw1r_32x512_8.gds exists.
[ERROR]: /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/../../gds/wbuart_wrap.gds doesn't exist.
[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: user_project_wrapper
Run Directory: /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper
Source not found.
----------------------------------------

LVS Summary:
Source: /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper/logs/finishing/user_project_wrapper.lvs.lef.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper/reports/final_summary_report.csv
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    (procedure "assert_files_exist" line 5)
    invoked from within
"assert_files_exist $::env(EXTRA_GDS_FILES)"
    (procedure "prep" line 375)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 11)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	puts_info "Note, that post_run_hook..."
    (file "/openlane/flow.tcl" line 412)
make[1]: *** [Makefile:43: user_project_wrapper] Error 1
make[1]: Leaving directory '/home/jure/Projekti/rvj1-caravel-soc/openlane'
make: *** [Makefile:72: user_project_wrapper] Error 2

```

### Environment
This github repo https://github.com/jurevreca12/rvj1-caravel-soc/tree/develop contains the relevant  code. To reproduce the issue run: make user_project_wrapper. Beware that the relevant code is in the develop branch, and not the main branch.

### Reproduction Material
I am also attaching the run directory of openlane as a zip file (user_project_wrapper.zip) in this issue.

### Expected behavior
A successfully generated final gds of the design.

### Logs
None

[user_project_wrapper.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8754509/user_project_wrapper.zip)

vijayank88, 2022-05-23:
@jurevreca12 this is not a valid test case

jurevreca12, 2022-05-23:
Sorry, about that. I attached the correct openroad_issue_reproducible.zip now.

mattvenn, 2022-05-23:
thanks for opening

vijayank88, 2022-05-23:
@jurevreca12 
your macro placed outside die area. 
[image]

jurevreca12, 2022-05-23:
How do I get this view? I've been (literally) drawing this on a piece of paper...

vijayank88, 2022-05-23:
Use `openroad -gui`

mattvenn, 2022-05-23:
or look at the floorplan def file

maliberty, 2022-05-23:
The error shown in the report is "[ERROR]: /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/../../gds/wbuart_wrap.gds doesn't exist." which doesn't seem to have anything to do with the title of this issue.

jurevreca12, 2022-05-23:
@maliberty  ah yes. I've compressed the repo and then ran again to get the message. I'll fix this.

jurevreca12, 2022-05-23:
@vijayank88 Thank you for the help. This fixed my issue.

vijayank88, 2022-05-23:
@jurevreca12 
If fixed, close this issue. Open new issue if you're facing any other issue again. Thanks.

closed 2022-05-23 by jurevreca12