The-OpenROAD-Project/OpenLane #1103: Openlane [Tag: 2022.05.18_02.12.32] low fails with [ERROR DPL-0036] Detailed placement failed Error

dineshannayya, 2022-05-23:
### Description
Latest openlane [Tag: a2022.05.18_02.12.32] tool flow fails with [ERROR DPL-0036] Detailed placement failed Error. Same design use work with older version tool sets

### Environment
Openlane git repo log

<pre><font color="#C4A000">commit 0dc6fb79c91082b94f8ded78d70f8bacbab96bf2 (</font><font color="#06989A"><b>HEAD -&gt; </b></font><font color="#4E9A06"><b>master</b></font><font color="#C4A000">, </font><font color="#C4A000"><b>tag: 2022.05.18_02.12.32</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/HEAD</b></font><font color="#C4A000">)</font>
Author: Anton Blanchard &lt;anton@linux.ibm.com&gt;
Date:   Wed May 18 01:24:25 2022 +1000

    Enable PL_ROUTABILITY_DRIVEN and PL_TIME_DRIVEN (#1092)
    
    ORFS enables both of these by default.
</pre>

<pre><font color="#C4A000">commit 6bcd930ce1e818cd86cfd491ba015a8cc5a98408 (</font><font color="#06989A"><b>HEAD -&gt; </b></font><font color="#4E9A06"><b>master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/HEAD</b></font><font color="#C4A000">)</font>
Merge: a9682175a 17bde4c8f
Author: Matt Liberty &lt;mliberty@eng.ucsd.edu&gt;
Date:   Wed May 18 22:20:47 2022 -0700

    Merge pull request #1883 from The-OpenROAD-Project-staging/ifp-obsolete
    
    Ifp obsolete
</pre>

### Reproduction Material
Attached the openroad_issue_reproducible directory.

[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8754668/openroad_issue_reproducible.tar.gz)

### Expected behavior

### Logs
Openlane failure log

<pre>[STEP 18]
[INFO]: Running Detailed Placement...
[ERROR]: during executing openroad script /openlane/scripts/openroad/opendp.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: uart_i2cm_usb_spi_top/runs/uart_i2cm_usb_spi_top/logs/routing/18-diode_legalization.log
[ERROR]: Last 10 lines:
[INFO DPL-0035]  ANTENNA_u_skew_uart.u_mux_level_10_S
[INFO DPL-0035]  ANTENNA_u_skew_uart.u_mux_level_00_S
[INFO DPL-0035]  ANTENNA_u_skew_uart.u_mux_level_11_S
[INFO DPL-0035]  ANTENNA_u_skew_uart.u_mux_level_11_S
[INFO DPL-0035]  ANTENNA_u_skew_uart.u_mux_level_07_S
[INFO DPL-0035]  ANTENNA_u_skew_uart.u_mux_level_06_S
[INFO DPL-0035]  ANTENNA_u_skew_uart.u_mux_level_06_S
[ERROR DPL-0036] Detailed placement failed.
Error: opendp.tcl, 32 DPL-0036
child process exited abnormally

</pre>

vijayank88, 2022-05-23:
@dineshannayya 
Update following in `config.tcl` will resolve the issue.
`set ::env(CELL_PAD) {2}`

antonblanchard, 2022-05-24:
https://github.com/The-OpenROAD-Project/OpenLane/issues/1100 might help here

jjcherry56, 2022-05-24:
or without gratuitous punctuation
set env(CELL_PAD) 2

dineshannayya, 2022-05-25:
Yes with CELL_PAD 2 working now ..  Closing the ticket

closed 2022-05-25 by dineshannayya