The-OpenROAD-Project/OpenLane #1104: Repost: [ERROR PDN-0165] Conflict found, instance dram_inst is part of two grid definitions (CORE_macro_grid_2, CORE_macro_grid_1).

jurevreca12, 2022-05-23:
### Description
This is a repost of a recently closed issue, which was closed to soon: https://github.com/The-OpenROAD-Project/OpenLane/issues/1102. The solution mentioned in the referenced issue works, but only if i comment out FP_PDN_MACRO_HOOKS, if not I still get the error bellow:
```
Yosys 0.12+45 (git sha1 UNKNOWN, gcc 8.3.1 -fPIC -Os)
Time spent: 67% 2x stat (0 sec), 20% 2x write_verilog (0 sec), ...
[INFO]: Running Floorplanning...
[INFO]: Running Initial Floorplanning...
[INFO]: Incremented step index to 2.
OpenROAD 8d53e9b018dec98fa63e907ddeb6c5406f035361 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper/tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 444 library cells
[INFO ODB-0226] Finished LEF file:  /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper/tmp/merged_unpadded.lef
[WARNING STA-0200] /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper/results/synthesis/user_project_wrapper.v line 307, instance rvj1_soc port dram_addr0 size 2 does not match net size 9.
[WARNING STA-0200] /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper/results/synthesis/user_project_wrapper.v line 307, instance rvj1_soc port iram_addr0 size 2 does not match net size 9.
[WARNING ORD-1011] LEF master sky130_sram_2kbyte_1rw1r_32x512_8 has no liberty cell.
[WARNING ORD-1011] LEF master rvj1_caravel_soc has no liberty cell.
[WARNING ORD-1011] LEF master wbuart_wrap has no liberty cell.
if {[info exists ::env(CLOCK_PORT)] && $::env(CLOCK_PORT) != ""} { 
    create_clock [get_ports $::env(CLOCK_PORT)]  -name $::env(CLOCK_PORT)  -period $::env(CLOCK_PERIOD)
} else {
    create_clock -name __VIRTUAL_CLK__ -period $::env(CLOCK_PERIOD)
    set ::env(CLOCK_PORT) __VIRTUAL_CLK__
}
set input_delay_value [expr $::env(CLOCK_PERIOD) * $::env(IO_PCT)]
set output_delay_value [expr $::env(CLOCK_PERIOD) * $::env(IO_PCT)]
puts "\[INFO\]: Setting output delay to: $output_delay_value"
[INFO]: Setting output delay to: 2.0
puts "\[INFO\]: Setting input delay to: $input_delay_value"
[INFO]: Setting input delay to: 2.0
set_max_fanout $::env(SYNTH_MAX_FANOUT) [current_design]
set clk_indx [lsearch [all_inputs] [get_port $::env(CLOCK_PORT)]]
#set rst_indx [lsearch [all_inputs] [get_port resetn]]
set all_inputs_wo_clk [lreplace [all_inputs] $clk_indx $clk_indx]
#set all_inputs_wo_clk_rst [lreplace $all_inputs_wo_clk $rst_indx $rst_indx]
set all_inputs_wo_clk_rst $all_inputs_wo_clk
# correct resetn
set_input_delay $input_delay_value  -clock [get_clocks $::env(CLOCK_PORT)] $all_inputs_wo_clk_rst
#set_input_delay 0.0 -clock [get_clocks $::env(CLOCK_PORT)] {resetn}
set_output_delay $output_delay_value  -clock [get_clocks $::env(CLOCK_PORT)] [all_outputs]
# TODO set this as parameter
set_driving_cell -lib_cell $::env(SYNTH_DRIVING_CELL) -pin $::env(SYNTH_DRIVING_CELL_PIN) [all_inputs]
set cap_load [expr $::env(SYNTH_CAP_LOAD) / 1000.0]
puts "\[INFO\]: Setting load to: $cap_load"
[INFO]: Setting load to: 0.033442
set_load  $cap_load [all_outputs]
puts "\[INFO\]: Setting clock uncertainity to: $::env(SYNTH_CLOCK_UNCERTAINITY)"
[INFO]: Setting clock uncertainity to: 0.25
set_clock_uncertainty $::env(SYNTH_CLOCK_UNCERTAINITY) [get_clocks $::env(CLOCK_PORT)]
puts "\[INFO\]: Setting clock transition to: $::env(SYNTH_CLOCK_TRANSITION)"
[INFO]: Setting clock transition to: 0.15
set_clock_transition $::env(SYNTH_CLOCK_TRANSITION) [get_clocks $::env(CLOCK_PORT)]
puts "\[INFO\]: Setting timing derate to: [expr {$::env(SYNTH_TIMING_DERATE) * 10}] %"
[INFO]: Setting timing derate to: 0.5 %
set_timing_derate -early [expr {1-$::env(SYNTH_TIMING_DERATE)}]
set_timing_derate -late [expr {1+$::env(SYNTH_TIMING_DERATE)}]
[WARNING IFP-0028] Core area lower left (5.520, 10.880) snapped to (5.520, 10.880).
[INFO IFP-0001] Added 1286 rows of 6323 sites.
[INFO]: Core area width: 2908.96
[INFO]: Core area height: 3498.24
[INFO]: Final Vertical PDN Offset: 5
[INFO]: Final Horizontal PDN Offset: 5
[INFO]: Final Vertical PDN Pitch: 180
[INFO]: Final Horizontal PDN Pitch: 180
[INFO]: Changing layout from 0 to /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper/tmp/floorplan/3-initial_fp.def
[INFO]: Setting Core Dimensions...
[INFO]: Incremented step index to 3.
[... 134 lines cut ...]
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	puts_info "Note, that post_run_hook..."
    (file "/openlane/flow.tcl" line 412)
make[1]: *** [Makefile:43: user_project_wrapper] Error 1
make[1]: Leaving directory '/home/jure/Projekti/rvj1-caravel-soc/openlane'
make: *** [Makefile:72: user_project_wrapper] Error 2

```

### Environment
This github repo https://github.com/jurevreca12/rvj1-caravel-soc/tree/develop contains the relevant  code. To reproduce the issue run: make user_project_wrapper. Beware that the relevant code is in the develop branch, and not the main branch.

### Reproduction Material
I am also attaching a new openroad reproducible as a zip file (openroad_issue_reproducible.zip) in this issue.

### Expected behavior
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8755398/openroad_issue_reproducible.zip)

A successfully generated final gds of the design.

### Logs
None

[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8754599/openroad_issue_reproducible.zip)

vijayank88, 2022-05-23:
@maliberty  While running with latest OR App getting exited by tool with below message:
```
openroad> source run.tcl
[WARNING STA-0163] pdk/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib line 23, default_fanout_load is 0.0.
[INFO ODB-0222] Reading LEF file: ./tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 444 library cells
[INFO ODB-0226] Finished LEF file:  ./tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: ./in.def
[INFO ODB-0128] Design: user_project_wrapper
[INFO ODB-0130]     Created 637 pins.
[INFO ODB-0131]     Created 4 components and 722 component-terminals.
[INFO ODB-0133]     Created 904 nets and 623 connections.
[INFO ODB-0134] Finished DEF file: ./in.def
[WARNING PDN-1024] -verbose has been deprecated.
[WARNING PDN-1000] Using legacy PDNGEN.
Consider using "convert_pdn_config openlane/scripts/openroad/pdn_cfg.tcl" to convert the legacy configuration.
[INFO PDN-9008] Design name is user_project_wrapper.
[INFO PDN-9009] Reading technology data.
[ERROR PDN-9165] Conflict found, instance dram_inst is part of two grid definitions (CORE_macro_grid_2, CORE_macro_grid_1).
PDN-9165
```

maliberty, 2022-05-23:
[WARNING PDN-1000] Using legacy PDNGEN.

You should move to the new pdngen as we are no longer enhancing the legacy one.  Are you on an old version of OL?

jurevreca12, 2022-05-23:
My project is based upon the caravel_user_project mpw-6c(https://github.com/efabless/caravel_user_project/releases/tag/mpw-6c). Which is the latest tagged version. In there, the makefile defines the OL version as: **OPENLANE_TAG=2022.02.23_02.50.41**. 
So I am not sure, if this is an old version, but since its in the template, I guessed it should work.

maliberty, 2022-05-23:
That is the old pdngen as 6c is using a tag from February.

In any case you can't have a macro in two different grids.  What are you trying to accomplish?

jurevreca12, 2022-05-23:
I have very limited experience with physical design, so I am not sure what "two different grids" mean. I am simply trying to create a layout for my design.

vijayank88, 2022-05-24:
@maliberty  @jurevreca12 
With latest OpenLane tag: [2022.05.18_02.12.32](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.05.18_02.12.32) OpenROAD has updated PDNGEN.

With above OpenLane tag its passed PDN stage for your repo: https://github.com/jurevreca12/rvj1-caravel-soc/tree/develop

Update tool_metadata.yml for OpenROAD commit: https://github.com/The-OpenROAD-Project/OpenLane/blob/2022.05.18_02.12.32/dependencies/tool_metadata.yml and run the flow.

But during detailed routing its facing error again.
```
[STEP 15]
[INFO]: Running Detailed Routing...
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/rvj1-caravel_soc/runs/RUN_2022.05.24_04.56.33/logs/routing/15-detailed.log
[ERROR]: Last 10 lines:
Units:                1000
Number of layers:     13
Number of macros:     444
Number of vias:       25
Number of viarulegen: 25

[INFO DRT-0150] Reading design.
[ERROR DRT-0305] Net zero_ of signal type GROUND is not routable by TritonRoute. Move to special nets.
Error: droute.tcl, 46 DRT-0305
child process exited abnormally

```

vijayank88, 2022-05-24:
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8759863/openroad_issue_reproducible.zip)

jurevreca12, 2022-05-24:
Thanks for the answer @vijayank88. This does indeed fix this issue. I also get the routing error, but that is a separate issue.

closed 2022-05-24 by jurevreca12