The-OpenROAD-Project/OpenLane #1110: Issue launching "make test" to run OpenLane

ishfaqahmed29, 2022-05-25:
I'm getting the following error message. Do I need to reinstall Docker? 

[image]

Thanks

vijayank88, 2022-05-26:
@ishfaqahmed29 
From 2nd screenshot `make pdk` not completed. Try with `sudo make pdk`

Then try `sudo make test` or an other way try following command:
```
make mount
./flow.tcl -design spm
```

ishfaqahmed29, 2022-05-26:
Hi @vijayank88 

```
Linux, 20.04.1-Ubuntu: Python 3.8.10: Docker version 20.10.14, build a224086349
```

I tried <mark>sudo make pdk</mark>, and then <mark>sudo make test</mark> 

[image]

vijayank88, 2022-05-26:
@ishfaqahmed29 
You have permission issue.

Just install `OpenLane` where user have `read/write` access.

After successfull installation try
```
make mount
./flow.tcl -design spm
```

closed 2022-05-26 by ishfaqahmed29