The-OpenROAD-Project/OpenLane #1111: SRAM macro rotation (ERROR: Guide is not connected to design)

jurevreca12, 2022-05-26:
### Description
I am having trouble rotating a sky130 SRAM macro. When I have all macros set to N (I am guessing this means north) routing finishes successfully. However, when I rotate the macro I get the following error:

```
[STEP 15]
[INFO]: Running Detailed Routing...
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: user_project_wrapper/runs/user_project_wrapper/logs/routing/15-detailed.log
[ERROR]: Last 10 lines:
[INFO DRT-0028]   Complete met5.
[INFO DRT-1000] Pin iram_inst/din0[27] not in any guide. Attempting to patch guides to cover (at least part of) the pin.
[WARNING DRT-1001] No guide in the pin neighborhood
[WARNING DRT-0215] Pin iram_inst/din0[27] not covered by guide.
[WARNING DRT-0215] Pin iram_inst/din0[27] not covered by guide.
[WARNING DRT-0225] iram_din0\[27\] 2 pin not visited, fall back to feedthrough mode.
[WARNING DRT-0224] iram_din0\[27\] 2 pin not visited, number of guides = 8.
[ERROR DRT-0218] Guide is not connected to design.
Error: droute.tcl, 46 DRT-0218
child process exited abnormally
```

I rotate the macro by setting the configuration macro.cfg to S as such:
```
rvj1_soc          1162  1552 N                                                                                          
iram_inst         1100  800  S                                                                                          
dram_inst         1100  2400 N                                                                                          
uart_inst         500   1500 N              
```         

### Environment
```
git clone https://github.com/jurevreca12/rvj1-caravel-soc/tree/6338bc687190d042a6a00f40ccaa7c07ffdfe7d7.git
make uncompress
make user_project_wrapper
```

### Reproduction Material
I am adding a reproducible as a zip file to this issue.

### Expected behavior
Expected to successfully complete routing of the design.

### Logs
Full log:
```
make user_project_wrapper 
export CARAVEL_ROOT=/home/jure/Projekti/rvj1-caravel-soc/caravel && cd openlane && make user_project_wrapper
make[1]: Entering directory '/home/jure/Projekti/rvj1-caravel-soc/openlane'
Makefile:43: warning: undefined variable 'MCW_ROOT'
Makefile:43: warning: undefined variable 'MCW_ROOT'
Makefile:43: warning: undefined variable 'MCW_ROOT'
Makefile:43: warning: undefined variable 'MCW_ROOT'
Makefile:43: warning: undefined variable 'MCW_ROOT'
Makefile:43: warning: undefined variable 'MCW_ROOT'
###############################################
OpenLane 0dc6fb79c91082b94f8ded78d70f8bacbab96bf2
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using design configuration at /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/config.tcl
[INFO]: Sourcing Configurations from /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/config.tcl
[INFO]: PDKs root directory: /home/jure/Projekti/PDK
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/jure/Projekti/PDK/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/config.tcl
[WARNING]: Removing existing run at /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper...
[INFO]: Current run directory is /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper
[... 100 lines cut ...]
    (procedure "detailed_routing_tritonroute" line 6)
    invoked from within
"detailed_routing_tritonroute"
    (procedure "detailed_routing" line 17)
    invoked from within
"detailed_routing"
    (procedure "run_routing" line 76)
    invoked from within
"run_routing"
    (procedure "run_routing_step" line 7)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 55)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	if { [info exists arg_values(-file)] } {
		run_file [file normalize $a..."
    (file "/openlane/flow.tcl" line 434)
make[1]: *** [Makefile:43: user_project_wrapper] Error 1
make[1]: Leaving directory '/home/jure/Projekti/rvj1-caravel-soc/openlane'
make: *** [Makefile:72: user_project_wrapper] Error 2
```
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8777337/openroad_issue_reproducible.zip)

jurevreca12, 2022-05-26:
Without flipping the macro, or in other words, with the following macro.cfg:
```
rvj1_soc          1162  1552 N                                                                                          
iram_inst         1100  800  N                                                                                          
dram_inst         1100  2400 N                                                                                          
uart_inst         500   1500 N
```
I get the following result from running make user_project_wrapper:
```
make user_project_wrapper 
export CARAVEL_ROOT=/home/jure/Projekti/rvj1-caravel-soc/caravel && cd openlane && make user_project_wrapper
make[1]: Entering directory '/home/jure/Projekti/rvj1-caravel-soc/openlane'
Makefile:43: warning: undefined variable 'MCW_ROOT'
Makefile:43: warning: undefined variable 'MCW_ROOT'
Makefile:43: warning: undefined variable 'MCW_ROOT'
Makefile:43: warning: undefined variable 'MCW_ROOT'
Makefile:43: warning: undefined variable 'MCW_ROOT'
Makefile:43: warning: undefined variable 'MCW_ROOT'
###############################################
OpenLane 0dc6fb79c91082b94f8ded78d70f8bacbab96bf2
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using design configuration at /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/config.tcl
[INFO]: Sourcing Configurations from /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/config.tcl
[INFO]: PDKs root directory: /home/jure/Projekti/PDK
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/jure/Projekti/PDK/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/config.tcl
[WARNING]: Removing existing run at /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper...
[INFO]: Current run directory is /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/runs/user_project_wrapper
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Looking for files defined in ::env(EXTRA_GDS_FILES)  /home/jure/Projekti/PDK/sky130A/libs.ref/sky130_sram_macros/gds/sky130_sram_2kbyte_1rw1r_32x512_8.gds  /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/../../gds/wbuart_wrap.gds  /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/../../gds/rvj1_caravel_soc.gds ...
[INFO]: /home/jure/Projekti/PDK/sky130A/libs.ref/sky130_sram_macros/gds/sky130_sram_2kbyte_1rw1r_32x512_8.gds exists.
[INFO]: /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/../../gds/wbuart_wrap.gds exists.
[INFO]: /home/jure/Projekti/rvj1-caravel-soc/openlane/user_project_wrapper/../../gds/rvj1_caravel_soc.gds exists.
[STEP 1]
[INFO]: Running Synthesis...
[STEP 2]
[INFO]: Running Single-Corner Static Timing Analysis...
[INFO]: Creating a netlist with power/ground pins.
[STEP 3]
[INFO]: Running Initial Floorplanning...
[INFO]: Setting Core Dimensions...
[STEP 4]
[INFO]: Running IO Placement...
[STEP 5]
[INFO]: Performing Manual Macro Placement...
[WARNING]: Skipping Tap/Decap Insertion.
[INFO]: Power planning with power {vccd1 vccd2 vdda1 vdda2} and ground {vssd1 vssd2 vssa1 vssa2}...
[STEP 6]
[INFO]: Generating PDN...
[STEP 7]
[INFO]: Performing Random Global Placement...
[INFO]: Skipping Placement Resizer Design Optimizations.
[STEP 8]
[INFO]: Running Detailed Placement...
[INFO]: Skipping Placement Resizer Timing Optimizations.
[INFO]: Routing...
[STEP 9]
[INFO]: Running Global Routing Resizer Timing Optimizations...
[STEP 10]
[INFO]: Writing Verilog...
[INFO]: Adding routing obstructions...
[WARNING]: Specifying a routing obstruction is now done using the coordinates
[WARNING]: of its bounding box instead of the now deprecated (x, y, size_x, size_y).
[INFO]: Obstructions added over li1  0     0    2920    3520,
                        met1 1100  800  1783.1  1216.54,
[... 67 lines cut ...]
    (procedure "try_catch" line 17)
    invoked from within
"try_catch $::env(OPENROAD_BIN) -python $::env(SCRIPTS_DIR)/tr_drc_to_klayout_drc.py  -i $::env(drc_prefix).tr  -o $::env(drc_prefix).klayout.xml  --de..."
    (procedure "run_magic_drc" line 20)
    invoked from within
"run_magic_drc"
    (procedure "run_drc_step" line 9)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 55)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	if { [info exists arg_values(-file)] } {
		run_file [file normalize $a..."
    (file "/openlane/flow.tcl" line 434)
make[1]: *** [Makefile:43: user_project_wrapper] Error 1
make[1]: Leaving directory '/home/jure/Projekti/rvj1-caravel-soc/openlane'
make: *** [Makefile:72: user_project_wrapper] Error 2
```

Unfortunately, because of the way openlane fails, the reproducible is not generated. I am attaching a git commit that can be used to reproduce the issue though: https://github.com/jurevreca12/rvj1-caravel-soc/tree/c297cdaefa119f267ad54b40c840a4373adfbb6c

The commit is identical to the original commit posted in the issue, with the exception of the macro.cfg change.

vijayank88, 2022-05-26:
@jurevreca12 
I have cloned your repo and running it.
Magic DRC failed with child killed signal, is out of memory kill in your end.

vijayank88, 2022-05-26:
@jurevreca12 
Its failed with Magic DRC error at my end.
```
[INFO]: Running Magic DRC...
[INFO]: Converting Magic DRC Violations to Magic Readable Format...
[INFO]: Converting Magic DRC Violations to Klayout XML Database...
[INFO]: Converting TritonRoute DRC Violations to Klayout XML Database...
[INFO]: Converting DRC Violations to RDB Format...
[ERROR]: There are violations in the design after Magic DRC.
[ERROR]: Total Number of violations is 11146789
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'runs/RUN_2022.05.26_09.33.59/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'runs/RUN_2022.05.26_09.33.59/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
````

jurevreca12, 2022-05-26:
Yes. When I ran that on another computer, with more memory, the same thing happens. And this DRC error happens right after my error, so its basically the same error.

vijayank88, 2022-05-26:
@maliberty FYI

Stephanommg, 2022-05-31:
I started attending to the problem initially mentioned (guide is not connected to the design). Is this still relevant?

Stephanommg, 2022-05-31:
I investigated the "guide is not connected to the design" problem and it really is not a DR problem. There is no guide even close to pin iram_inst/din0[27], so that the router could create a patch guide to connect to the pin.

@jurevreca12 are you rotation the macro and performing the entire place and route or are you rotating it after global routing result?

jurevreca12, 2022-05-31:
@Stephanommg yes it is still relevant.

Stephanommg, 2022-05-31:
So this is a GR problem. @maliberty do you have anyone to look at GR?

Stephanommg, 2022-06-07:
A fix was implemented and it will go through more testing.