The-OpenROAD-Project/OpenLane #1112: Add clock tree metal layer constraints

antonblanchard, 2022-05-26:
@jjcherry56 points out that Openlane is not constraining the clock tree metal layers. Depending on the process, we might want to use upper/larger metal layers.

 We need a few changes:

1. Environment variables in open_pdks for the clock min/max layers. open_pdks does have:

```
set ::env(CLOCK_WIRE_RC_LAYER) "met5"
```

However met5 seems a bit of an odd setting. We really want variables for min and max layers (eg met3-met5). `CLOCK_MIN_LAYER/CLOCK_MAX_LAYER` perhaps?

2. Telling global routing the layers to use (remembering to cap the max layer if we are taping out a macro):

```
set_routing_layers -clock $::env(CLOCK_MIN_LAYER)-$::env(CLOCK_MAX_LAYER)
```

3. Telling STA what layer to use for RC estimates:

```
set_wire_rc -clock -layer $::env(CLOCK_MIN_LAYER)
```

donn, 2022-09-01:
`RT_CLOCK_MIN_LAYER` and `RT_CLOCK_MAX_LAYER` exist?

donn, 2023-04-13:
Staled out.

closed 2023-04-13 by donn