The-OpenROAD-Project/OpenLane #1117: Run several verilog code

ashkanr65, 2022-05-31:
Hi,
I was wondering how I can run a project with several verilog file with OpenLane?
For example this project:
https://github.com/alexforencich/verilog-i2c
have 9 verilog files which should run with each other.
How can i do that?

mousaq92, 2022-05-31:
Hello,

Try this:

set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v]

OR this:

set ::env(VERILOG_FILES) "[glob $::env(DESIGN_DIR)/src/file1.v] \
                                         [glob $::env(DESIGN_DIR)/src/file2.v] \
                                         [glob $::env(DESIGN_DIR)/src/file3.v] "

maliberty, 2022-05-31:
There is no need to use glob if you are giving a path without wildcards (your second option).

ashkanr65, 2022-06-01:
Thanks for your answers, but when I run it, I receive several errors:(copy-paste from errors.log)

[ERROR]: during executing: "yosys -c /openlane/scripts/yosys/synth.tcl -l /openlane/designs/verilog-i2c-master/runs/first_run/logs/synthesis/1-synthesis.log |& tee /dev/null"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
Successfully finished Verilog frontend.

8. Executing Verilog-2005 frontend: /openlane/designs/verilog-i2c-master/rtl/i2c_slave_axil_master.v
Parsing SystemVerilog input from `/openlane/designs/verilog-i2c-master/rtl/i2c_slave_axil_master.v' to AST representation.
Generating RTLIL representation for module `\i2c_slave_axil_master'.
Successfully finished Verilog frontend.

9. Executing Verilog-2005 frontend: /openlane/designs/verilog-i2c-master/rtl/i2c_init.v
ERROR: No such module: verilog-i2c-master
child process exited abnormally

[ERROR]: Flow failed.