The-OpenROAD-Project/OpenLane #1118: Syntax Error in PDKs

Eswar-Akula, 2022-06-01:
Hello Team,
Our team encountered syntax error while executing GLS on the verilog file using "FUNCTIONAL" directive in the path **OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v** in line number 49162. We should be commenting this  SKY130_FD_SC_HD__LPFLOW_BLEEDER_FUNCTIONAL_V in that line then its giving accurate result. 

[image]

So, we are reporting this to bring to your notice.

Thanks,
Eswar

donn, 2022-06-02:
@RTimothyEdwards @mithro Wasn't a fix merged for this a while back?

donn, 2022-06-02:
Oh, also, please do not ignore the issue template. Without knowing what version of the PDK you're using, we *cannot* help you.

closed 2022-06-02 by donn