The-OpenROAD-Project/OpenLane #1120: make test does not work correctly / .flow.tcl does not work correctly

OmurCeran, 2022-06-02:
Hello,
When I make `sudo make test` at linux mint 20.3 after installing openlane, I gave error at below.
[image]

How can I solve this?

vijayank88, 2022-06-03:
@OmurCeran 
Try following way:
```
make mount
./flow.tcl -design spm
```

The above executed by short `make test`

OmurCeran, 2022-06-03:
I tried it as you said but it is same also.
[image]

It did not solve problem actually.

vijayank88, 2022-06-03:
@OmurCeran 
Execute below command and post here:

```
Please run the following set of commands in the OpenLane folder:

make survey || python3 ./env.py issue-survey

And copy and paste the ENTIRE output between the triple-backticks. Please do not gzip and upload the output.

If neither command succeeds, you are using an out of date version of OpenLane and should probably update. 
```

vijayank88, 2022-06-03:
From the  screenshot Openlane version shows rc7 looks very old.

https://github.com/The-OpenROAD-Project/OpenLane#prerequisites do a fresh installation in new path and let us know

OmurCeran, 2022-06-03:
When I make `make`, it gives me error as shown first figure.
[image]
However also docker was installed before.
You can see it at second figure.
[image]

vijayank88, 2022-06-03:
@OmurCeran 
`Container Engine: UNKNOWN` from your survey
Try to remove previous docker and re-install latest version.

OmurCeran, 2022-06-03:
I tried to install docker by using https://docs.docker.com/engine/install/ubuntu/ this. But it gives me error at below.
[image]

vijayank88, 2022-06-03:
@OmurCeran 
Follow this instruction : https://computingforgeeks.com/install-docker-and-docker-compose-on-linux-mint-19/

vijayank88, 2022-06-03:
Inside docker magic already installed. Just use `magic <path to>/spm*.mag

vijayank88, 2022-06-03:
so spm test completes successfully. Installation working fine, so closing this issue. For other issue, open a new issue.

closed 2022-06-03 by vijayank88