The-OpenROAD-Project/OpenLane #1121: Syntax Error in PDKs

Eswar-Akula, 2022-06-03:
### Description

Our team encountered syntax error while executing GLS on the verilog file using "FUNCTIONAL" directive in the path OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v in line number 49162. We should be commenting this SKY130_FD_SC_HD__LPFLOW_BLEEDER_FUNCTIONAL_V in that line then its giving accurate result.

### Environment
```

Kernel: Linux v3.10.0-1160.59.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 0c611a3422f0a9691c10776f951efd32887e85e5
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

0c611a3 2022-05-09T08:39:57-07:00 Update to use the new C++ pdngen in OR (#1059) - Matt Liberty -  (HEAD -> master, tag: 2022.05.11_01.53.47, origin/master, origin/HEAD)
47c9f22 2022-05-06T14:32:57+02:00 Remove Hard-coded Paths in Sample Designs (#1083) - Balint Cristian -  (tag: 2022.05.07_01.50.30)
0c2d66a 2022-05-04T11:11:24+02:00 Global Placement Fixes (#1039) - Anton Blanchard -  (tag: 2022.05.05_02.01.30)

```
PDKs version:- open_pdks 41c0908b47130d5675ff8484255b43f66463a7d6

### Expected behavior
 We should be commenting this SKY130_FD_SC_HD__LPFLOW_BLEEDER_FUNCTIONAL_V in line number 49162, then its giving accurate result.

[image]

### Logs
```
We are getting syntax error at OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v in line number 49162
```

Thanks,
Eswar

vijayank88, 2022-06-03:
@RTimothyEdwards 
FYI

donn, 2022-06-03:
@vijayank88 We thought we fixed this a long time ago, tbh

vijayank88, 2022-06-04:
@donn 
I checked that file. That particular line was not commented.

kareefardi, 2024-07-17:
Fixed

closed 2024-07-17 by kareefardi