The-OpenROAD-Project/OpenLane #1122: [ERROR GRT-0080] Invalid pin placement. Error: groute.tcl, 69 GRT-0080 [ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl when executig the first make

Ricardo-Ren, 2022-06-03:
### Description
A clear and concise description of what the bug is.
when executing make ***
[ERROR GRT-0080] Invalid pin placement. Error: groute.tcl, 69 GRT-0080 [ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl when executig the first make
### Environment
```
Please run the following set of commands in the OpenLane folder:

make survey || python3 ./env.py issue-survey
Kernel: Linux v5.4.0-100-generic
Distribution: ubuntu 20.04
Python: v3.7.6 (OK)
Container Engine: docker v19.3.6 (BELOW MINIMUM: UPDATE DOCKER)
OpenLane Git Version: -dev
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)
And copy and paste the ENTIRE output between the triple-backticks. Please do not gzip and upload the output.

If neither command succeeds, you are using an out of date version of OpenLane and should probably update. 
```

### Reproduction Material
* Upload a tarball containing the relevant design.
* List the commands used to run the design.

If you see a message like `Reproducible pac
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8830125/openroad_issue_reproducible.zip)
kaged: Please tarball and upload <PATH> if you're going to submit an issue` in your logs, please also tarball and include that path. This will greatly speed up the fixing process.

### Expected behavior
A clear and concise description of what you expected to happen.

### Logs
```
Add any relevant logs here. Please do ensure they're enclosed by the triple-backticks.
```

vijayank88, 2022-06-03:
@Ricardo-Ren 
Check this slack channel conversation: https://skywater-pdk.slack.com/archives/C016UL7AQ73/p1654236878248299?thread_ts=1654077013.941219&cid=C016UL7AQ73
I checked your repo. Update following in basic marco config.tcl and try:
```
set ::env(FP_SIZING) absolute
set ::env(DIE_AREA) "0 0 600 600"

set ::env(FP_PIN_ORDER_CFG) $script_dir/pin_order.cfg

set ::env(PL_BASIC_PLACEMENT) 1
```

vijayank88, 2022-06-03:
also update your docker version as per https://github.com/The-OpenROAD-Project/OpenLane#prerequisites

Ricardo-Ren, 2022-06-03:
Thank you , but met another ERROR 
XOR CHECK FILE NOT FOUND in Projects/caravel_tutorial/caravel_example/precheck_results/03_JUN_2022___07_54_50/logs/xor_check.total
{{XOR CHECK FAILED}} The GDS file has non-conforming geometries.

vijayank88, 2022-06-03:
@Ricardo-Ren 
`XOR_CHECK_FAILED` related to `precheck`. Its not related to OpenLane flow.
Follow slack channel discussion to fix `precheck` issues.
Closing this issue.

closed 2022-06-03 by vijayank88