The-OpenROAD-Project/OpenLane #1125: Fail at

minghungumich, 2022-06-05:
### Description
A clear and concise description of what the bug is.

### Environment
```
Kernel: Linux v3.10.0-1160.53.1.el7.x86_64
Distribution: centos 7
Python: v3.8.13 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 67e08299d4babfd672a86978b1a29b31efc973e9
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

67e0829 2022-06-05T03:59:26+02:00 Update OpenROAD (#1123) - Mohamed Gaber -  (HEAD -> master, origin/master, origin/HEAD)
1f1508e 2022-06-04T19:19:10+02:00 Revert Power Ring Order (#1109) - Dinesh Annayya -  (tag: 2022.06.05_01.39.14)
01e9510 2022-06-02T00:06:03+02:00 Fix tag names on Jenkins (#1119) - Vitor Bandeira -  (tag: 2022.06.02_01.56.32)
```

### Reproduction Material
[Pulpino Testcase]()

### Expected behavior
This design setup should pass the detailed placement. Already succeed in ORFS flow but with bad timing.

### Logs
In `openlane.log`
```
[INFO]: Changing layout to 'designs/pulpino/runs/v1/tmp/placement/8-global.def'...
[INFO]: Running Placement Resizer Design Optimizations...
[INFO]: Executing OpenROAD with script 'scripts/openroad/resizer.tcl'...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/pulpino/runs/v1/logs/placement/9-resizer.log
[ERROR]: Last 10 lines:
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_1__31_
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_29__16_
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_2__1_
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_12__29_
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_12__29_
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_16__9_
[INFO DPL-0035] message limit reached, this message will no longer print
[ERROR DPL-0036] Detailed placement failed.
Error: resizer.tcl, 79 DPL-0036
child process exited abnormally
```
In `9-resizer.log`
```
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_8__10_
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_11__3_
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_27__3_
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_1__31_
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_29__16_
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_2__1_
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_12__29_
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_12__29_
[INFO DPL-0035]  u_core/core_region_i_CORE_RISCV_CORE_id_stage_i_registers_i/mem_reg_16__9_
[INFO DPL-0035] message limit reached, this message will no longer print
[ERROR DPL-0036] Detailed placement failed.
Error: resizer.tcl, 79 DPL-0036
```

minghungumich, 2022-06-05:
@mithro @msaligane I am facing an issue that detail placement does not work for my design.

vijayank88, 2022-06-06:
@minghungumich  
can you share `designs/pulpino/runs/v1/config.tcl`?

minghungumich, 2022-06-06:
@vijayank88 This is my `designs/pulpino/runs/v1/config.tcl` file
```
# Run configs
set ::env(PDK_ROOT) {/home/kevinchen/OpenLane/pdks}
set ::env(BASE_SDC_FILE) {/openlane/designs/pulpino/base.sdc}
set ::env(BOTTOM_MARGIN_MULT) {4}
set ::env(CACHED_NETLIST) {/openlane/designs/pulpino/src/pulpino_top.v}
set ::env(CARRY_SELECT_ADDER_MAP) {/home/kevinchen/OpenLane/pdks/sky130A/libs.tech/openlane/sky130_fd_sc_hd/csa_map.v}
set ::env(CELLS_LEF) {/home/kevinchen/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lef/sky130_fd_sc_hd.lef}
set ::env(CELLS_LEF_OPT) {/home/kevinchen/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lef/sky130_fd_sc_hd.lef}
set ::env(CELL_CLK_PORT) {CLK}
set ::env(CELL_PAD) {0}
set ::env(CELL_PAD_EXCLUDE) {sky130_fd_sc_hd__tap* sky130_fd_sc_hd__decap* sky130_ef_sc_hd__decap* sky130_fd_sc_hd__fill*}
set ::env(CHECK_ASSIGN_STATEMENTS) {0}
set ::env(CHECK_UNMAPPED_CELLS) {1}
set ::env(CLK_BUFFER) {sky130_fd_sc_hd__clkbuf_4}
set ::env(CLK_BUFFER_INPUT) {A}
set ::env(CLK_BUFFER_OUTPUT) {X}
set ::env(CLOCK_BUFFER_FANOUT) {16}
set ::env(CLOCK_PERIOD) {10}
set ::env(CLOCK_PORT) {clk}
set ::env(CLOCK_TREE_SYNTH) {0}
set ::env(CLOCK_WIRE_RC_LAYER) {met5}
set ::env(CONFIGS) {/openlane/configuration/checkers.tcl /openlane/configuration/cts.tcl /openlane/configuration/extraction.tcl /openlane/configuration/general.tcl /openlane/configuration/lvs.tcl /openlane/configuration/routing.tcl /openlane/configuration/synthesis.tcl /openlane/configuration/floorplan.tcl /openlane/configuration/placement.tcl}
set ::env(CORE_AREA) {60 60 3510 3510}
set ::env(CTS_CLK_BUFFER_LIST) {sky130_fd_sc_hd__clkbuf_8 sky130_fd_sc_hd__clkbuf_4 sky130_fd_sc_hd__clkbuf_2}
set ::env(CTS_CLK_MAX_WIRE_LENGTH) {0}
set ::env(CTS_DISABLE_POST_PROCESSING) {0}
set ::env(CTS_DISTANCE_BETWEEN_BUFFERS) {0}
set ::env(CTS_MAX_CAP) {1.53169}
set ::env(CTS_REPORT_TIMING) {1}
set ::env(CTS_ROOT_BUFFER) {sky130_fd_sc_hd__clkbuf_16}
set ::env(CTS_SINK_CLUSTERING_MAX_DIAMETER) {50}
set ::env(CTS_SINK_CLUSTERING_SIZE) {25}
set ::env(CTS_SQR_CAP) {0.258e-3}
set ::env(CTS_SQR_RES) {0.125}
set ::env(CTS_TARGET_SKEW) {200}
set ::env(CTS_TECH_DIR) {N/A}
set ::env(CTS_TOLERANCE) {100}
set ::env(DATA_WIRE_RC_LAYER) {met2}
set ::env(DECAP_CELL) {sky130_ef_sc_hd__decap_12 sky130_fd_sc_hd__decap_8 sky130_fd_sc_hd__decap_6 sky130_fd_sc_hd__decap_4 sky130_fd_sc_hd__decap_3}
set ::env(DEFAULT_MAX_TRAN) {0.75}
set ::env(DEF_UNITS_PER_MICRON) {1000}
set ::env(DESIGN_CONFIG) {/openlane/designs/pulpino/config.tcl}
set ::env(DESIGN_DIR) {/openlane/designs/pulpino}
set ::env(DESIGN_IS_CORE) {1}
set ::env(DESIGN_NAME) {pulpino_top}
set ::env(DETAILED_ROUTER) {tritonroute}
set ::env(DIE_AREA) {0 0 3570 3570}
set ::env(DIODE_CELL) {sky130_fd_sc_hd__diode_2}
set ::env(DIODE_CELL_PIN) {DIODE}
set ::env(DIODE_INSERTION_STRATEGY) {3}
set ::env(DIODE_PADDING) {2}
set ::env(DRC_EXCLUDE_CELL_LIST) {/home/kevinchen/OpenLane/pdks/sky130A/libs.tech/openlane/sky130_fd_sc_hd/drc_exclude.cells}
set ::env(DRC_EXCLUDE_CELL_LIST_OPT) {/home/kevinchen/OpenLane/pdks/sky130A/libs.tech/openlane/sky130_fd_sc_hd/drc_exclude.cells}
set ::env(DRT_OPT_ITERS) {64}
set ::env(ECO_ENABLE) {1}
set ::env(ECO_FINISH) {0}
set ::env(ECO_ITER) {0}
set ::env(ECO_SKIP_PIN) {1}
set ::env(ECO_STARTED) {0}
set ::env(EXTRA_GDS_FILES) {/openlane/designs/pulpino/macros/gds/sky130_sram_2kbyte_1rw_32x512_8.gds}
set ::env(EXTRA_LEFS) {/openlane/designs/pulpino/macros/lef/sky130_sram_2kbyte_1rw_32x512_8.lef}
set ::env(EXTRA_LIBS) {/openlane/designs/pulpino/macros/lib/sky130_sram_2kbyte_1rw_32x512_8_TT_1p8V_25C.lib}
set ::env(FAKEDIODE_CELL) {sky130_ef_sc_hd__fakediode_2}
set ::env(FILL_CELL) {sky130_fd_sc_hd__fill*}
set ::env(FILL_INSERTION) {1}
set ::env(FP_ASPECT_RATIO) {1}
set ::env(FP_CORE_MARGIN) {0}
set ::env(FP_CORE_UTIL) {50}
[... 245 lines cut ...]
set ::env(floorplan_tmpfiles) {/openlane/designs/pulpino/runs/v1/tmp/floorplan}
set ::env(placement_logs) {/openlane/designs/pulpino/runs/v1/logs/placement}
set ::env(placement_reports) {/openlane/designs/pulpino/runs/v1/reports/placement}
set ::env(placement_results) {/openlane/designs/pulpino/runs/v1/results/placement}
set ::env(placement_tmpfiles) {/openlane/designs/pulpino/runs/v1/tmp/placement}
set ::env(routing_logs) {/openlane/designs/pulpino/runs/v1/logs/routing}
set ::env(routing_reports) {/openlane/designs/pulpino/runs/v1/reports/routing}
set ::env(routing_results) {/openlane/designs/pulpino/runs/v1/results/routing}
set ::env(routing_tmpfiles) {/openlane/designs/pulpino/runs/v1/tmp/routing}
set ::env(signoff_logs) {/openlane/designs/pulpino/runs/v1/logs/signoff}
set ::env(signoff_reports) {/openlane/designs/pulpino/runs/v1/reports/signoff}
set ::env(signoff_results) {/openlane/designs/pulpino/runs/v1/results/signoff}
set ::env(signoff_tmpfiles) {/openlane/designs/pulpino/runs/v1/tmp/signoff}
set ::env(synthesis_logs) {/openlane/designs/pulpino/runs/v1/logs/synthesis}
set ::env(synthesis_reports) {/openlane/designs/pulpino/runs/v1/reports/synthesis}
set ::env(synthesis_results) {/openlane/designs/pulpino/runs/v1/results/synthesis}
set ::env(synthesis_tmpfiles) {/openlane/designs/pulpino/runs/v1/tmp/synthesis}
set ::env(SYNTH_MAX_TRAN) {0.75}
set ::env(CURRENT_DEF) {0}
set ::env(CURRENT_GUIDE) {0}
set ::env(CURRENT_INDEX) 2
set ::env(CURRENT_NETLIST) /openlane/designs/pulpino/runs/v1/results/synthesis/pulpino_top.v
set ::env(PREV_NETLIST) 0

```

minghungumich, 2022-06-06:
And this is my `config.tcl` file.
```
# User config
set ::env(DESIGN_NAME) pulpino_top

set ::env(FP_SIZING) absolute
set ::env(DIE_AREA) "0 0 3570 3570"
set ::env(CORE_AREA) "60 60 3510 3510"

set ::env(MACRO_PLACEMENT_CFG) $::env(DESIGN_DIR)/macro_placement.cfg
set ::env(BASE_SDC_FILE) $::env(DESIGN_DIR)/base.sdc

set ::env(FP_PDN_CORE_RING) 1
set ::env(FP_PDN_CORE_RING_VWIDTH) 3.1
set ::env(FP_PDN_CORE_RING_HWIDTH) 3.1
set ::env(FP_PDN_CORE_RING_VOFFSET) 14
set ::env(FP_PDN_CORE_RING_HOFFSET) $::env(FP_PDN_CORE_RING_VOFFSET)
set ::env(FP_PDN_CORE_RING_VSPACING) 1.7
set ::env(FP_PDN_CORE_RING_HSPACING) $::env(FP_PDN_CORE_RING_VSPACING)

set ::env(FP_PDN_VWIDTH) 3.1
set ::env(FP_PDN_HWIDTH) 3.1
set ::env(FP_PDN_VSPACING) [expr 5*$::env(FP_PDN_CORE_RING_VWIDTH)]
set ::env(FP_PDN_HSPACING) [expr 5*$::env(FP_PDN_CORE_RING_HWIDTH)]

set ::env(VDD_NETS) [list {vccd1}]
set ::env(GND_NETS) [list {vssd1}]
# set ::env(SYNTH_USE_PG_PINS_DEFINES) "USE_POWER_PINS"

# PDN Pitch
set ::env(FP_PDN_VPITCH) 180
set ::env(FP_PDN_HPITCH) $::env(FP_PDN_VPITCH)

# PDN Offset
set ::env(FP_PDN_VOFFSET) 5
set ::env(FP_PDN_HOFFSET) $::env(FP_PDN_VOFFSET)

set ::env(PL_TARGET_DENSITY) 0.35
set ::env(PL_BASIC_PLACEMENT) 1
set ::env(CELL_PAD) 0

set ::env(CLOCK_PERIOD) "10"
set ::env(CLOCK_PORT) "clk"
set ::env(CLOCK_TREE_SYNTH) 0

# Change if needed
set ::env(CACHED_NETLIST) "$::env(DESIGN_DIR)/src/$::env(DESIGN_NAME).v"
# set ::env(VERILOG_FILES) "$::env(DESIGN_DIR)/src/$::env(DESIGN_NAME).v"
# set ::env(VERILOG_FILES_BLACKBOX) "$::env(DESIGN_DIR)/src/sky130_sram_2kbyte_1rw_32x512_8.v"

set ::env(EXTRA_LIBS) [glob $::env(DESIGN_DIR)/macros/lib/*.lib]
set ::env(EXTRA_LEFS) [glob $::env(DESIGN_DIR)/macros/lef/*.lef]
set ::env(EXTRA_GDS_FILES) [glob $::env(DESIGN_DIR)/macros/gds/*.gds]

set ::env(ECO_ENABLE) {1}

set filename $::env(DESIGN_DIR)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
	source $filename
}
```

vijayank88, 2022-06-06:
@minghungumich 
Can you try to change FP_CORE_UTIL to 30 and try it out?

vijayank88, 2022-06-06:
Add this variable if you're using auto macro placement.
```
set ::env(PL_MACRO_CHANNEL) {20 20}
set ::env(PL_MACRO_HALO) {10 10}
```

minghungumich, 2022-06-06:
@vijayank88 Do I need to change `FP_CORE_UTIL` if `DIE_AREA` is specified? I see there is two mode in ifp [here](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/openroad/floorplan.tcl#L58). I think I accidentally use the `PL_BASIC_PLACEMENT = 1` setting and cause problem.

minghungumich, 2022-06-06:
Thanks! Now it passes the detailed placement.

closed 2022-06-06 by minghungumich