The-OpenROAD-Project/OpenLane #1126: Rocket Mini to GDSII failed at step 15

Ryabkomv, 2022-06-06:
### Description
I have successfully generated GDSII file with open lane for picorv32 with configuration using docker container.
However when I try to get GDSII for Rocket chip with Tiny configuration I got error at step 15. 

```
[STEP 15]
[INFO]: Running Global Routing Resizer Timing Optimizations...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/ExampleRocketSystem/runs/RUN_2022.06.03_12.54.37/logs/routing/15-resizer.log
[ERROR]: Last 10 lines:
met2       Vertical      6182445       4937214          20.14%
met3       Horizontal    4121630       3293440          20.09%
met4       Vertical      2472978       1650582          33.26%
met5       Horizontal     824326        410880          50.16%
---------------------------------------------------------------

[INFO GRT-0101] Running extra iterations to remove overflow.
[ERROR GRT-0169] Net _0432988_: Invalid index for position (2590950, 3605250). Net degree: 6.
Error: resizer_routing_timing.tcl, 53 GRT-0169
child process exited abnormally

```

Googling I came to the following config.tcl

```
set ::env(DESIGN_NAME) ExampleRocketSystem
set ::env(FP_CORE_UTIL) 30

set ::env(CELL_PAD) 2

set ::env(PL_BASIC_PLACEMENT) 0
set ::env(PL_TARGET_DENSITY) 0.3

set ::env(GLB_RT_ADJUSTMENT) 0.2

set ::env(PL_ROUTABILITY_DRIVEN) 1

# Change if needed
set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v]

# Fill this
set ::env(CLOCK_PERIOD) "10.0"
set ::env(CLOCK_PORT) "clk"

set filename $::env(DESIGN_DIR)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
        source $filename
}
```

### Environment
```
Please run the following set of commands in the OpenLane folder:

python3 ./env.py issue-survey
Kernel: Linux v5.4.0-113-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: UNKNOWN vUNKNOWN (UNSUPPORTED)
OpenLane Git Version: 2653e6e4d69c8a8270c494e2ce61cbfd151fae38
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

2653e6e 2022-05-25T19:27:29+02:00 Synthesis Parameters, More Report Metrics (#1105) - Arya Reais-Parsi -  (HEAD -> master, tag: 2022.05.26_01.40.55, origin/master, origin/HEAD)
4298bf2 2022-05-25T17:42:21+02:00 Remove `endcap_cpp` option from `tapcell` (#1085) - Anton Blanchard -  ()
601eede 2022-05-21T18:47:55+02:00 Support Multi-Arch Docker Builds (#1075) - Mohamed Gaber -  (tag: 2022.05.22_02.07.28)
```

### Reproduction Material
* Upload a tarball containing the relevant design.
* List the commands used to run the design.

If you see a message like `Reproducible packaged: Please tarball and upload <PATH> if you're going to submit an issue` in your logs, please also tarball and include that path. This will greatly speed up the fixing process.

### Expected behavior

Successful GDSII generation

vijayank88, 2022-06-06:
@Ryabkomv 
`openroad_issue_reproducible` attachment missing

vijayank88, 2022-06-06:
`Container Engine: UNKNOWN vUNKNOWN (UNSUPPORTED)` Also update docker version to latest one

@Donn is possible to print `issue-survey` with existing container version followed by `UNKNOWN` message. So its easy to convey the message to user.

Ryabkomv, 2022-06-06:
Added sources, and docker version

vijayank88, 2022-06-06:
@Ryabkomv 
Test case attached is not valid format.
`designs/ExampleRocketSystem/runs/RUN_2022.06.03_12.54.37/openroad_issue_reproducible` exist or not?

Ryabkomv, 2022-06-06:
> @Ryabkomv Test case attached is not valid format. `designs/ExampleRocketSystem/runs/RUN_2022.06.03_12.54.37/openroad_issue_reproducible` exist or not?
> 
> If not exist Please package a standalone test case with https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

Added corresponding folder as an archive at [github](https://github.com/Ryabkomv/openroad_issue_reproducible). It pretty lengthy.

vijayank88, 2022-06-06:
@Ryabkomv 
can you share me git repo link, where you found this project?

vijayank88, 2022-06-06:
If you're using default configuration try increase `FP_CORE_UTIL` to 30 and try

Ryabkomv, 2022-06-06:
> @Ryabkomv can you share me git repo link, where you found this project?

https://github.com/chipsalliance/rocket-chip
I used TinyConfig configuration

Ryabkomv, 2022-06-06:
> If you're using default configuration try increase `FP_CORE_UTIL` to 30 and try

Yes, I copied my config and  `FP_CORE_UTIL` is already set to 30

vijayank88, 2022-06-06:
Can you share src/*.v files?

Ryabkomv, 2022-06-06:
> Can you share src/*.v files?

These files are already attached to the post, can you download them?

vijayank88, 2022-06-06:
`CLOCK_PORT` is `clock` right?

Ryabkomv, 2022-06-06:
> set ::env(CLOCK_PORT) "clk"

Should be...
So you mean I need to change
`set ::env(CLOCK_PORT) "clk"`
for
`set ::env(CLOCK_PORT) "clock"`
right?

vijayank88, 2022-06-06:
yes. I am also running at my end with changes. Will update you the status.

Ryabkomv, 2022-06-06:
oh, looks like a stupid mistake...I will rerun the flow with updated config file.
By the way, Is there a way to run the flow step by step? Because the process is pretty time consuming, in case of failure at some stage I'm looking to make modifications in config.tcl and re-run just failed stage.

vijayank88, 2022-06-06:
Use `./flow.tcl -interactive`

vijayank88, 2022-06-07:
@maliberty  Can you please  look into this?

vijayank88, 2022-06-07:
@Ryabkomv Attach latest run test case

Ryabkomv, 2022-06-08:
Unfortunately got the issue at step 15.
updated test case at [git repository](https://github.com/Ryabkomv/openroad_issue_reproducible)

maliberty, 2022-06-08:
I am out this week, perhaps @luis201420 can help

vijayank88, 2022-06-15:
@luis201420 
do you had time to look into this issue?

vijayank88, 2022-06-15:
@donn 
Is there any configuration available to change `set_routing_alpha 0` in openlane flow?

@luis201420 
what is default setting for `set_routing_alpha`? In which scenario we need to pass this variable?

maliberty, 2022-06-15:
The alpha=0 is a workaround to a router problem that we are fixing.  You would need it when you hit this error.

Ryabkomv, 2022-06-16:
Thanks for suggestion.
I wonder if I need to set variable in config file? What is there name of the variable? Should it looks like:
`set ::env(set_routing_alpha) 0 ?`
I cannot see such variable in [this list](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/configuration/README.md)

vijayank88, 2022-06-16:
@Ryabkomv 
Check for following file in your local path: https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/openroad/groute.tcl

In line 45 add following lines and run the flow.
`set ::env(set_routing_alpha) 0`

Ryabkomv, 2022-06-16:
Thank, will do

vijayank88, 2022-06-16:
@Ryabkomv 
updated command format

Ryabkomv, 2022-06-16:
Got same error

[STEP 15]
[INFO]: Running Global Routing Resizer Timing Optimizations...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/ExampleRocketSystem/runs/RUN_2022.06.16_06.21.38/logs/routing/15-resizer.log
[ERROR]: Last 10 lines:
met2       Vertical      6182445       4528494          26.75%
met3       Horizontal    4121630       2882560          30.06%
met4       Vertical      2472978       1614742          34.70%
met5       Horizontal     824326        410880          50.16%

[INFO GRT-0101] Running extra iterations to remove overflow.
[ERROR GRT-0169] Net _0430322_: Invalid index for position (1983750, 3805350). Net degree: 6.
Error: resizer_routing_timing.tcl, 53 GRT-0169
child process exited abnormally

part of groute.tcl with set set_routing_alpha

```
source $::env(SCRIPTS_DIR)/openroad/set_routing_layers.tcl

set_macro_extension $::env(GLB_RT_MACRO_EXTENSION)
**set ::env(set_routing_alpha) 0**
source $::env(SCRIPTS_DIR)/openroad/layer_adjustments.tcl

set arg_list [list]
lappend arg_list -congestion_iterations $::env(GLB_RT_OVERFLOW_ITERS)
lappend arg_list -verbose
if { $::env(GLB_RT_ALLOW_CONGESTION) == 1 } {
    lappend arg_list -allow_congestion
}

```

Ryabkomv, 2022-06-16:
oh, sorry

Ryabkomv, 2022-06-17:
still failing at the same step with the same error

vijayank88, 2022-06-17:
@luis201420 
If I use `set_routing_alpha 0` before `global_route` command openroad crashed.
```
[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
/opt/rh/devtoolset-8/root/usr/include/c++/8/bits/stl_vector.h:932: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::reference = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Stack trace:
 0# 0x0000000000BF53D6 in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 1# 0x00007FD67ADD0400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# boost::stacktrace::detail::this_thread_frames::collect(void const**, unsigned long, unsigned long) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 5# 0x0000000002724FC2 in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 6# grt::FastRouteCore::updateRouteType1(int, grt::TreeNode const*, int, int, int, int, int, grt::TreeEdge*, int, int) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 7# grt::FastRouteCore::mazeRouteMSMD(int, int, float, int, int, bool, int, float, int, int, int) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 8# grt::FastRouteCore::run() in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 9# grt::GlobalRouter::findRouting(std::vector<grt::Net*, std::allocator<grt::Net*> >&, int, int) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
10# grt::GlobalRouter::globalRoute() in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
11# 0x0000000001416B8F in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
```

luis201420, 2022-06-18:
@vijayank88 I cloned the [github repository](https://github.com/Ryabkomv/openroad_issue_reproducible).
I unzipped the file **openroad_issue_reproducible.tar.gz** and I ran openROAD on the directory "openroad_issue_reproducible/home/ryabko/OpenLane/designs/ExampleRocketSystem/runs/RUN_2022.06.07_10.44.44/openroad_issue_reproducible/":
`./openroad -no_init run.tcl`
And I got the error:
```
[ERROR GRT-0169] Net _0428215_: Invalid index for position (2680650, 2176950). Net degree: 5.
[ERROR GUI-0070] Error: resizer_routing_timing.tcl, 54 GRT-0169
```
Then, I added on **openlane/scripts/openroad/resizer_routing_timing.tcl** file before the global_route command (Line 53-54) the `set_routing_alpha 0` command to solve the invalid index error.
With this command the design finishes the extra overflow iterations without problems but in the end the output gives **Routing congestion too high**.

dralabeing, 2022-06-24:
@luis201420  What is the next step here?

dralabeing, 2022-06-28:
@vijayank88  create a new GH with the ORFS crash you see on this test case. Please re-test and If there is no congestion issues close this one.

Ryabkomv, 2022-06-28:
Dear all, I'm little bit lost. Is there a way to resolve the issue? 
Is so, can you kindly let me know the needed config file for that?
In my case I still not able to pass stage 15. 
I tried default config for rocket chip and still have issue. The only successes was with picorv32 processor and simple components.

vijayank88, 2022-06-28:
@Ryabkomv 
Do you have any idea how much clock frequency it can be operated?

Ryabkomv, 2022-06-28:
This is good question. I used 10 MHz. I thought it is pretty reasonable. Never used less. 
Is it too fast?

vijayank88, 2022-06-28:
I tried with 30ns = 33.33MHz. Its passing with ORFS flow.
OpenLane Its crashed during global routing. Need to debug that.

vijayank88, 2022-06-28:
Final routing view from ORFS flow.
[image]

vijayank88, 2022-06-28:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts

vijayank88, 2022-06-28:
> @vijayank88 create a new GH with the ORFS crash you see on this test case. Please re-test and If there is no congestion issues close this one.

@dralabeing With ORFS flow mini rocket chip routing got completed. I think that crash due to memory issue. took 25GB RAM for post routing timing check and report.

Ryabkomv, 2022-06-28:
By the way,
in my case I have about 500Gb of RAM

antonblanchard, 2022-06-28:
@Ryabkomv using either DFFRAM or OpenRAM for those arrays would likely help a lot.

Ryabkomv, 2022-06-28:
Can you let me know which file you need?
 I think I provided the link [link](https://github.com/Ryabkomv/openroad_issue_reproducible) to the tarball with all possible logs along with config file containing all used parameters.

maliberty, 2022-06-28:
@Ryabkomv **no** that tarball does not contain all log files.

Ryabkomv, 2022-06-29:
Please, let me know which log file you need. Where are they located?
Sorry for my questions, but I am a newbie with OpenLane, just using step by
step instructions.

Ryabkomv, 2022-06-29:
> @Ryabkomv <https://github.com/Ryabkomv>
> designs/ExampleRocketSystem/runs/LatestRUNtag/logs
> Zip complete logs directory and upload here.
> Example: designs/ExampleRocketSystem/runs/RUN_2022.06.16_06.21.38/logs/

Ok, I've cleaned all the runs history, so will make a fresh run and upload
in a few hours.

Ryabkomv, 2022-06-30:
Please, use the [link ](https://github.com/Ryabkomv/openroad_issue_reproducible)to find all the file with log folder.

Ryabkomv, 2022-07-05:
Guys, were the logs I uploaded enough? Any others files needed?

vijayank88, 2022-07-05:
@Ryabkomv Have you checked your RTL to update memory blocks with OpenRAM/DFFRAM change?

Ryabkomv, 2022-07-05:
No, I didn't. I thought rocket project can be synthetized as it is.
Is the a simple way to update to OpenRAM/DFFRAM.
Should it be done through config file?

vijayank88, 2022-07-05:
Source RTL update can't do with config.tcl.

Ryabkomv, 2022-07-05:
Ok,
so in summary, current version of OpenLane can't synthetize rocket core with the [source ](https://github.com/chipsalliance/rocket-chip) as it is, right?
And the workaround is to dive into RTL?

Just for information, my interest in RISC-V is not about designing come advanced cores, I want to use standard core and supplement it with photonic accelerator through ROCC interface. So I did not expect to modify any RTL, just work on GDSII files from RISC-V  and photonic chip to connect corresponding wires.

vijayank88, 2022-07-05:
@Ryabkomv 
Please don't mislead it.
OpenLane synthesize the current rocket core. Its facing issue while running global routing.
So i am asking whether is it possible to optimize source RTL?

Ryabkomv, 2022-07-05:
I'm sorry if I said something wrong.  I meant the process of getting GDSII from open source rocket core written in chisel.
I will check what is DFFRAM/OpenRAM and how to use it in the project. RTL is a huge verilog file, If I will be able to find step by step instruction I will do. Otherwise it seems impossible in reasonable time with my background.

tspyrou, 2022-07-06:
@vijayank88 We should treat the RTL as fixed in this case. Could you try it with a lower utilization?

maliberty, 2022-07-07:
Sorry I've been off on an important task.  I looked at 8-resizer.log and I see
```
[INFO ODB-0131]     Created 1098296 components and 6431219 component-terminals.
...
[INFO RSZ-0038] Inserted 4000 buffers in 3063 nets.
[INFO RSZ-0039] Resized 790199 instances.
```
The vast majority of your gates are needing to be upsized.  This is likely to be the cause of the excess congestion (unfortunately it doesn't record the area increase due to sizing).  

In 7-global.log you have a fair number of ERC violations.
```
max slew violation count 86353
max fanout violation count 2
max cap violation count 14436
```
Its odd that at the end of synthesis you have very few (730 max slew).  Something seem off that synthesis see such a different view.  Perhaps there is an issue with wire load models?

Also a fair amount of slack repair needed:
```
tns -664458.94
wns -22.90
```
after CTS I see:
```
max fanout violation count 19090
```
I'm not sure why the max fanout is set to 5 but it seems quite low.  It does seem to be an OL default rather than specific to your design.

In all it seems like the design requires a lot of repair and that is increasing the utilization.  Are you trying to hit a particularly high frequency?  The debugging really needs to look at the full flow.   Is there a repo with the full design collateral necessary to run OL?

antonblanchard, 2022-07-07:
> I'm not sure why the max fanout is set to 5 but it seems quite low. It does seem to be an OL default rather than specific to your design.

I think 5 is overly pessimistic. I've been overriding the default, and using 10 (which still may be low).

antonblanchard, 2022-07-07:
I created https://github.com/The-OpenROAD-Project/OpenLane/pull/1187

vijayank88, 2022-07-07:
> @vijayank88 We should treat the RTL as fixed in this case. Could you try it with a lower utilization?

@tspyrou 
Yes. I am using RTL as it is and Re-Running the flow again with changes suggested above. Update you once got flow status.

[73 more replies]