The-OpenROAD-Project/OpenLane #1127: STA Failed with syntax error, unexpected REG

vijayank88, 2022-06-06:
### Description
Trying to run MPW design, from repo https://github.com/shalan/Caravel_Chameleon_SoC/tree/master/openlane/DFFRAM.

But its failed during STA with [ERROR STA-0164]

### Environment
```
python3 ./env.py issue-survey
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: c22c37b5121b0fa82c833e067841932dbb096972
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---

```

### Reproduction Material
* Upload a tarball containing the relevant design.
* List the commands used to run the design.
* $BIN/openroad
* source run.tcl

### Expected behavior
STA should pass.

### Logs
```
[STEP 2]
[INFO]: Running Single-Corner Static Timing Analysis...
[ERROR]: during executing openroad script /openlane/scripts/openroad/sta.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/cham_DFFRAM/runs/RUN_2022.06.06_06.56.38/logs/synthesis/2-sta.log
[ERROR]: Last 10 lines:
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /openlane/designs/cham_DFFRAM/runs/RUN_2022.06.06_06.56.38/tmp/merged.nom.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/designs/cham_DFFRAM/runs/RUN_2022.06.06_06.56.38/tmp/merged.nom.lef
[ERROR STA-0164] /openlane/designs/cham_DFFRAM/runs/RUN_2022.06.06_06.56.38/results/synthesis/DFFRAM.v line 283, syntax error, unexpected REG
STA-0164
child process exited abnormally

```

jjcherry56, 2022-06-06:
reg is a verilog rtl construct. opensta does not support rtl netlists because there is not timing information for rtl primitives.

closed 2022-06-06 by jjcherry56

radhe-2022, 2022-12-04:
I am also facing the same issue. How did you solve this ?

vijayank88, 2022-12-05:
@radhe-2022 
Can you share verilog file or repo link?

Eyantra698Sumanto, 2023-06-01:
Facing same issue