The-OpenROAD-Project/OpenLane #1131: Error during Generating PDN

ashkanr65, 2022-06-07:
I'm trying to learn how to work with OpenLane and I'm trying to run [this example](https://github.com/sudhamshu091/32-Verilog-Mini-Projects/tree/main/Universal_Shift_Register) on OpenLane, but I have several issues:
I made these changes in the config.tcl:
```
set ::env(PL_TARGET_DENSITY) "0.7"
set ::env(FP_CORE_UTIL) "70"
set ::env(PL_RANDOM_GLB_PLACEMENT) "1"
set ::env(PL_RANDOM_INITIAL_PLACEMENT) "1"
```
But I received this error:
[ERROR PDN-0108] Spacing (1.1200 um) specified for layer met5 is less than minimum spacing (1.6000 um).
Error: pdn_cfg.tcl, 129 PDN-0108
How can I fix it?

[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8852170/openroad_issue_reproducible.zip)

vijayank88, 2022-06-07:
@ashkanr65 
Its missing: `make survey || python3 ./env.py issue-survey`
Its difficult to proceed without survey

ashkanr65, 2022-06-07:
> 
`python3 ./env.py issue-survey
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v5.13.0-44-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 0dc6fb79c91082b94f8ded78d70f8bacbab96bf2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

0dc6fb7 2022-05-17T08:24:25-07:00 Enable PL_ROUTABILITY_DRIVEN and PL_TIME_DRIVEN (#1092) - Anton Blanchard -  (HEAD -> master, tag: 2022.05.18_02.12.32, origin/master, origin/HEAD)
fba1ad0 2022-05-17T14:45:07+02:00 Remove unnecessary calls to `read_verilog` (#1084) - Anton Blanchard -  ()
9d463c0 2022-05-17T14:43:10+02:00 Add `FP_IO_UNMATCHED_ERROR` flag (#1086) - Kareem Farid -  ()
`

ashkanr65, 2022-06-07:
> @ashkanr65 Its missing: `make survey || python3 ./env.py issue-survey` Its difficult to proceed without survey
@vijayank88 
Sorry, This is the survey:

python3 ./env.py issue-survey
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v5.13.0-44-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 0dc6fb79c91082b94f8ded78d70f8bacbab96bf2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

0dc6fb7 2022-05-17T08:24:25-07:00 Enable PL_ROUTABILITY_DRIVEN and PL_TIME_DRIVEN (#1092) - Anton Blanchard -  (HEAD -> master, tag: 2022.05.18_02.12.32, origin/master, origin/HEAD)
fba1ad0 2022-05-17T14:45:07+02:00 Remove unnecessary calls to `read_verilog` (#1084) - Anton Blanchard -  ()
9d463c0 2022-05-17T14:43:10+02:00 Add `FP_IO_UNMATCHED_ERROR` flag (#1086) - Kareem Farid -  ()

ashkanr65, 2022-06-07:
@vijayank88 
If I remove these lines I get this error:

[ERROR GPL-0302] Use a higher -density or re-floorplan with a larger core area.
Given target density: 0.55
Suggested target density: 0.61
Error: replace.tcl, 91 GPL-0302
child process exited abnormally

vijayank88, 2022-06-07:
set ::env(PL_TARGET_DENSITY) "0.61"

Add above variable and run again

vijayank88, 2022-06-07:
@ashkanr65 
You have reduced core area by increasing `CORE_UTIL ` to `70 ` causing the issue for power domain creation. With above logs, by using default `CORE_UTIL` of `50 `resolving the issue. So closing this.

closed 2022-06-07 by vijayank88