The-OpenROAD-Project/OpenLane #1140: Thin areas of sky130 met1 rails have no tap cells

antonblanchard, 2022-06-13:


maliberty, 2022-06-13:
@RTimothyEdwards I've moved this to OL as it is a setup issue not a tool issue.  I think the OL team should address this.

kareefardi, 2023-08-01:
I don't see a problem here too with regards to the original issue.

> > There is no problem here. You have set the decap as your endcap and they are within the 13um tapcell dist. The only issue is choosing decap as a endcap which it is not really.
> 
> ```
> # welltap and endcap cells
> set ::env(FP_WELLTAP_CELL) "sky130_fd_sc_hd__tapvpwrvgnd_1"
> set ::env(FP_ENDCAP_CELL) "sky130_fd_sc_hd__decap_3"
> ```
> 
> @RTimothyEdwards should we be using something other than `decap_3` for the endcap cells? (eg something with a tap)

If you would like to explore that more, please open a new issue.

closed 2023-08-01 by kareefardi