The-OpenROAD-Project/OpenLane #1141: Error while running python3 ./env.py local-install for non docker installation on Ubuntu 22.04

titan73, 2022-06-13:
Hi,

While following the installation guide on Ubuntu 22.04.

When I run:
     python3 ./env.py local-install

I give the following answers:
     Which UNIX/Unix-like OS are you using? [OTHER/ubuntu-20.04/centos-7/arch/macos] > ubuntu-20.04
     Do you want to install dependencies using your package manager? [NO/yes] > yes

The script failed later with the error:

make[2] : on entre dans le répertoire « /home/titan/eda/ol/install/build/repos/cvc/src_py »
pyinstaller -F check_cvc.spec --clean
90 INFO: PyInstaller: 5.1
90 INFO: Python: 3.10.4
92 INFO: Platform: Linux-5.15.0-37-generic-x86_64-with-glibc2.35
option(s) not allowed:
  --onedir/--onefile
makespec options not valid when a .spec file is given
make[2]: *** [Makefile:453 : check_cvc] Erreur 1
make[2] : on quitte le répertoire « /home/titan/eda/ol/install/build/repos/cvc/src_py »
make[1]: *** [Makefile:403 : all-recursive] Erreur 1
make[1] : on quitte le répertoire « /home/titan/eda/ol/install/build/repos/cvc »
make: *** [Makefile:344 : all] Erreur 2

Thanks

vijayank88, 2022-06-14:
@titan73 
Provide system survey `make survey || python3 ./env.py issue-survey`
https://github.com/The-OpenROAD-Project/OpenLane#prerequisites is per-requsities installed?

donn, 2022-06-14:
Ubuntu 22.04 is not supported. Please use the Docker image.

closed 2022-06-14 by donn

titan73, 2022-06-14:
titan@titan-host:~/eda/ol$ make survey || python3 ./env.py issue-survey
python3 ./env.py issue-survey
Kernel: Linux v5.15.0-37-generic
Distribution: ubuntu 22.04
Python: v3.10.4 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found.
OpenLane Git Version: 5a767e7fd208359f89d6784c8bc7533843013962
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: FAILED
/home/titan/eda/ol/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/titan/eda/ol/dependencies/verify_versions.py", line 72, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/titan/eda/ol/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

5a767e7 2022-06-10T15:28:44+02:00 Updates to `save_final_views` (#1137) - Mohamed Gaber -  (HEAD -> master, tag: 2022.06.11_01.51.35, origin/master, origin/HEAD)
ad24025 2022-06-07T08:52:04+02:00 Replace -slack_margin with -setup_margin/-hold_margin (#1130) - Anton Blanchard -  (tag: 2022.06.08_01.54.04)
b29b715 2022-06-07T08:43:35+02:00 Python Cleanup (#864) - Mohamed Gaber -  ()

titan73, 2022-06-14:
I run: apt install -y build-essential python3 python3-venv python3-pip
Docker has been removed to test local install.

titan73, 2022-06-14:
I tested with docker. The docker environment works but is rather empty as my daily tools are not present and I was unable to run klayout as it couldn't connect to display. This problems was easily worked around. I used klayout from the distribution.
More generally, I would like to test also the local installation as the target platform is redhat and with the IT policy I don't know yet how to install and build openlane with the correct dependency and if docker can be used. I just explore with my personal computer for now t get some insight and I don't mind debugging on unbuntu 22.04.

donn, 2022-06-14:
Well, the build scripts for most of the tools assume certain versions of tools are used.

The build system for the tool `cvc` breaks on later versions of Python (@d-m-bailey, would be nice if you can look into it). You can try `export SKIP_TOOLS=cvc` to bypass this issue.

d-m-bailey, 2022-06-14:
@donn Does this problem exist in cvc version 1.1.3 or later? I removed the pyinstaller steps.

titan73, 2022-06-14:
I had pyInstaller 5.1 installed. Using 4.1 with
pip install PyInstaller==4.1
seems to help as described in https://github.com/slundberg/shap/issues/2439

titan73, 2022-06-14:
This work only if I do a make in cvc directory.
python3 ./env.py local-install fetch the latest version of pyInstaller. I don't know yet how to change the version.
Btw do we use the standalone check_cvc? I don't see any call by grepping.

titan73, 2022-06-14:
This patch does it:

diff --git a/dependencies/python/compile_time.txt b/dependencies/python/compile_time.txt
index c013f73..a60f8c5 100644
--- a/dependencies/python/compile_time.txt
+++ b/dependencies/python/compile_time.txt
@@ -1,2 +1,2 @@
 cmake
-pyinstaller
\ No newline at end of file
+pyinstaller>=3.1.1,<5.0
I can do a PR.

donn, 2022-06-14:
Thanks for the info- let me try updating cvc first though and seeing how that goes

d-m-bailey, 2022-06-14:
> Btw do we use the standalone check_cvc? I don't see any call by grepping.
@titan73 The current Makefile will not install `check_cvc` by default. It's a GUI for managing the results that is not currently supported for openlane.

titan73, 2022-06-17:
Ah ok. So the dependency on pyinstaller can be removed when openlane switches to a new version of cvc.