The-OpenROAD-Project/OpenLane #1144: Unable to launch flow.tcl in local installation

titan73, 2022-06-14:
After worked around the pyinstaller problem (version 4.1 to use instead of latest 5.1) I managed to run the installl script
python3 ./env.py local-install.
According to the doc at https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/local_installs.md,
I should just do ./flow.tcl but I get:

ERROR]: run_non_interactive_mode missing required -design

    while executing
"parse_key_args "run_non_interactive_mode" args arg_values $options flags_map $flags -no_consume"
    (procedure "run_non_interactive_mode" line 10)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 384)

If I use /flow.tcl -design spm, it does not work either with:

[INFO]: Using design configuration at /home/titan/eda/ol/designs/spm/config.tcl
[INFO]: Sourcing Configurations from /home/titan/eda/ol/designs/spm/config.tcl
[ERROR]: PDK_ROOT is not specified. Please make sure you have it set.

    while executing
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 11)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 384)

donn, 2022-06-14:
That's uh. I mean you can invoke flow.tcl with the same arguments as inside the Docker container.

As for PDK_ROOT, you do need to set that separately, in addition to using a tool like [volare](https://github.com/efabless/volare) to get the PDK installed. 

We'll update the documentation to reflect all of that.

titan73, 2022-06-17:
Ah yeah indeed. Option -design is needed. I just followed blindy the doc at https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/local_installs.md.

Ok thanks. I will try once the doc is updated.

PR #1333 (2022-09-07): Documentation Fixes
\+ Local installer re-documented, volare integrated
\+ Create new "Macros/Chip Integration" section in `configuration.md`, document EXTRA_LIBS
\+ Document `CVC_SCRIPTS_DIR` in PDK variables ~ Replace broken links with permalinks `chip_integration.md`
~ `SYNTH_CLOCK_UNCERTAINITY` -> `SYNTH_CLOCK_UNCERTAINTY` (with translation behavior)
\- Remove unused `FP_CORE_MARGIN`

---
Resolves #1082
Resolves #1124 
Resolves #1144
Resolves #1332
closed 2022-09-07 by donn