The-OpenROAD-Project/OpenLane #1146: Max slew violations

Prit1097, 2022-06-14:
Hello, 

I have a design which actually passes the flow but it reports max slew violations. Is there any way to fix ?

The OL version I am using is : openlane:2022.05.11_01.53.47

This is the error I am getting in 29-rcx_sta.slew.rpt.
[image]

Thanks.

vijayank88, 2022-06-15:
Missing `make survey || python3 ./env.py issue-survey`

Can you provide test case?
If you want try your own, try to change following variable and check
```
set ::env(PL_RESIZER_MAX_SLEW_MARGIN) {20}
set ::env(GLB_RESIZER_MAX_SLEW_MARGIN) {10}
```

Prit1097, 2022-06-16:
@vijayank88 For the testcase I would have to talk to my manager. I will get back to you. Moreover, those two commands did not make any impact.

Thanks,
Prit

closed 2022-06-20 by donn