The-OpenROAD-Project/OpenLane #1149: Is there a way to visualize the gds file into the OpenROAD gui?

GabbedT, 2022-06-18:
Hi, I come from the ORFS flow and I found pretty useful the gui. I just finished synthetizing a design and i would like to visualize the gds file with the gui. The documentation doesn't seems to talk about OpenROAD commands etc.

maliberty, 2022-06-18:
Note that OR shows the final DEF not the gds.  You would have to use magic or klayout for that.

I believe you can just run with -gui.

GabbedT, 2022-06-19:
@maliberty This is what i get
```
OpenLane Container (074a92b):/openlane$ openroad -gui
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-'
process 9: The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.
Most likely, the application was supposed to call dbus_connection_close(), since this is a private connection.
  D-Bus not built with -rdynamic so unable to print a backtrace
Stack trace:
 0# 0x0000000001035AA6 in openroad
 1# 0x00007F2584CEA400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x00007F257A16B655 in /lib64/libdbus-1.so.3
 5# _dbus_warn_check_failed in /lib64/libdbus-1.so.3
 6# 0x00007F257D422BFB in /lib64/libQt5DBus.so.5
 7# QObject::event(QEvent*) in /lib64/libQt5Core.so.5
 8# 0x00007F258601BA55 in /lib64/libQt5Core.so.5
 9# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib64/libQt5Core.so.5
10# QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) in /lib64/libQt5Core.so.5
11# 0x00007F258606AA03 in /lib64/libQt5Core.so.5
12# g_main_context_dispatch in /lib64/libglib-2.0.so.0
13# 0x00007F2581C823F8 in /lib64/libglib-2.0.so.0
14# g_main_context_iteration in /lib64/libglib-2.0.so.0
15# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
16# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
17# QThread::exec() in /lib64/libQt5Core.so.5
18# 0x00007F257D4203B5 in /lib64/libQt5DBus.so.5
19# 0x00007F2585E71E71 in /lib64/libQt5Core.so.5
20# 0x00007F2587A34EA5 in /lib64/libpthread.so.0
21# clone in /lib64/libc.so.6
QThread::wait: Thread tried to wait on itself
QMutex: destroying locked mutex
QThread: Destroyed while thread is still running
Stack trace:
 0# 0x0000000001035AA6 in openroad
 1# 0x00007F2584CEA400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x00007F2585E5ECE1 in /lib64/libQt5Core.so.5
 5# QThread::~QThread() in /lib64/libQt5Core.so.5
 6# 0x00007F257D4205D9 in /lib64/libQt5DBus.so.5
 7# 0x00007F2584CEDCE9 in /lib64/libc.so.6
 8# 0x00007F2584CEDD37 in /lib64/libc.so.6
 9# 0x0000000001035ACB in openroad
10# 0x00007F2584CEA400 in /lib64/libc.so.6
11# gsignal in /lib64/libc.so.6
12# abort in /lib64/libc.so.6
13# 0x00007F257A16B655 in /lib64/libdbus-1.so.3
14# _dbus_warn_check_failed in /lib64/libdbus-1.so.3
15# 0x00007F257D422BFB in /lib64/libQt5DBus.so.5
16# QObject::event(QEvent*) in /lib64/libQt5Core.so.5
17# 0x00007F258601BA55 in /lib64/libQt5Core.so.5
18# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib64/libQt5Core.so.5
19# QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) in /lib64/libQt5Core.so.5
20# 0x00007F258606AA03 in /lib64/libQt5Core.so.5
21# g_main_context_dispatch in /lib64/libglib-2.0.so.0
22# 0x00007F2581C823F8 in /lib64/libglib-2.0.so.0
23# g_main_context_iteration in /lib64/libglib-2.0.so.0
24# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
25# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
26# QThread::exec() in /lib64/libQt5Core.so.5
27# 0x00007F257D4203B5 in /lib64/libQt5DBus.so.5
28# 0x00007F2585E71E71 in /lib64/libQt5Core.so.5
29# 0x00007F2587A34EA5 in /lib64/libpthread.so.0
30# clone in /lib64/libc.so.6
```

vijayank88, 2022-06-20:
@GabbedT 
Issue template always important when ever raising issue here.
```
make survey || python3 ./env.py issue-survey
```

vijayank88, 2022-06-20:
@GabbedT 
Check this link: https://l10nn.medium.com/running-x11-applications-with-docker-75133178d090

GabbedT, 2022-06-20:
@vijayank88 
```
gabriele@Gabbed:~/OpenLane$ make survey || pyton3 ./env.py issue-survey
python3 ./env.py issue-survey
Kernel: Linux v5.13.0-51-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: 074a92b7895be96a66db4318015ecb9f2c0f93c2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

074a92b 2022-06-14T15:01:11+02:00 Fix Magic Antenna Check (#1142) - Mohamed Gaber -  (HEAD -> master, tag: 2022.06.15_01.35.48, origin/master, origin/HEAD)
5a767e7 2022-06-10T15:28:44+02:00 Updates to `save_final_views` (#1137) - Mohamed Gaber -  (tag: 2022.06.11_01.51.35)
ad24025 2022-06-07T08:52:04+02:00 Replace -slack_margin with -setup_margin/-hold_margin (#1130) - Anton Blanchard -  (tag: 2022.06.08_01.54.04)
```

I checked the link and two things are not clear:
  * Where should I add the lines of code? 
  * `RUN apt-get install -y firefox ` What should i write instead of `firefox`
  
Then after doing this should i run `make mount` and everything is ok? Sorry if my question is stupid but i have 0 experience with docker and this type of environment.

vijayank88, 2022-06-20:
@GabbedT 
I took that link from previous github issue.
Normally X11 environment should be enabled. 
Maybe try this https://github.com/The-OpenROAD-Project/OpenLane/pull/740

vijayank88, 2022-06-20:
@donn 
Is that need any other libraries/packages to be updated to run `openroad -gui` inside docker?

GabbedT, 2022-06-20:
> @GabbedT I took that link from previous github issue.
> 
> Normally X11 environment should be enabled. Maybe try this #740
> 
> Just post complete log after running `make mount`
> 
> Also `echo $DISPLAY`

@vijayank88  I changed Makefile from this:

```
.PHONY: mount
mount:
	cd $(OPENLANE_DIR) && \
		$(ENV_START) -ti $(OPENLANE_IMAGE_NAME)-$(DOCKER_ARCH)
```
To
```
.PHONY: mount
mount:
	cd $(OPENLANE_DIR) && \
		docker run -it --rm \
			-e PDK_ROOT=$(PDK_ROOT) \
			-v $(OPENLANE_DIR):/openlane \
			-v $(PDK_ROOT):$(PDK_ROOT) \
			$(DOCKER_OPTIONS) $(OPENLANE_IMAGE_NAME)
```
This is the output:
```
gabriele@Gabbed:~/OpenLane$ make mount
cd /home/gabriele/OpenLane && \
	docker run -it --rm \
		-e PDK_ROOT=/home/gabriele/OpenLane/pdks \
		-v /home/gabriele/OpenLane:/openlane \
		-v /home/gabriele/OpenLane/pdks:/home/gabriele/OpenLane/pdks \
		--user 1000:1000 -e DISPLAY=:0 -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/gabriele/.Xauthority:/.Xauthority --network host efabless/openlane:074a92b7895be96a66db4318015ecb9f2c0f93c2
OpenLane Container (074a92b):/openlane$ echo $DISPLAY
:0
```
I'm not sure if I did the right thing.

> @GabbedT Until you get solution try standalone `openroad` for viewing def file. I think you already installed `OpenROAD-flow-scripts` So invoke gui following way `path/OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad` `read_lef path/name.lef` `read_def path/name.def` `gui::show`

LEF:
```
% read_lef /home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_12.02.10/results/final/lef/uart.lef

[INFO ODB-0222] Reading LEF file: /home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_12.02.10/results/final/lef/uart.lef
[WARNING ODB-0220] WARNING (LEFPARS-2008): NOWIREEXTENSIONATPIN statement is obsolete in version 5.6 or later.
The NOWIREEXTENSIONATPIN statement will be ignored. See file /home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_12.02.10/results/final/lef/uart.lef at line 2.
...
LIST OF UNDEFINED LAYER ERRORS
...
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  /home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_12.02.10/results/final/lef/uart.lef
```
DEF:
```
% read_def /home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_12.02.10/results/final/def/uart.def

[INFO ODB-0127] Reading DEF file: /home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_12.02.10/results/final/def/uart.def
[INFO ODB-0128] Design: uart
... 
LIST OF UNDEFINED SITES/LAYERS
...
ERROR (DEFPARS-6010): An error has been reported in callback. Error in file /home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_12.02.10/results/final/def/uart.def at line 180, on token ;.
Update the def file before parsing the file again.
[WARNING ODB-0149] DEF parser returns an error!
```

vijayank88, 2022-06-20:
For openlane flow by default the lef file read from `AutoC_UART/runs/RUN_2022.06.20_12.02.10/tmp/merged.unpadded.nom.lef`. So can follow
`read_lef home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_12.02.10/tmp/merged.unpadded.nom.lef`
`read_def home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_12.02.10/results/final/def/uart.def

donn, 2022-06-20:
@vijayank88 Nope. It works fine on every machine I tried.

GabbedT, 2022-06-20:
@vijayank88 Ok, now that works with ORFS even tho I can't visualize some heatmaps like routing congestions (not populated with data), are there any particular files that i should let OpenROAD read?

I also tried the same procedure on OpenLane, this time with the Makefile changes that i did. In this case DEF and LEF files don't even get read:

```
openroad> read_def /home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_13.47.38/results/final/def/uart.def
[ERROR ORD-0003] /home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_13.47.38/results/final/def/uart.def does not exist.
ORD-0003
while evaluating {read_def /home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_13.47.38/results/final/def/uart.def}

openroad> read_lef /home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_13.47.38/tmp/merged.unpadded.nom.lef
[ERROR ORD-0001] /home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_13.47.38/tmp/merged.unpadded.nom.lef does not exist.
ORD-0001
while evaluating {read_lef /home/gabriele/OpenLane/designs/AutoC_UART/runs/RUN_2022.06.20_13.47.38/tmp/merged.unpadded.nom.lef}
```

vijayank88, 2022-06-30:
@GabbedT 
OpenLane flow Inside docker you shouldn't use complete path. Just try relative path for reading lef file in openroad.

mbaykenar, 2022-07-14:
on ubuntu 20.04 I have the same issue

mbaykenar, 2022-07-14:
openroad -gui simply gives the same output above @GabbedT provided

maliberty, 2022-07-14:
How are you setting up your run?  Do you use 'make mount' to start docker?

GabbedT, 2022-07-15:
@maliberty Yes

maliberty, 2022-07-15:
In the Makefile it already has
```
# Allow using GUIs
UNAME_S = $(shell uname -s)
ifeq ($(UNAME_S),Linux)
DOCKER_OPTIONS += -e DISPLAY=$(DISPLAY) -v /tmp/.X11-unix:/tmp/.X11-unix -v $(HOME)/.Xauthority:/.Xauthority --network host
```

so you should not need to modify the mount target.  Did modifying it make it work for you?

maliberty, 2022-07-16:
I brought up a ubuntu 20.04 LTS x86/64 instance on gcp and got the same crash reported.  I see that running docker with "--security-opt apparmor=unconfined" or "--privileged" fixes the crash.

I guess apparmor is interfering with qt/dbus somehow.  Is anyone a docker pro who might know how to setup a custom apparmor config?  (@quantumhd @proppy @mithro )

vijayank88, 2022-07-18:
@donn 
For Ubuntu this issue occurs, not for Centos. Can you please fix this?

donn, 2022-07-18:
@vijayank88 I will when I get the chance. Please stop @'ing me.

maliberty, 2022-09-11:
@donn Its almost two months so I am checking in

donn, 2022-09-11:
https://github.com/The-OpenROAD-Project/OpenLane/pull/740#issuecomment-983145639

hmzsaeed, 2022-09-14:
I'm also getting this error when running `openroad -gui` after `make mount` in OpenLane:

```
hsaeed@merledu1:~/OpenLane$ make mount
cd /home/hsaeed/OpenLane && \
	docker run --rm -v /home/hsaeed/OpenLane:/openlane -v /home/hsaeed/OpenLane/designs:/openlane/install -v /home/hsaeed/OpenLane/pdks:/home/hsaeed/OpenLane/pdks -e PDK_ROOT=/home/hsaeed/OpenLane/pdks -e PDK=sky130A  --user 1034:1034 -e DISPLAY=:3 -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/hsaeed/.Xauthority:/.Xauthority --network host -ti efabless/openlane:63b5966c2874da8bf01a831f0ee9dc68382a4958-amd64
```
	
```
OpenLane Container (63b5966):/openlane$ openroad -gui
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-'
process 9: The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.
Most likely, the application was supposed to call dbus_connection_close(), since this is a private connection.
  D-Bus not built with -rdynamic so unable to print a backtrace
Stack trace:
 0# 0x0000000000C78EE6 in openroad
 1# 0x00007FA929DF2400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x00007FA91EFEB655 in /lib64/libdbus-1.so.3
 5# _dbus_warn_check_failed in /lib64/libdbus-1.so.3
 6# 0x00007FA9222E3BFB in /lib64/libQt5DBus.so.5
 7# QObject::event(QEvent*) in /lib64/libQt5Core.so.5
 8# 0x00007FA92B123A55 in /lib64/libQt5Core.so.5
 9# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib64/libQt5Core.so.5
10# QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) in /lib64/libQt5Core.so.5
11# 0x00007FA92B172A03 in /lib64/libQt5Core.so.5
12# g_main_context_dispatch in /lib64/libglib-2.0.so.0
13# 0x00007FA926D6E3F8 in /lib64/libglib-2.0.so.0
14# g_main_context_iteration in /lib64/libglib-2.0.so.0
15# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
16# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
17# QThread::exec() in /lib64/libQt5Core.so.5
18# 0x00007FA9222E13B5 in /lib64/libQt5DBus.so.5
19# 0x00007FA92AF79E71 in /lib64/libQt5Core.so.5
20# 0x00007FA92C8F6EA5 in /lib64/libpthread.so.0
21# clone in /lib64/libc.so.6
QThread::wait: Thread tried to wait on itself
QMutex: destroying locked mutex
QThread: Destroyed while thread is still running
Stack trace:
 0# 0x0000000000C78EE6 in openroad
 1# 0x00007FA929DF2400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x00007FA92AF66CE1 in /lib64/libQt5Core.so.5
 5# QThread::~QThread() in /lib64/libQt5Core.so.5
 6# 0x00007FA9222E15D9 in /lib64/libQt5DBus.so.5
 7# 0x00007FA929DF5CE9 in /lib64/libc.so.6
 8# 0x00007FA929DF5D37 in /lib64/libc.so.6
 9# 0x0000000000C78F0B in openroad
10# 0x00007FA929DF2400 in /lib64/libc.so.6
11# gsignal in /lib64/libc.so.6
12# abort in /lib64/libc.so.6
13# 0x00007FA91EFEB655 in /lib64/libdbus-1.so.3
14# _dbus_warn_check_failed in /lib64/libdbus-1.so.3
15# 0x00007FA9222E3BFB in /lib64/libQt5DBus.so.5
16# QObject::event(QEvent*) in /lib64/libQt5Core.so.5
17# 0x00007FA92B123A55 in /lib64/libQt5Core.so.5
18# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib64/libQt5Core.so.5
19# QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) in /lib64/libQt5Core.so.5
20# 0x00007FA92B172A03 in /lib64/libQt5Core.so.5
21# g_main_context_dispatch in /lib64/libglib-2.0.so.0
22# 0x00007FA926D6E3F8 in /lib64/libglib-2.0.so.0
23# g_main_context_iteration in /lib64/libglib-2.0.so.0
24# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
25# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
26# QThread::exec() in /lib64/libQt5Core.so.5
27# 0x00007FA9222E13B5 in /lib64/libQt5DBus.so.5
28# 0x00007FA92AF79E71 in /lib64/libQt5Core.so.5
29# 0x00007FA92C8F6EA5 in /lib64/libpthread.so.0
30# clone in /lib64/libc.so.6
```

hmzsaeed, 2022-09-14:
I tried and changed the MakeFile as follows:

**From:**
```
.PHONY: mount
mount:
        cd $(OPENLANE_DIR) && \
                $(ENV_START) -ti $(OPENLANE_IMAGE_NAME)-$(DOCKER_ARCH)
```
**To:**
```
.PHONY: mount
mount:
        cd $(OPENLANE_DIR) && \
                docker run -it --rm \
                        -e PDK_ROOT=$(PDK_ROOT) \
                        -v $(OPENLANE_DIR):/openlane \
                        -v $(PDK_ROOT):$(PDK_ROOT) \
                        --volume /tmp/.X11-unix:/tmp/.X11-unix \
                        --volume ${HOME}/.Xauthority:/.Xauthority \
                        --security-opt seccomp=unconfined \
                        --privileged \
                        $(DOCKER_OPTIONS) $(OPENLANE_IMAGE_NAME)
```
Although it worked but when I tried running OpenLane flow it gave me this error:
```
hsaeed@merledu1:~/OpenLane$ make mount
cd /home/hsaeed/OpenLane && \
	docker run -it --rm \
                        -e PDK_ROOT=/home/hsaeed/OpenLane/pdks \
                      	-v /home/hsaeed/OpenLane:/openlane \
                        -v /home/hsaeed/OpenLane/pdks:/home/hsaeed/OpenLane/pdks \
                        --volume /tmp/.X11-unix:/tmp/.X11-unix \
                        --volume /home/hsaeed/.Xauthority:/.Xauthority \
                       	--security-opt seccomp=unconfined \
                        --privileged \
                        --user 1034:1034 -e DISPLAY=:3 -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/hsaeed/.Xauthority:/.Xauthority --network host efabless/openlane:63b5966c2874da8bf01a831f0ee9dc68382a4958
OpenLane Container (63b5966):/openlane$ ./flow.tcl -design spm
OpenLane 63b5966c2874da8bf01a831f0ee9dc68382a4958
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[ERROR]: Failed to compare PDKs.
'NoneType' object has no attribute 'startswith'
Traceback (most recent call last):
  File "/openlane/dependencies/verify_versions.py", line 68, in verify_versions
    if pdk.startswith("sky130"):
AttributeError: 'NoneType' object has no attribute 'startswith'

[ERROR]: Please update your environment. OpenLane will now quit.
```

maliberty, 2022-09-14:
I  think it would be simpler to just

DOCKER_OPTIONS += --privileged 

probably something has gotten missed in your manually expanding the rule.

proppy, 2022-10-05:
I was unclear why `--privileged` is indeed given that the Makefile does volume mount the x11 unix socket, it seems related to: https://github.com/moby/moby/issues/38442

We could try to add a custom [seccomp configuration](https://docs.docker.com/engine/security/seccomp/) to OpenLane instead of adding `--privileged`, wdyt?

maliberty, 2022-10-05:
@proppy if you know how that would be great

mattvenn, 2022-10-07:
Adding ```DOCKER_OPTIONS += --privileged``` to the Makefile fixes it for me! I can finally try the openroad gui!

maliberty, 2022-11-14:
I see this cropped up again in slack.  If nobody wants to find a better solution can we just adopt DOCKER_OPTIONS += --privileged ?

proppy, 2022-11-15:
in order to evaluate potential candidate solution to fix this, it'd be good to know how many people affected by this issue do have setcomp enabled in their kernel configuration.

```
grep CONFIG_SECCOMP= /boot/config-$(uname -r)
```
can you hit below ↓↓↓
:ok_person: for `CONFIG_SECCOMP=y`
:no_good: for `CONFIG_SECCOMP=n`

maliberty, 2022-11-28:
@proppy you probably aren't going to get any more votes.  I suggest you move forward with something.

donn, 2022-11-28:
```
% zcat /proc/config.gz | grep SECCOMP
CONFIG_HAVE_ARCH_SECCOMP=y
CONFIG_HAVE_ARCH_SECCOMP_FILTER=y
CONFIG_SECCOMP=y
CONFIG_SECCOMP_FILTER=y
```

PR #1528 (2022-12-28): Makefile: disable seccomp to enable openroad ui
Resolves #1149

As described in https://docs.docker.com/engine/security/seccomp/#run-without-the-default-seccomp-profile
commit 4e910618 (2022-12-28): Makefile: disable seccomp to enable openroad ui (#1528)

Resolves #1149
closed 2022-12-28 by donn