The-OpenROAD-Project/OpenLane #1157: PDNGen issue in for a SOC with Mix of hard-macro with met3/met4 and met4/met5 power mesh

dineshannayya, 2022-06-22:
### Description
My SOC project has Multiple Hardmarco and all these hardmacro had power hook up in met-3/met-4.  Recently i have added a new macro which had power hook-up of met-4/met-5.   Openlane PDN generation flow does not shown any failure during PDN generation, But when we review the design, none of the hardmacro power is connected.

Here is working case power-up:
[image]

Here is Non-working case power-up:
[image]

### Environment

This issue is observed with latest MPW-6 openlane flow;  But PDN flows does not fails , No not clear on how to share the complete openlane setup here.  My Git work area: https://github.com/dineshannayya/riscduino

I have created a PDN standalone setup to re-produce the issue.

### Reproduction Material

Attached the Non working case:

[riscduino_pdn_non_working.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8954914/riscduino_pdn_non_working.tar.gz)

Attached the working case:
[riscduino_pdn_working.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8954952/riscduino_pdn_working.tar.gz)

Run command: run_cmd

To check the failure review the output/user_project_wrapper.def

### Expected behavior
PDN flow should fails to bring the user attention and also if there is way to handle these type of macro integration through Openlane flow.

### Lo
[24-pdn.log](https://github.com/The-OpenROAD-Project/OpenLane/files/8954960/24-pdn.log)
gs

vijayank88, 2022-06-22:
@dineshannayya 
Can you share me `pdn_working` case `GLB_RT_OBS`?

dineshannayya, 2022-06-22:
Is this one ?
set ::env(GLB_RT_OBS) "                              \
	                li1   150 130  833.1  546.54,\
	                met1  150 130  833.1  546.54,\
	                met2  150 130  833.1  546.54,\
                        met3  150 130  833.1  546.54,\

	                li1   950 130  1633.1 546.54,\
	                met1  950 130  1633.1 546.54,\
	                met2  950 130  1633.1 546.54,\
                        met3  950 130  1633.1 546.54,\

                        li1   150  750     833.1  1166.54,\
                        met1  150  750     833.1  1166.54,\
                        met2  150  750     833.1  1166.54,\
                        met3  150  750     833.1  1166.54,\
                        met1  950  650     1760   660 ,\
                        met3  950  650     1760   660 ,\
	                met5  0 0 2920 3520"

vijayank88, 2022-06-22:
As `digital_pll` macro used `met5` layer is the different from both testcase lef files.
As per maco hardening `met5` not allowed to use. plz check link here: https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/hardening_macros.md#power-grid-pdn

dineshannayya, 2022-06-22:
This digital_pll is hardmarco picked from caravel project. If caravel/efabless team able to integrate this macro why it's issue for my project ?   In MPW-5 time I have integrated DFFRAM from caravel which also had power hook in met4/met5 and I was able work with MPW-5 tool set which has DRC issue which got fixed with MACRO location movement. 
Is this new requirement for MPW-6 ?

vijayank88, 2022-06-22:
Ok. I will try to clone your repo and check it.

dineshannayya, 2022-06-22:
Here is working case with MPW-5 tool with same MPW-6 DEF and LEF included with digital_pll 

[image]

Attached my MPW-5 PDN setup
[riscduino_working_mpw5.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8955598/riscduino_working_mpw5.tar.gz)

vijayank88, 2022-06-22:
@maliberty fyi

vijayank88, 2022-06-22:
@dineshannayya 
Can you give `make survey || python3 ./env.py issue-survey`

dineshannayya, 2022-06-22:
python3 ./env.py issue-survey
Kernel: Linux v5.13.0-44-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 0dc6fb79c91082b94f8ded78d70f8bacbab96bf2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

0dc6fb7 2022-05-17T08:24:25-07:00 Enable PL_ROUTABILITY_DRIVEN and PL_TIME_DRIVEN (#1092) - Anton Blanchard -  (HEAD -> master, tag: 2022.05.18_02.12.32, origin/master, origin/HEAD)
fba1ad0 2022-05-17T14:45:07+02:00 Remove unnecessary calls to `read_verilog` (#1084) - Anton Blanchard -  ()
9d463c0 2022-05-17T14:43:10+02:00 Add `FP_IO_UNMATCHED_ERROR` flag (#1086) - Kareem Farid -  ()

arlpetergadfort, 2022-06-22:
@maliberty I'm unable to look at the files until late today. But if they are using the OpenLane flow, I don't know if there is a way to specify multiple macros in it. The default flow just makes a single -default grid for all macros (assuming a single layer for contact). If there are macros with different requirements, they need a grid defined for them.

arlpetergadfort, 2022-06-22:
In that case, the current setup will not be able to connect them. OL will need to be augmented to support multiple different definitions of macros (or the macro will need to have additional layers of metal inserted to comply with all the macros in the design.)

dineshannayya, 2022-06-22:
If we see the power mesh inside the digital pll ,  Vertical  => MET4 and Horizontal => MET5
[image]
There should not be any issue in Golbal Horizontal PDN Power Strip MPW5 to connecting the Macro Vertical MET4.  This was the case until the MPW-5 openlane tool set.  I used to face DRC issue due to over-lapping Global Horizontal Power Strip MET5 with Macro Horizontal MET5.  By moving the Hardmacro relative position I used to clear the DRC issue. 
.

dineshannayya, 2022-06-22:
If you see the current issue, PDN hook-up is missed for even macro with MET3/MET4 power mesh and I don't see any PDN error reporting during this case

vijayank88, 2022-06-25:
@arlpetergadfort 
Please check `digital_pll` macro lef. While integrating causing the issue.

`Additional info Shared by @dineshannayya :` In the failure setup, when i deleted the digital_pll definition in the .def file. I see PDN flow works fine and power-hook connected to all the hardmarco.

vijayank88, 2022-06-25:
@dineshannayya 
with latest openroad commit non-working log as follows:
```
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (-8.480um, 1910.880um), layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (-8.480um, 1820.880um), layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (-8.480um, 1730.880um), layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (-8.480um, 1640.880um), layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (-8.480um, 1550.880um), layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (-8.480um, 1460.880um), layer: 6.
[ERROR PSM-0069] Check connectivity failed.
Error: pdn.tcl, 117 PSM-0069
```
There was connectivity issue. 

But other working case its passing fine no error.
```
[WARNING PSM-0030] VSRC location at (2805.520um, 3090.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (2810.520um, 3080.880um).
[WARNING PSM-0030] VSRC location at (565.520um, 3230.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (570.520um, 3260.880um).
[WARNING PSM-0030] VSRC location at (1405.520um, 3230.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (1370.520um, 3260.880um).
[WARNING PSM-0030] VSRC location at (2245.520um, 3230.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (2250.520um, 3260.880um).
[INFO PSM-0031] Number of PDN nodes on net vccd1 = 1555.
[INFO PSM-0064] Number of voltage sources = 88.
[INFO PSM-0040] All PDN stripes on net vccd1 are connected.
########## IR report #################
Worstcase voltage: 1.80e+00 V
Average IR drop  : 2.01e-13 V
Worstcase IR drop: 3.26e-12 V
######################################
```

arlpetergadfort, 2022-06-25:
@dineshannayya I found a bug in the instance grid code, see PR: https://github.com/The-OpenROAD-Project/OpenROAD/pull/1997
For the PLL, because it has metal on met5, PDNGEN will not place straps over it, so it will remain disconnected. Looking at the grid in the PLL and the grid PDNGEN is building, it would short vssd1 and vccd1, so my recommendation would be to limit the macros to met4 to PDNGEN can connect at the last level.

dineshannayya, 2022-06-25:
@arlpetergadfort As i informed in in mail thread,  I am re-using this hardmarco available in Efabless/caravel project. As this block has some analog component, I will not be able to re-harden it my side.  
Regarding the latest PDNGEN integration issue with Hard macro with MET4/MET5.  Previously I was able integrated similar IP in MPW-5 tool set & I used to face DRC issue due to over-lapping at MET5. By moving the Hard macro relative position I used to clear the DRC issue. I have already attached the working MPW-5 example. 
I see this PDN issue is new restriction occurred in MPW-6 tool set onward.  
Currently I am planning to move ahead with using MPW-5 tools for my integration - As I am able to complete the Openlane flow without LVS issue with MPW-5 tools set.

maliberty, 2022-06-29:
If the old pdgen routed over the macro it was a bug.  The fact that you had to manually align the macro suggests it was so.  It is not a supported feature and would be an enhancement request.  You could 

- Unplace the macro, run pdngen and then replace the macro as you did manually before.
- Strip the top layer power shapes from the macro and allow pdngen to rebuild the connections.

dineshannayya, 2022-06-30:
@maliberty You will be able to move this ticket to enhancement request or you want me to raise a separate request. 

Just because pdgen need some spaces in met5 for power strip routing/power hook and restricting macro's from using met5 for routing does not look to be ideal solution. 
I can say old  pdngen bug was a nice to have feature (May be with extra option like allow DRC violation).
Previously it was one time effort to get the DRC failure location due to Global PDN strip overlap with Macro Power Strip and adjust macro relative position accordingly.   

I will cross-check with efabless team on their implementation plan for caravel project as i see they have multiple sub macro (listed below) uses MET5 for power routing 
caravel macro's using MET5: caravan_power_routing, caravel_clocking, chip_io_alt, chip_io, digital_pll, gpio_control_block,gpio_defaults_block,housekeeping, simple_por, spare_logic_block,user_id_programming

maliberty, 2022-07-11:
We can consider it as an enhancement request.  I know Peter is busy at the moment - how urgent is this for you?

dineshannayya, 2022-07-12:
This is not blocking my current work, I am using latest Openlane for Macro hardening and Previous openlane (MPW5) for top-level integration + Power Hook up. This will be creating confusion for any one try to fork my repo for their analysis.

dineshannayya, 2022-08-01:
@maliberty Is there is any plan to prioritize fixing this issue. Currently I need to switch between MPW-5 and MPW-7 tool set get clean database

maliberty, 2022-08-01:
There was no plan to raise the priority since you said it is not blocking your work.  Has that changed?

dineshannayya, 2022-08-02:
Now I am trying to integrate ADC macro which has two power domain. Macro with two power domain PDN integration only supported from MPW-7 tool set only.  Now i need to hack the digital pll lef to work with MPW-7 tool set. 
I also notice other project also reporting similar issue while integrating caravel digital pll macro. 
I appreciate if you can raise priority to have clean flow.

maliberty, 2022-08-02:
@arlpetergadfort when you are back can you give this some attention?  If not I'll find someone else to assign it to.

arlpetergadfort, 2022-08-09:
@dineshannayya this should allow you to connect the macros: https://github.com/The-OpenROAD-Project/OpenROAD/pull/2134
It will require placement to ensure the straps can go through the macro (the placement uploaded doesn't).

dineshannayya, 2022-08-09:
@arlpetergadfort I have picked your src/pdn/src/grid.cpp and locally compiled the openroad tool and validated  the pdn changes.  After moving the digital pll by 5 micro (100 to 105 micron) in Y Direction, I see there is vccd1/vssd1 feed through the digital pll in MET5.  
One additional observation is this first time i see MET4 also feed through this block.  I have not yet run full openlane flow to see any issue.

[image]

arlpetergadfort, 2022-08-09:
> 

There is no OBS blockage in met4 or met5 (so the behavior is going to be the same). If you move it over by 5um in X I think both the met4 straps will avoid the pins in met4 and connect. Or if you don't want met4, you will need to add a routing blockage over the block to prevent this.

dineshannayya, 2022-08-10:
@arlpetergadfort I am fine with met4 and met5 power strap inside the digital-pll as it help in better IR DROP. With Horizontal/Vertical adjustment, Now Both met4/met5 Power strip connected through the digital pll - That's Good. 
[image]

With Latest PDN flow, I am facing issue in global routing stages.  I see PDN met-4 routing at very near to Macro boundary and it's blocking the signal routing during global routing stage. How to manage this ? Is there is way to add PDN Offset ?  Note: all the SRAM macros has their Pins in MET-4 level.
[image]

Attaching previous working case, where i see there is enough space between macro and PDN met4
[image]

dineshannayya, 2022-08-10:
Last time i have faced similar issue : [https://github.com/The-OpenROAD-Project/OpenLane/issues/640] , But Now i see FP_PDN_HORIZONTAL_HALO & FP_PDN_VERTICAL_HALO value has no impact.

dineshannayya, 2022-08-10:
Here is standalone test case setup
[exp4.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9300443/exp4.tar.gz)

Run command: run_cmd

Input Lef file: ./input/merged.nom.lef
input Def file: ./input/4-io.macro_placement.def
Output Def file: ./output/user_project_wrapper.def

In Openroad gui you can look around the SRAM macro (u_tsram0_2kb,u_icache_2kb,u_dcache_2kb)

Let me know, if you need openlane based test case.

dineshannayya, 2022-08-10:
@arlpetergadfort This issue is not due to latest changes, but may be due to some of previous PDN changes.  I have been using 2 to 3 month old Openlane for top-level integration due to digital_pll PDN issue.  Any project with SRAM should have notice this issue.

arlpetergadfort, 2022-08-10:
> @arlpetergadfort This issue is not due to latest changes, but may be due to some of previous PDN changes. I have been using 2 to 3 month old Openlane for top-level integration due to digital_pll PDN issue. Any project with SRAM should have notice this issue.

The issue here is that you are specifying that you want met4/met5 as pins (PDNgen) will not trim back the pin layers. You can either remove that specification and it should work just fine, or PDNgen will need to be updated to allow trimming on the pin layers (it might make connecting power to blocks harder by making smaller targets). Since this is toplevel, I would remove the pin specification.

dineshannayya, 2022-08-11:
SRAM are hard macro for us, we don't have control on it's pin specification. I see it as a PDNGen issue not honoring the FP_PDN_HORIZONTAL_HALO & FP_PDN_VERTICAL_HALO value.  Do i need to raise separate issue for this ?

arlpetergadfort, 2022-08-11:
> SRAM are hard macro for us, we don't have control on it's pin specification. I see it as a PDNGen issue not honoring the FP_PDN_HORIZONTAL_HALO & FP_PDN_VERTICAL_HALO value. Do i need to raise separate issue for this ?

I think you misunderstand, I mean: https://github.com/The-OpenROAD-Project/OpenLane/blob/6ab944bc23688cae6dc6fa32444891a1e57715c8/scripts/openroad/pdn_cfg.tcl#L87

It's also possible the HALOs are not honored correctly. It's possible enforcing the halos will prevent you from correctly placing straps through the block. Please open a separate issue with a clean testcase. For now, I can tell you that removing the line I pointed to will trim away the subs, solving your problem.

dineshannayya, 2022-08-11:
@arlpetergadfort  With above pdn_cfg.tcl  change, I see there is additional halo space near macro and Global routing worked fine. But Openlane flow failed at "Writing Powered Verilog" stage with saying "No power ports found at the top-level."  below is details error message.  

[INFO]: Writing Powered Verilog (log: ../../../../../../../dev/null)...

----------------------------------------------------------
ERROR]: during executing: "openroad -python /openlane/scripts/odbpy/power_utils.py write_powered_def --input-lef /home/dinesha/workarea/opencore/git/riscduino/openlane/user_project_wrapper/runs/user_project_wrapper/tmp/merged.nom.lef --power-port vccd1 --ground-port vssd1 --powered-netlist /home/dinesha/workarea/opencore/git/riscduino/openlane/user_project_wrapper/runs/user_project_wrapper/tmp/synthesis/pg_define.v --output /home/dinesha/workarea/opencore/git/riscduino/openlane/user_project_wrapper/runs/user_project_wrapper/tmp/signoff/27-powered_def.def /home/dinesha/workarea/opencore/git/riscduino/openlane/user_project_wrapper/runs/user_project_wrapper/results/routing/user_project_wrapper.def |& tee /dev/null /home/dinesha/workarea/opencore/git/riscduino/openlane/user_project_wrapper/runs/user_project_wrapper/logs/signoff/28-write_powered_def.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/openlane/scripts/odbpy/power_utils.py", line 88, in write_powered_def
    ), "No power ports found at the top-level. Make sure that they exist and have the USE POWER|GROUND property or they match the arguments specified with --power-port and --ground-port."
AssertionError: No power ports found at the top-level. Make sure that they exist and have the USE POWER|GROUND property or they match the arguments specified with --power-port and --ground-port.
child process exited abnormally
----------------------------------------------------

I feel pdn_cfg.tcl change is impacting top-level power port hook-up. I will be raising separate ticket without pdn_cfg.tcl change.

dineshannayya, 2022-08-11:
I have submitted new issue at [https://github.com/The-OpenROAD-Project/OpenLane/issues/1262]

closed 2022-11-11 by dineshannayya