The-OpenROAD-Project/OpenLane #1158: Choosing the best configuration after an exploration run using run_designs.py.

ZeyadZaki, 2022-06-22:
### Prompt
In the readme file explaining the regression and exploration run: https://github.com/The-OpenROAD-Project/OpenLane/blob/master/regression_results/README.md. It was mentioned that the best configuration is chosen based on the least number of violations. In the example I've attached all of the successful runs had 0 violations so the function just chose the first successful run as the best one.  

### Proposal
The selection of the best design could be based first on the number of violations, if they are the same then take a look on the antenna violations (which were not taken into consideration in my case), after that if they are also equal you could look into other metrics such as the configuration that gave the least diearea, or the least physical cells. After that if all these metrics were identical you could also compare the power consumption of each configuration. In that way the process of choosing the best configurations would be much more accurate and more helpful for the user. 

[reg_20_06_2022_13_13.csv]
(https://github.com/The-OpenROAD-Project/OpenLane/files/8955918/reg_20_06_2022_13_13.csv)
[reg_20_06_2022_13_13_best.csv]
(https://github.com/The-OpenROAD-Project/OpenLane/files/8955919/reg_20_06_2022_13_13_best.csv)

vijayank88, 2022-06-22:
@ZeyadZaki 
Missing `make survey || python3 ./env.py issue-survey`

donn, 2022-06-23:
@vijayank88 That's for issues, not enhancement proposals